roleoroleo/yi-hack-Allwinner-v2 #454: Yi Factory Firmware Update - Camera now will not boot - Yi Outdoor 1080p

tunerooster, 2022-06-16:
Today, I responded to a request from the Yi  Android App to update the Yi factory firmware:
* From: 9.0.26.07.202104231123
* To:      9.0.26.08.202203181055

After the update the camera will not boot.  Yellow light comes on, flashes once, the goes solid yellow forever, and never boots.

**What must I do to recover?**

The camera and cable is glued in with silicone waterproofing, so it will mean a lot of work to remove it.

Does this mean **never upgrade firmware from the Yi App?**

Camera: Yi Outdoor 1080p IFUSY59UOS Model: YHS.3020
Hack: yi-hack-Allwinner-v2 0.2.2

roleoroleo, 2022-06-16:
> **What must I do to recover?**

You should follow the wiki: https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam

> The camera and cable is glued in with silicone waterproofing, so it will mean a lot of work to remove it.

If the cam is not working in this moment, you need to access the sd card to restore it.

tunerooster, 2022-06-16:
> If you are able, please, send me a backup of the new firmware.

Sorry, but how do I do that?  I assume you mean the new Yi Factory firmware...

tunerooster, 2022-06-20:
I got the cameras restored to their pre-factory-update state, so the Yi App is again prompting me with new firmware, which I am ignoring.

> If you are able, please, send me a backup of the new firmware.

I still do not know how to get you a copy of the new firmware.  I am more than happy to get it to you if there is a way.  Please advise..

Is there any way to know (in the future, say after a yi-hack 0.2.3 update) if is is safe to install new yi factory firmware?

Thanks!

roleoroleo, 2022-06-25:
I will try to download the new firmware from the yi website and to analyze it.
I can't know if it's safe to install a new firmware from yi because I don't know which partitions it contains.
For example, if it's only home partition it's safe.

tunerooster, 2022-06-25:
> For example, if it's only home partTTition it's safe.

Got it...  Thanks!  To your knowledge, has this happened to anyone else?

roleoroleo, 2022-06-29:
> I will try to download the new firmware from the yi website and to analyze it.

No luck, it's not available from yi website.

> To your knowledge, has this happened to anyone else?

I don't think so.

tunerooster, 2022-07-01:
If there is anything I can do to help, let me know.  Thanks

tunerooster, 2022-08-27:
Is there any change in the status on this issue?

The reason I ask is that I have recently tried to change "motion detection" to "ai_human_detection" and have encountered the bug referred to in #75.  If I understand it correctly, this is considered a yi-firmware bug and cannot be fixed in yi-hack.

So I am wondering if it might be fixed in the latest yi-firmware update?

However installing the new firmware breaks my yi-hack and the camera will not boot (i.e., the subject of *this* issue).

1. Is there any change in a possible fix for the ai_human_detection start issue?
2. Is there any way to install the new yi-firmware without bricking my camera?

Thanks!

Thanks!

roleoroleo, 2022-08-27:
I changed the human detection behavior in this commit: https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/9eed937497cd4c7d754a942089f43d20d69df860
Probably I fixed the problem.
About yi firmware update, I can't help you without a dump of the new firmware.
Did you make a backup before to restore the old firmware?

tunerooster, 2022-08-27:
> I changed the human detection behavior in this commit: 9eed937

I was not able to make  a backup of the new firmware because, after
updating, the camera would not boot.  Am I missing something?

Thank you, I will check out the commit!

tunerooster, 2022-08-28:
Regarding the commit 9eed937...

It appears that it is already included in the v0.2.2 I am running, is that correct?   I am confused as I thought you were saying this was likely a fix for the ai_human_detection  *not*  sending the `start` message (only `stop`).

My v0.2.2 release is correctly showing the ai_human_detection options on the mqtt web page and in the mqttv4.conf file and check_conf.sh file, which are the only files the commit includes.  I am still experiencing the the "no motion_start" issue.

1. Is the 9eed937 commit already included in 0.2.2?
2. Is it supposed to fix the #75 issue?

Thanks!

roleoroleo, 2022-08-28:
> It appears that it is already included in the v0.2.2 I am running, is that correct?

No, the last change is about 10 days ago.
If you want I can't send you a beta version.

tunerooster, 2022-08-28:
> > It appears that it is already included in the v0.2.2 I am running, is that correct?
> 
> No, the last change is about 10 days ago. If you want I can't send you a beta version.

You mean you CAN send me a beta?  Yes! Please do.  Thanks!

roleoroleo, 2022-08-28:
> You mean you CAN send me a beta?  Yes! Please do.  Thanks!

Yes... sorry :)
This afternoon.

tunerooster, 2022-08-28:
> > You mean you CAN send me a beta?  Yes! Please do.  Thanks!
> 
> Yes... sorry :) This afternoon.

No hurry... Thanks again.

roleoroleo, 2022-08-28:
Here it is:
[h30ga_0.2.2.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/9439709/h30ga_0.2.2.tar.gz)

tunerooster, 2022-08-28:
> Here it is: [h30ga_0.2.2.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/9439709/h30ga_0.2.2.tar.gz)

Got it, thanks!  I don't want to break things so I want to check with you first...

Can I install this *OVER* the existing root of the sd card (/tmp/sd) without retrieving the "hard to access" sd card from the camera?  My plan is to:
- mv /tmp/sd/Factory.done mv /tmp/sd/Factory.done.orig
- use lftp to upload the beta yi-hack (with overwrite existing files option) to the camera /tmp/sd
- reboot

I.e., I hope to retain my existing customizations (which are more than just the config files).

There is no way to do this from the web interface, right?  Like pass a file name to the "UPGRADE FIRMWARE" button.

roleoroleo, 2022-08-31:
In this manner, you will find some problem because binary files are in use: dropbear, pure-ftpd, busybox, rRTSPServer, etc...

There is a trick: copy the upgrade file in /tmp/sd with name h30ga_x.x.x.tgz (where 'x' is exactly the char 'x').
Go to the maintenance page and click "Upgrade".

tunerooster, 2022-09-01:
Outstanding!  Great to know!  Glad I asked first...  I'll try it now...  Thanks!

tunerooster, 2022-09-01:
Human detection is now working...     Thanks, you always come through!    :thumbsup:  :smiley:

roleoroleo, 2022-09-02:
Good news!
Thank your for your test.

closed 2023-01-08 by stale[bot]