roleoroleo/yi-hack-Allwinner-v2 #455: Issue with dead sd card

carlosvelosa, 2022-06-17:
My pt202 escam that was already hacked for 1 year ended up losing the card. I'm trying to hack again with two new sd card's and nothing. It just doesn't work anymore, I tried to reset the camera several times with the hack and nothing. When I remove the sd card from the camera, the Factory.Done folder appears and the hack doesn't work. I already tested removing the hack with a file in this part (https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Remove-the-hack-(deprecated)) and reinstalling the hack and it didn't work . The last process I need to do is "unbrick the cam" but I think the process is very complicated, I just don't know how to work with linux and I don't have any ttl-rs232/usb adapter. So both processes are out of the question. What more options do I have? Is there anything I can put on an sd card and unlock it, just like it is for installing the hack? I don't know what else to do, I've tried all the options at my disposal... My old sd card with the hack died...

roleoroleo, 2022-06-17:
New unbrick procedure doesn't require serial connection.
Try it.

dagiccross, 2022-06-21:
It's weird my sd card became corrupted after I deployed to my outdoor cam, it has 1GB from 8GB, and tried a lot of ways including different OS also no luck getting the 6.5GB back. Hope it doesn't happen to my other SD card again and this is only once...

carlosvelosa, 2022-06-21:
Even the new process to unbrick for me is quite complicated. I've tried but I just don't understand the instructions, it says putting it on a linux machine... But I don't understand how, if I had an image of how to do it I think it would be easier. The instructions are very superficial and for those who don't understand anything it's confusing.

carlosvelosa, 2022-06-21:
The old method of putting a file on the sd card and waiting for the camera to recognize it is much easier. How was the process remove the hack which is now deprecated. It would be much easier for me.

roleoroleo, 2022-06-28:
@carlosvelosa 
I can't simplify this part very much because I can't prepare a file for each cam. The procedure must use your backup file.
Send me your backup and I will create the file for you.

carlosvelosa, 2022-06-28:
[https://www14.zippyshare.com/v/junY1xiS/file.html](url)
@roleoroleo Thank you, sorry for my lack of knowledge in linux. And thank you very much.

carlosvelosa, 2022-06-28:
[https://www.sendspace.com/file/95cmnn](url)

roleoroleo, 2022-06-29:
Here it is:
[b091qp_unbrick.tgz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/9008279/b091qp_unbrick.tgz)

I think that backup partition should be enough to restore your cam.
But I created home and rootfs also, if you need them.
There are 2 different backup partitions:
- "factory" is the original partition
- "hacked" is like the original but it's ready for the hack (some files changed)

carlosvelosa, 2022-06-29:
@roleoroleo How do I run this file? Just put it on the sd card and that's it? Or do I have to change some name?

roleoroleo, 2022-06-29:
Just copy the file backup_091qp into your sd card and switch on the cam.

roleoroleo, 2022-06-29:
> Just copy the file backup_091qp into your sd card and switch on the cam.

I forgot a "b": backup_b091qp

If the cam renames Factory to Factory.done when you try to apply the hack, it means that the cam starts.
So the restore procecure described above with the file backup_b091qp must work.

carlosvelosa, 2022-06-30:
@roleoroleo Continues not working for me. I try with different sd card's and nothing. The camera simple don't work... Probably a internal problem or something. The cam renames Factory to Factory.done but the hack simple not work...

carlosvelosa, 2022-06-30:
You need a better and a faster way to unhack the camera like the same way you hack. If the camera don't works more, i will buy a new camera and put this in garbage.

roleoroleo, 2022-06-30:
Yi doesn't publish the firmwares so it's not possible to create a standard restore file.
Each user must create their own restore file.
I can assure you that your cam is not broken (or at least not completely).
If the folder is renamed, the cam boots and the main script (init.sh) is executed.
And the sd card is properly read and mounted.

We should restart from scratch and check every step.

closed 2023-01-07 by stale[bot]