roleoroleo/yi-hack-Allwinner-v2 #456: China's version Yi Outdoor O30 YHS.3020

antonsrc, 2022-06-20:
Hello!
My new camera from China can't detected by my iPhone, Android phone and PC version of YI Home. The camera speaks Chinese...

Camera spec:
Yi Outdoor O30
YHS.3020
IFCN...

I'm trying to install your h30ga, r40ga but apps on smartphones don't detect camera.
I tried to install the latest version of the application (Chinese version) on android, but when I tried to detect it, the camera said that it only works in China. But I'm not in China...

Does anyone have any idea?
Thanks!
Anton

roleoroleo, 2022-06-25:
First, you need to make a backup of your flash.
With a backup we can check the model of the cam and try to use a proxy to setup it.

antonsrc, 2022-06-25:
Here is what I have after the backup. Look please.
[https://github.com/antonsrc/temp/tree/main/backup](url)

antonsrc, 2022-07-01:
I did what you recommended to me, but camera says "this camera can only be used within china"...

I try to change \yi-hack\etc\proxychains.conf with the following content:
`socks5 59.49.33.221 7302`
`http 121.199.78.228 8888`
`http 124.112.115.137 9091`

Proxy took from [geonode.com](https://spys.one/free-proxy-list/CN/), [spys.one](https://spys.one/free-proxy-list/CN/) and etc.
But still fail...

antonsrc, 2022-07-02:
Now I read your [tutorial](https://github.com/roleoroleo/yi-hack-MStar/wiki/Use-proxychains-ng-to-connect-your-app-(thanks-to-@x571)) and tried to do something similar.

1. Add big proxy list (~100 items) with China's ip (socks4 and sock5) in proxychains.conf 
2. Insert sd in camera.
3. Using web ip:8080 to turn off unnecessary features, enable proxychains-ng, restart. 
4. Open proxychains page. Press button SAVE AND TEST CONFIGURATION.
5. A few minutes later I saw in form "Test with proxy":
```
{
	"ip": "124.164.249.226",
	"hostname": "226.249.164.124.adsl-pool.sx.cn",
	"city": "Tianjin",
	"region": "Tianjin",
	"country": "CN",
	"loc": "39.1422,117.1767",
	"org": "AS4837 CHINA UNICOM China169 Backbone",
	"timezone": "Asia/Shanghai",
	"readme": "https://ipinfo.io/missingauth"
}
```
6. Then I turn on unnecessary features (which turn off on step 3)
7. Reload camera
8. And fail again... "this camera can only be used within china"...

I work through Windows 10. I use Web server (web configutation interface (port 8080)). 
Tried to use FTP server via FileZilla (with port 21 and Default user is root), but there is no proxychains.conf  file in the etc folder.

What am I doing wrong?

roleoroleo, 2022-07-04:
What's your goal?
Do you want to use the yi cloud features? Do you want to use the original app?
Or not?

antonsrc, 2022-07-04:
The goal is to be able to access the camera remotely.  I want to watch the recording live or download the recordings.
The original app is not required.

ghanashyaml, 2022-07-04:
I am using 8 Yi Home 1080p cameras for exactly this purpose. Only need the
hack, SD card, enable swap file, and keep some % SD space free using an
option in the UI.

I had to do port forwarding for both the UI port 8080 and camera RPST port.

Working most of the times for all 8 cameras though sometimes some cameras
have stopped recording but live is working.

roleoroleo, 2022-07-05:
> The goal is to be able to access the camera remotely. I want to watch the recording live or download the recordings. The original app is not required.

If you don't want to use the app, switch on "Disable cloud" option and check if it works.

> I had to do port forwarding for both the UI port 8080 and camera RPST port.

It's not a good idea to forward http and rtsp ports.
Both protocols don't support tls/ssl encryption.
Consider using a vpn on your router.

antonsrc, 2022-07-11:
> If you don't want to use the app, switch on "Disable cloud" option and check if it works.

Great! Thank you! It works! :D

closed 2022-07-11 by antonsrc