roleoroleo/yi-hack-Allwinner-v2 #457: Compiling Error

tv4you2016, 2022-06-22:
hello I tried to compile the code as it is on the wiki and it gives me this error:

```
make[2]: Entering directory '/workspaces/yi-hack-Allwinner-v2-master/src/alsa-lib/alsa-lib-1.1.4.1/test'
make[2]: Nothing to be done for 'all-am'.
make[2]: Leaving directory '/workspaces/yi-hack-Allwinner-v2-master/src/alsa-lib/alsa-lib-1.1.4.1/test'
Making all in lsb
make[2]: Entering directory '/workspaces/yi-hack-Allwinner-v2-master/src/alsa-lib/alsa-lib-1.1.4.1/test/lsb'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/workspaces/yi-hack-Allwinner-v2-master/src/alsa-lib/alsa-lib-1.1.4.1/test/lsb'
make[1]: Leaving directory '/workspaces/yi-hack-Allwinner-v2-master/src/alsa-lib/alsa-lib-1.1.4.1/test'
Making all in utils
make[1]: Entering directory '/workspaces/yi-hack-Allwinner-v2-master/src/alsa-lib/alsa-lib-1.1.4.1/utils'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/workspaces/yi-hack-Allwinner-v2-master/src/alsa-lib/alsa-lib-1.1.4.1/utils'
make[1]: Entering directory '/workspaces/yi-hack-Allwinner-v2-master/src/alsa-lib/alsa-lib-1.1.4.1'
make[1]: Nothing to be done for 'all-am'.
make[1]: Leaving directory '/workspaces/yi-hack-Allwinner-v2-master/src/alsa-lib/alsa-lib-1.1.4.1'
Installing 'install.alsa-lib' in the firmware...

Done!

MOD_DIR:        /workspaces/yi-hack-Allwinner-v2-master/src/busybox
MOD_NAME:       busybox
MOD_INIT:       init.busybox
MOD_COMPILE:    compile.busybox
MOD_INSTALL:    install.busybox

Initializing busybox...

Compiling busybox...

make: *** No targets specified and no makefile found.  Stop.
```

roleoroleo, 2022-06-28:
Did you use the command to clone submodules?
`git submodule update --init`

roleoroleo, 2022-06-28:
There is a new procedure to build the hack, using devcontainer.
Check the readme.