roleoroleo/yi-hack-Allwinner-v2 #459: Escam PT-202 bricked ?

macedpo, 2022-06-25:
I have a PT202 (with hack) in service since several weeks and he stopped working today. The only thing he does is beeping 2 times every 30 seconds.
I'm afraid he's bricked. Anyone has an idea what those 2 beeps mean before i do the "unbricking" procedure ?
I already removed the hack and resetted the camera + installed a new hack but nothing helps...

roleoroleo, 2022-06-29:
Did you fix the problem?

macedpo, 2022-06-29:
Hi Roleo,
No... not yet. I'm struggeling with Putty.
Found the "gnd-rx-tx" connection on the PT202 circuit board and wired it to a PL2303HXA (old driver to run it on W11) but not getting any data on the putty screen.
Using 115200 8N1 settings on both the PC COM4 port and Putty.
There is a data flow though because i see the blue data LED flashing on the PL2303 when booting the PT202 but no data on the screen.
Ordered a more recent CP2102 USB --> TTL UART.
Regards,

macedpo, 2022-07-01:
Reveived the CP2102 uart today and now getting data on the terminal screen.
Did the "unbrick" procedure as discribed but after resetting the PT202 still having the same issue.
I attached a log-file. No problem if it is irreparable... was fun though playing with Putty-USB-->TTL uart.
[PT202.pdf](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/9029044/PT202.pdf)

macedpo, 2022-07-03:
.

closed 2022-07-03 by macedpo

roleoroleo, 2022-07-04:
The log shows errors about video buffer.
But it's not clear if it's a hw failure or not.