roleoroleo/yi-hack-Allwinner-v2 #462: y21ga | BFUS | v0.2.2: Continuous "click sound" from the cameras after updates?

outofsight, 2022-07-09:
Installed few Yi Hack cameras (also Dome X) in April. Updated firmwares in June. 

Now, several times a day, these camera start making a "click sound", in a loop of 20-30 times every 1sec.

The cameras are not rebooting.

There is SD on everyone with swap enabled.

roleoroleo, 2022-07-09:
If the camera is not in a boot loop the reason could be a the light of the room.
Dark to switch on ir leds and filter but not enough to mantain it.
So the cam switches on the night mode and then comes back to day mode.

zimgui, 2022-08-08:
My h51ga camera do the same sound, but only when switch video is closed and camera parked, I disabled the IR led and the sound is still there.

zimgui, 2022-08-08:
Another test, hide camera using PTZ commands (and status switch always ON) and no click sounds detected.

zimgui, 2022-09-01:
One more test, compiling with the latest commits make this sound disappear.

roleoroleo, 2022-09-02:
Which commit?

zimgui, 2022-09-02:
Sorry, I don't remember, I compiled it about 2 weeks ago, where I can find this information (SD, linux compilation dir)?

roleoroleo, 2022-09-02:
"2 weeks ago" is enough, thank you.

closed 2023-01-08 by stale[bot]

ashish-tandon, 2023-02-03:
Hi is there a solution for this

roleoroleo, 2023-02-04:
I didn't find a fix, but I'm, not able to repeat this behavior with my cams.