roleoroleo/yi-hack-Allwinner-v2 #463: Have snapshots, but no RTSP or ONVIF

Zeilloc, 2022-07-19:
Hello,
I manage to see snapshots using http://IPCAM:8080/cgi-bin/snapshot.sh?res=high&watermark=yes

But I got nothing using VLC open network stream rtsp://IPCAM/ch0_0.h264 or with Synology Surveillance Station using ONVIF IPCAM:80 (I have entered the login/passwords set in the http configuration page, but nothing appear). While configuring Surveillance Station, Connexion Test is OK, but when I save the page, it tries to activate ("Activation" is prompted) and then it shows "Deconnected".

Any idea?

roleoroleo, 2022-07-19:
About Synology, check the other issues: ONVIF implementation is not complete.
About VLC, it must work, start checking low res and try again.

Zeilloc, 2022-07-19:
The low res and only audio stream are not working either 😭

closed 2022-07-19 by Zeilloc