roleoroleo/yi-hack-Allwinner-v2 #469: MQTT Settings not working

hiagocosta, 2022-07-23:
I have Yi home camera with model y291ga
and this optoins on home assistant via mqtt doesn't change, only in camera webpage

could be something with this model, is on seria RFUS with 9.0.35 version firmware, using this version [https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/445#discussioncomment-2936174](url) of the hack:

[image]

roleoroleo, 2022-07-25:
How did you install these controls?

hiagocosta, 2022-07-26:
installed via MQTT, using HA integration and MQTT

roleoroleo, 2022-07-30:
I didn't write the mqtt discovery part.
I need some time to check it.

hiagocosta, 2022-08-01:
ok,

I will Wait

duhow, 2022-10-23:
Oh, I thought this was already working... 🤔 
I'll try have a look!

duhow, 2022-10-23:
For me in f2ddb4034a4a4632b2c4f25064eed03b7c0a149a (`v0.2.3`) only `ROTATE` change works.

closed 2022-10-26 by roleoroleo