roleoroleo/yi-hack-Allwinner-v2 #473: Process recording mp4 video file have 2 hours difference

mcanteri, 2022-08-18:
I've noticed that the process that records video file have 2 hours difference so file regarding 17 (5 PM) will go in the directory of 15 (3 PM) as we can see in the following log line:

2022-08-18 [17-02-35] [INFO] checkFiles: uploadToFtp SUCCEEDED - [/tmp/sd/record/2022Y08M18D15H/01M00S60.mp4]

Is there something to do to avoid this, or does the original software have to be fixed?

roleoroleo, 2022-08-18:
I think this is the normal behavior, but I'm not really sure.
Does the problem occur only with the hack?
What happens if you remove the sd card (so disable the hack) and configure the timezone with the app?

mcanteri, 2022-08-19:
I haven't tried without the hack, I donì't use the camera with the app. I'll try to understand a bit more when I have some time.

roleoroleo, 2022-08-19:
Me too, as soon as I have time and I'm close to the cam.

mcanteri, 2022-09-01:
I can see that using directly the program mp4record it seems to have the right time and date:
`
root@LittleSpy:/home/app# ./mp4record

mp4record.c(main-1254)[14:37:09.967]:init_finish(1), start_with_reset(0)

mp4record.c(main-1290)[14:37:09.968]:init_finish(1), start_with_reset(0)

mp4record.c(main-1292)[14:37:09.969]:mp4 mode start init_finish

mp4record.c(main-1298)[14:37:09.970]:fshare_open ok

mp4record.c(record_init-774)[14:37:09.971]:got video main vps=0 sps=20 pps=8, ts=4178433862, framerate=20 width=1920 height=1080

mp4record.c(record_init-796)[14:37:09.972]:got video sub sps=19 pps=8, ts=4178433884, framerate=20 width=640 height=360

mp4record.c(record_init-829)[14:37:09.973]:got audio aac config, ts=4178432190
main(1307) record_init ok
main(1311) create_thread_handyman ok, is_sd_exist = 1

mp4record.c(main-1317)[14:37:09.974]:mp4 mode got sd exist
file time list:

mp4record.c(main-1324)[14:37:10.000]:recordevent_init finish

mp4record.c(record_file-1037)[14:37:10.000]:record_init finish

mp4record.c(record_init-774)[14:37:10.000]:got video main vps=0 sps=20 pps=8, ts=4178433862, framerate=20 width=1920 height=1080

mp4record.c(record_init-796)[14:37:10.000]:got video sub sps=19 pps=8, ts=4178433884, framerate=20 width=640 height=360

mp4record.c(record_init-829)[14:37:10.000]:got audio aac config, ts=4178432190
[Peter:] need_split_file = 0

mp4record.c(record_file-1208)[14:37:24.810]:mp4 file begin record: ...
[Peter:] need_split_file = 0

recordevent_add_file_info(193) start:1662035844,duration:18

mp4record.c(record_file-1189)[14:38:00.112]:mp4 file end record: /tmp/sd/record/2022Y09M01D14H/37M24S18.mp4

mp4record.c(record_file-1208)[14:38:02.083]:mp4 file begin record: ...
[Peter:] need_split_file = 0

recordevent_add_file_info(193) start:1662035882,duration:31

mp4record.c(record_file-1189)[14:39:02.097]:mp4 file end record: /tmp/sd/record/2022Y09M01D14H/38M02S31.mp4
^C
root@LittleSpy:/home/app# date
Thu Sep  1 14:39:08 CEST 2022
root@LittleSpy:/home/app#
`
But more than this I haven't discovered. If you have a suggestion to understand more. it is welcome.

roleoroleo, 2022-09-02:
The key is check if the time is correct when the hack is not running.
Simply remove the sd card and try it.

mcanteri, 2022-09-02:
I've reset many times the camera without the SD card but it won't connect with the app and ask to reset it every time. I don't know what to do to make it connect again with the app, it wasn't much time ago the last time I reconnect with the app (actually now I remember it was the other camera, not this one - they are the same). 
I noticed that the date, connecting in telnet when it is online and not connected with the app, is exactly 2 hours difference respect our time, so from there, it is the error.
I set using the command date a correct date/time, but I saw that the program "cloud" take care of it and fix it bringing back the date as before.
Now I try to re-apply the hack hoping all will go fine ... 
Any suggestion on how to reset as fabric the camera?

mcanteri, 2022-09-03:
Today I was able to reconnect the camera with the app (had to choose another server, which means another area location - Korea) and I was able to verify the time zone in the setting of the camera, set it to Rome, and verify if something changed connecting in telnet: I saw that the date was unchanged, it remains like Sat Sep  3 13:40:11 GMT 2022 when actually it is now: sab 3 set 2022, 15:40:40, CEST. So for my time zone the two hours difference remains and I don't see a configuration that solves the problem. Probably the FTP script on the camera should query the date and do the work to correct the difference in time zones. It is not so difficult about the correction, it is a bit harder to make it general, querying the date and calculating the difference regarding the time zone configured in the hack, which looks a bit harder for me. Hope it helps, let me know if some other details could be of help :-) 
Thank you for your work ;-)

roleoroleo, 2022-09-03:
I think yi firmware works in GMT.
It probably saves dirs and files in GMT and the app converts the time when it shows them to the user.
I will remove the hack from my cam to confirm this behavior.

roleoroleo, 2022-09-08:
I can confirm it.
Without the hack, the cam uses GMT to save record files.
To fix the FTP upload problem we need to change the directory/file name inside ftppush.sh
Read with GMT and write with the configured timezone.

mcanteri, 2022-12-09:
It is still an open issue, if I am not wrong.

roleoroleo, 2022-12-09:
I had solved but I had not pushed.
Try this commit: https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/1cb84dcfa4d2c9ec14e47d422421f03418c4f2aa

mcanteri, 2022-12-09:
Okay, fine, thank you, I'll try asap :)

closed 2022-12-09 by mcanteri

mcanteri, 2026-11-09:
Hi @roleoroleo,

Coming back to this old issue because the behaviour has changed since it was
closed.

The 2022 fix (`1cb84dcf`, "Translate file name with TZ in ftppush") is no
longer used: commit `aafa864a` (19/01/2024, "Fix bug in ftppush.sh") removed
both calls to `translateFULLFNWithTz`. I checked the file at each commit:

| Commit | Calls to `translateFULLFNWithTz` |
|---|---|
| `1cb84dcf` (12/2022) | 4 |
| `e6a25b33` (02/2023) | 5 |
| `aafa864a` (01/2024) | 0 |
| `master` | 0 |

The function is still in `ftppush.sh`, but it is dead code in 0.3.7. As a
result `uploadToFtp` uses the directory and file names exactly as `rmm` writes
them: UTC when `TZ` is not set, local time when it is.

On my cameras (`y21ga` fw `9.0.19.02` and `y291ga` fw `12.x`, 0.3.7) I export
`TZ` before `rmm`/`mp4record` starts, and the folders now match the local
hour: record dir `2026Y11M09D11H` on the camera,
`.../Outdoor/2026Y11M09D11H` and `.../Studio/2026Y11M09D11H` on the FTP
server, with the last files in `10H` uploaded at 11:00 carrying a 10:59
timestamp. No offset.

So the remaining problem is the one in #1194: without setting `TZ` the
directories are in UTC. I am leaving this note for future readers, because
this issue still shows up when searching for the time offset.

Thanks!