roleoroleo/yi-hack-Allwinner-v2 #474: Enable Motion Detection

KayGundhardt, 2022-08-18:
Hello everybody,
Hi roleoroleo,

is there any way enabling the motion-detection WITHOUT revealing all my personal data and complete all the yi-app-account-register-process just for one setting-option?

:)

Any help would be appreciated!
Thank you!

Kay

roleoroleo, 2022-08-19:
I tried to implement this function several times but no luck.

darkfuncat, 2022-10-26:
Hello
Very +1 for this one feature request.
I understand that there is no way to prevent camera from recording non-stop (at least at each move).
An API call or MQTT message to disable motion detection record or even simple record would be great (with all other cameras, i enable this recording feature only in "alarm" mode, not "normal" mode)