roleoroleo/yi-hack-Allwinner-v2 #477: RTSP Connection refused by peer on YI Home BFUS

syb3ria, 2022-08-22:
Hi, as per the title I cannot connect to the live stream via the web browser, recording and snapshots works just fine. Last night I went to the whole proccess again deleting the cam from the YI app > formating the SD card > putting the yi hack on the SD card > pairing the cam in the YI app > then inserting the SD card again, all is working as it should apart from the live stream in the browser (it works fine in the YI app). 
Firmware : 0.2.2
Model : y21ga
VLC returns:

live555 error: Failed to connect with rtsp://192.168.1.12:554/ch0_0.h264
main error: connection failed: Connection refused by peer
satip error: Failed to connect to RTSP server 192.168.1.12:554
main error: connection failed: Connection refused by peer
access_realrtsp error: cannot connect to 192.168.1.12:554

I have another one with firmware 0.1.8 and the RTSP works as it should. Any ideas or should I just downgrade for the time being?

Regards,

roleoroleo, 2022-08-22:
I'm using the same model and it's working properly.
Try to reset to default from the web page (I don't mean the reset by back button).
And try to reconfigure manually the cam from scratch.

syb3ria, 2022-08-22:
It workded this way, reset settings and reboot but I find it strange since it was fresh install and I haven't changed anything apart from disabling the led light.

roleoroleo, 2022-08-24:
> I find it strange

I agree with you.

closed 2023-01-07 by stale[bot]