roleoroleo/yi-hack-Allwinner-v2 #480: Motion detection gets crazy when night mode is on

josecarre, 2022-08-24:
Hi, I have this camera:
Yi 1080p Home IFUS | 9.0.36* | y211ga

Loaded with last allwinner-v2 (0.2.2). It works fine... but the motion detector gets crazy detecting motion when there isnt. I suspect is related to night mode, because during the days it doesnt seem to get so crazy.
The motion detection sensitive is set to Low.

This is a history of the sensor... kind of maps to my day/night theory:
[image]

Any hint or similar experience? or any fix? :D

Thanks!

roleoroleo, 2022-08-24:
I have a similar experience.
In my case IR leds are not stable: the intensity of the light changes continuously.
And the cam detect a motion.

roleoroleo, 2022-08-24:
I have a similar experience.
In my case IR leds are not stable: the intensity of the light changes continuously.
And the cam detects a motion.

josecarre, 2022-08-25:
Thanks for your reply @roleoroleo , any possible solution? or just buy other camera? I dont see this issues in sonoff camera with sonoff-hack based on yi-hack, the motion detector in that camera works just great

Or is this something you have on roadmap to fix...? or I just trash this cam 🤣 

Thanks!

roleoroleo, 2022-08-25:
I think it depends on the hardware.
I have 3 yi_hack cameras and only 1 has this problem.
If you enable human detection it works better, but I don't know if this feature is ok for you.
Otherwise the trash is a good solution :)

josecarre, 2022-08-25:
oh, you mean 3 same Yi 1080p Home IFUS one some has the issue and others not? or you mean different serial numbers hence maybe some minor parts of the cam (IR led...etc) may change?

I'd love not to destroy our planet and recycle... I'll try with human detection but honestly I wanted the cam to detect **any** movement (vampires? 😆 )

Are you planning any upgrade for y211ga motion detection? or not much to do?

Congratulations about this amazing work, at least my sonoff-cam works just great!

roleoroleo, 2022-08-25:
> oh, you mean 3 same Yi 1080p Home IFUS one some has the issue and others not? or you mean different serial numbers hence maybe some minor parts of the cam (IR led...etc) may change?

Different models.

> I'd love not to destroy our planet and recycle... I'll try with human detection but honestly I wanted the cam to detect **any** movement (vampires? 😆 )
> 
> Are you planning any upgrade for y211ga motion detection? (forza bolognia! if it helps xD) or not much to do?

I will publish a new release in a few days with all the pushed commits.

> Congratulations about this amazing work, at least my sonoff-cam works just great!

Thank you. I published a new release today. Enjoy it!

Staini, 2022-08-29:
I think it's the hardware, too. Pixel artifacts occur when lighting conditions are poor. Even without night vision.

Here is an example:
[image]

roleoroleo, 2022-09-01:
Sorry but Tuya cams are not supported.

josecarre, 2022-09-02:
Ok.

Any recommendations of other cams to buy usable with this hack, I see a big table... ideally something that motion detector works fine is a MUST, night visio also (is por a garage), PTZ not needed.

btw, I will update here with the coming software update you mentiones just in case anything about the motion detection changed...? could is be possible to have deeper levels to put it in lower than current "low"?

roleoroleo, 2022-09-02:
> Any recommendations of other cams to buy usable with this hack, I see a big table... ideally something that motion detector works fine is a MUST, night visio also (is por a garage), PTZ not needed.

My knowledge about different model is not enough, maybe you could open a new issue to have a feedback from other users.

> btw, I will update here with the coming software update you mentiones just in case anything about the motion detection changed...? could is be possible to have deeper levels to put it in lower than current "low"?

This part is under the control of the yi processes, I can't change it.

josecarre, 2022-09-14:
I've updated to 0.2.3 and it still doesnt work fine, I had some hope 🤣 
@Staini did you find any solution? or anyone? thanks