roleoroleo/yi-hack-Allwinner-v2 #481: High RTSP Stream Fails to Work

Indianb0y016, 2022-08-28:
So my h30ga Outdoor Camera suffered a failed SD card. Replaced it with a new one, and the app updated the camera to a new firmware. 9.26.8 or something. From march 2022, from the firmware tag.
Camera boots fine, but the high rtsp stream will not work no matter what. Before this firmware update, it didnt have many issues. It would work. But now it doesnt. I SUSPECT that the rtsp stream is crashing. Weirdly, the grep command shows that rtsp server is running. But vlc will connect, run a few seconds, but not show any frames. 
The stream does work when set to low, but the resolution is 360p, which is... not ideal. 
I am not sure whats going on, but I have some time to troubleshoot so if there is anything I can try, please let me know! I dont know if this is from a yi hack issue, or a core camera firmware issue.

roleoroleo, 2022-08-28:
Please, send me a backup of the new firmware.

Indianb0y016, 2022-08-28:
Sure thing! Give me a few hours, as it is mounted outside high up, so it might take some time. Thank you for reaching out!

Indianb0y016, 2022-08-28:
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/9440110/backup.zip)

Should be complete. Loaded the backup archive on a clean sd card, and booted it. Performed nicely, and delivered a backup. Hopefully this is it.

Indianb0y016, 2022-08-28:
Had a look at the top command in ssh. ./rmm is using upwards of 80% of CPU. This is with everything disabled, apart from the low rtsp stream, httpd, ntp, and ssh.
If this is the usage with low rtsp stream being used, then I have no doubt in my mind that high stream is simply unable to run with little cpu room. Swap is also enabled, but hardly being used. 
Hope this also helps

Indianb0y016, 2022-08-28:
So I dont think thats the issue. After some persistence, I disabled as many services as possible, including the web page, with only ssh enabled. 
Flipped rtsp settings a few times, and I got the high preset to run, but it took a long time to load in vlc, and every several seconds, it would stop receiving new frames and just recycle older frames. Essentially, the high preset stops delivering frames. I have no clue why its not functioning, other than an inefficient rtsp service change after the last core firmware update.

roleoroleo, 2022-08-28:
rmm should use 30-50%.
imho 80% is not normal.
I will check your backup.

Indianb0y016, 2022-08-28:
Thank you! I will wait for a response and act accordingly. I had a feeling that rmm was causing a problem because its cpu usage is way too high

Indianb0y016, 2022-08-29:
Well, in a bizarre and really unfortunate change of events, I changed the stream quality on my camera to high, and rebooted. This time, it didnt come back. No light, no shutter clicks, nothing. Tried the unbrick procedure, with no dice. No response. The CPU heatsink does get warm, so there is some power being given, but I cannot see any rx or tx pads to serial solder to. At this point, I am giving up and I am shelving this camera for now until I have more time to experiment. I dont think its dead, unless it is and I am incorrect. EIther way, I am replacing it with a proper IP wifi camera by amcrest.
@roleoroleo Thank you so much for your skills, repos, and commitment to these cameras. I still have some yi cams that I use, so I am still going to be a user of your Allwinner yi hack, but V2 will no longer be applicable.
Again, thank you so much for your hard work! I am definitely buying you a beer mate!

roleoroleo, 2022-09-01:
I'm sorry for you partial brick.
Let me know if you will resume your experiments.

closed 2023-01-08 by stale[bot]