roleoroleo/yi-hack-Allwinner-v2 #482: Hack for h60ga completed but camera loses get stuck

avellma, 2022-08-29:
Hello here is the situation
Camera Type: 
- h60ga
- Model: Dome U Pro Official Firmware update (version 9.0.27.20_202108161046)
- I created a SD Card formatted with 32 FAT per the instructions
- I followed the instruction per 
- https://github.com/roleoroleo/yi-hack-Allwinner-v2
- I added the following files in the root dir
- 
[image]

- The files that I added were the following:

[image]

[image]

and from the 

[image]

Once I uploaded the SD card, I turned on the camera and I was not longer able to see the camera in the YI app and I was never able to connect to the camera using my web browser using the http://192.168.86.46:8080

I took the SD card out, rebooted the camera and I was able to connect to the camera using the YI App. So it seems I did not break anything.

I looked at the content of the SD card and I have the following in the root file:

[image]

Reading the repository, it was said if there was a file called "Factory.done", the the hack was successfully installed.

I am not sure what to do next.... any suggestions ?
I would love to be able to RTSP this camera........ Thanks for all those smart people.

avellma, 2022-08-29:
After searching the different documentation finally I was able to fix the problem using @roleoroleo file.
Thank you @roleoroleo for your hard work.

Here is what made my camera work

[h60ga_0.1.9_u.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7595993/h60ga_0.1.9_u.tar.gz)

roleoroleo, 2022-09-01:
You should use files from Release section: https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases
Not directly from the repo.
Decompress the content of the archive in the sd card.

avellma, 2022-09-02:
> You should use files from Release section: https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases Not directly from the repo. Decompress the content of the archive in the sd card.

Thank you for the input.

roleoroleo, 2022-09-03:
There is a issue that explains how to restore firmware 9 because firmware 11 is bugged.
But the cam is supported by this hack.

closed 2023-01-08 by stale[bot]