roleoroleo/yi-hack-Allwinner-v2 #484: Yi Dome Guard YRS3019

franciscofsales, 2022-08-30:
I got this Yi Dome Guard **YRS3019** with base version **9.0.05.33**.
Tried both r35gb and r30gb. 

- **r35gb**: loads up, but RTSP and snapshots do not work.
- **r30gb**: loads up, seems to work but RTSP is unstable - seems to have some frame drops, I can't get them on home assistant or frigate.

Any idea or debug strategy?

Thanks

franciscofsales, 2022-08-30:
Adding on to this, I left it running the entire day while at work and it crashes from time to time.. seems to be more frequent when tracking people.

roleoroleo, 2022-09-01:
Your cam is a r30gb and it's supported in beta stage.
Because it's the only cam where the stream is hevc compressed.
The stream probably crashes because the cam doesn't have enough resources.
Did you try to use the low res stream?

franciscofsales, 2022-09-01:
@roleoroleo just tried that... low res seems to work much much better... is low res not hevc?
what else can be done? someway i can help?

roleoroleo, 2022-09-02:
Low stream is h264.
There is an experimental method that reconfigures the cam in h264, editing mtdblock7 partition.
You should try but it's a risky operation (make a backup copy!).
Create a shell script file switch2h264.sh:
```
#!/bin/sh

printf "\01\00\00\00" | dd of=/dev/mtdblock7 bs=1 seek=780 count=4 conv=notrunc
```
And another shell script file switch2h265.sh:
```
#!/bin/sh

printf "\02\00\00\00" | dd of=/dev/mtdblock7 bs=1 seek=780 count=4 conv=notrunc
```

The 1st configure high stream to h264, the 2nd to hevc (reboot is needed).
After you change to h264. I don't know if the cam is supported by the grabber.
Maybe I need to make some change.
So, if the stream doesn't work, please send me 3 copies of the file /dev/shm/fshare_frame_buf
I will check it.

vipp1991, 2022-09-03:
I also have this camera and tried h264 script.
RTSP in H264 is now more stable.  I have tested Home Assistant streaming, recording files after movement to sd and everything looks good.

franciscofsales, 2022-09-05:
ran the script, and now its great. way more stable, working like a charm! thank you so much @roleoroleo !

closed 2022-09-05 by franciscofsales

roleoroleo, 2022-09-06:
Does it still work with the app too?

franciscofsales, 2022-09-06:
@roleoroleo snapshot is working properly, everything seems to work. I haven't tested the app... I don't use the app at all.
do you @vipp1991 ? if not I can try it for the sake of learning it

roleoroleo, 2022-09-06:
Thank you for your feedback.

vipp1991, 2022-09-07:
Yes, It still works in app. I tested it today.

roleoroleo, 2022-09-08:
Good news, I will add a link to this issue in the readme page.

Torin89, 2022-09-17:
> Low stream is h264. There is an experimental method that reconfigures the cam in h264, editing mtdblock7 partition. You should try but it's a risky operation (make a backup copy!). Create a shell [...]

Hi, I'm having the same issue but I don't know where to put the scripts or how to run them.

roleoroleo, 2022-09-20:
Decompress this archive into your sd card and run the script from a console.
[switch2h26x.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/9604814/switch2h26x.tar.gz)

Tom-1001, 2023-09-18:
I got this Yi Dome Guard and after install r30gb I can't connect. Error 3006

roleoroleo, 2023-09-24:
Please. explain me better.

Tom-1001, 2023-09-25:
I have camera YI DOME GUARD 
IFUSY36
SN: IR30GB
and I installed via SD card (FAT32):
Yi 1080p Dome *FUSE 12.1.05* r30gb
After restart camera can't connect with wifi. In APP I receive info: "The camera is offline, please reconnect it(-3006)"

roleoroleo, 2023-09-26:
Which version? 0.2.9?
How did you connect the cam to the wifi? Through the yi app?

Tom-1001, 2023-09-28:
Version: 12.1.05.1_202302091435

I inserted the SD card to the camera
Factory/
yi-hack/
lower_half_init.sh
Camera connect direct to router (LAN)
The camera turned and stopped (as always, like without hack)
If I want to open the web interface http://Ip my camera, no connection
If I go to App YI Home, I receive:
"The camera is offline, please reconnect it(-3006)"

After removing the SD card I have it:
backup/
Factory.done/
log/
yi-hack/
lower_half_init_sh

JordiMirror, 2023-09-29:
I just tried the scripts and they left me without an image in Yi-Hack, the official app does work well.

I haven't read that you had to make a backup copy and now I don't know what to do, can I reset the camera somehow?

Thank you.

roleoroleo, 2023-09-30:
> Version: 12.1.05.1_202302091435
>
> I inserted the SD card to the camera Factory/ yi-hack/ lower_half_init.sh Camera connect direct to router (LAN) The camera turned and stopped (as always, like [...]

Reset the cam using the button.
The cam should restart the wizard for the network connection.

roleoroleo, 2023-09-30:
> I just tried the scripts and they left me without an image in Yi-Hack, the official app does work well.
> 
> I haven't read that you had to make a backup copy and now I don't know what to do, can I reset the camera somehow?
> 
> Yi Dome Guard: Firmware version: 12.01.05.1_202302091435 Yi Hack version: rg30 - Firmware 0.2.9
> 
> Thank you.

Please, explain me better.

Tom-1001, 2023-09-30:
> > Wersja: 12.1.05.1_202302091435
> > Włożyłem kartę SD do kamery Factory/ yi-hack/ less_half_init.sh Kamera łączy się bezpośrednio z routerem (LAN) Kamera obróciła się i zatrzymała (jak zawsze, jak bez hackowania) Jeśli chcę otworzyć interfejs WWW http: [// IP](http://Ip) mojej kamery, brak połączenia Jeśli przejdę do aplikacji YI Home, otrzymuję: „Kamera jest w trybie offline, podłącz ją ponownie (-3006)”
> > Po wyjęciu karty SD mam ją: kopia zapasowa/ Factory.done/ log/ yi-hack/ less_half_init_sh
> 
> Zresetuj kamerę za pomocą przycisku. Kamera powinna ponownie uruchomić kreatora połączenia sieciowego.

Thank you for your help, but it doesn't work
After restart ( long press the button) camera started and blink red led. On router I can't find the camera (is no connected). The camera stands and nothing happens.

Tom-1001, 2023-09-30:
> Reset the cam using the button. The cam should restart the wizard for the network connection.

Thank you for your help, but it doesn't work
After restart ( long press the button) camera started and blink red led. On router I can't find the camera (is no connected). The camera stands and nothing happens.

roleoroleo, 2023-09-30:
You could try the unbrick procedure, to restore the factory fw.

Solarbotics, 2023-10-17:
Thanks so much for the work you put into this. I didn't salvage my Yi Dome Guard, but your instructions  and code (even compiled from your sources first time) were fine otherwise. I'm sure I just don't have a supported submodel.

FYI, it's the Dome Guard YRS-3019, with a PCB marked R30GB_IN_V1_0 2019/11/04C3, firmware 9.0.05.99_202011091624

I tried the candidates: 

- Yi Dome Guard YRS | 9.0.46* | r35gb (matched the prefix, but older firmware), and 
- Yi 1080p Dome *FUS | 9.0.05* | r30gb (prefix wrong, but matched the firmware & PCB markings)

I followed through on the script hacks for the h254/265, but neither produced video/stills. BUT the 2nd firmware DID move the Pan/Tilt!

Thanks again for trying to open up these devices - the effort is appreciated. Too bad these cams don't have a Pi-compatible pinout - I'd love to throw a Pi zero into one.

roleoroleo, 2023-10-19:
The label printed into the PCB doesn't match the model.
The important thing is the firmware version.
So, your cam is a R30GB.
Try again with the proper hack.

Solarbotics, 2023-11-04:
r30gb_0.3.0.tgz: This is so close to useful, it hurts. 

- I have Pan/Tilt **(YAY)**.
- I have motion-recording video to the SD card **(YAY)**.
- I am only lacking some sort of RTSP stream I can open with VLC (booo).

I've been beating up 5 or 6 types of cams to get some RTSP streams to my NAS & MotionEye, and this is the last cam on the list I want to add. I'm going to play with more of the RTSP settings; might have something misconfigured...

Again, thanks for the effort in putting these together. I am SO appreciating direct-to-cam access (Eufy needs cloud access to control PT? Jeebuz...)

Solarbotics, 2023-11-05:
Did some digging, and checked over the wd_rtsp.sh, and enabled the log file.
Something about the rRTSPServer is hung, as the CPU usage is returning 0, so it's in a "restarting RTSP" loop. 
`
root@YiPTZ:~/etc# top -b -n 2 -d 1 | grep rRTSPServer | grep -v grep | tail -n 1 | awk '{print $8}'
0.0
root@YiPTZ:~/etc#
`
The digging continues. Again, thanks for this great project. Looks like something a person could sink their teeth into!

roleoroleo, 2023-11-08:
> root@YiPTZ:~/etc# top -b -n 2 -d 1 | grep rRTSPServer | grep -v grep | tail -n 1 | awk '{print $8}' 0.0

This is strange...
What about a normal top command?

> I am only lacking some sort of RTSP stream I can open with VLC (booo).

Were you able to use the rtsp stream with MotionEye?
Have you tried onvif?

Solarbotics, 2023-11-19:
I just moved all my Motioneye installs to a new location, so I'm sitting here without one to test against. 
ONVIF? Good idea; I didn't check that. Let's see... ODM does detect it, but it fails to show any stream. It does detect it as (null data excluded):

- Manufacturer: Yi
- Model: Yi Hack
- Hardware: HFUS
- Firmware: 0.3.0
- Device ID: HFUSY36KHW2QH4200526
- ONVIF version: 20.12
- URI: http://192.168.1.166/onvif/device_service
---

Top results:
```shell
Mem: 57524K used, 3384K free, 1836K shrd, 1740K buff, 7852K cached
CPU: 66.6% usr 33.3% sys  0.0% nic  0.0% idle  0.0% io  0.0% irq  0.0% sirq
Load average: 5.00 4.07 3.75 4/109 17700
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
 1246     1 root     S    43072 70.4   0 39.9 ./rmm
17106  1769 root     S<    4100  6.7   0  6.6 rRTSPServer -m r30gb -r high -a no -p 554
 4983  4931 root     R     1332  2.1   0  6.6 {thumb.sh} /bin/ash /tmp/sd/yi-hack/script/thumb.sh cron
17689 17485 root     R      988  1.6   0  6.6 top
    7     2 root     RW       0  0.0   0  6.6 [rcu_preempt]
 6034  6030 root     S     3512  5.7   0  0.0 /tmp/sd/yi-hack/bin/imggrabber -m r30gb -r high -w
 1592     1 root     S     3172  5.1   0  0.0 ./mp4record
 1769     1 root     S     2888  4.7   0  0.0 {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
12192     1 root     S     1488  2.4   0  0.0 /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_su
 4931  4930 root     S     1332  2.1   0  0.0 {thumb.sh} /bin/ash /tmp/sd/yi-hack/script/thumb.sh cron
  598     1 root     S     1164  1.9   0  0.0 ./dispatch
12409     1 root     S     1048  1.7   0  0.0 udhcpc -i wlan0 -b -s /backup/tools/default.script -x hostname:YiPTZ
15005 14465 root     S      996  1.6   0  0.0 top
    1     0 root     S      980  1.6   0  0.0 /sbin/init
14465 14324 root     S      980  1.6   0  0.0 -ash
17485 17279 root     S      980  1.6   0  0.0 -ash
  541     1 root     S      980  1.6   0  0.0 -/bin/sh
 6030  1655 root     S      980  1.6   0  0.0 sh -c export MOD=$(cat /tmp/sd/yi-hack/model_suffix); /tmp/sd/yi-hack/bin
 4930  1858 root     S      980  1.6   0  0.0 /bin/sh -c /tmp/sd/yi-hack/script/thumb.sh cron
 1858     1 root     S      968  1.5   0  0.0 {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/crond -c /va
 1742     1 root     S      964  1.5   0  0.0 {ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/ntpd -p pool.
 1635     1 root     S      964  1.5   0  0.0 {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/httpd -p 80
14324  1651 root     S      916  1.5   0  0.0 /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p
13474  1651 root     S      916  1.5   0  0.0 /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p
17279  1651 root     S      916  1.5   0  0.0 /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p
 1651     1 root     S      892  1.4   0  0.0 /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p
 1655     1 root     S      880  1.4   0  0.0 mqttv4
  446     2 root     SW       0  0.0   0  0.0 [irq/302-sunxi-m]
  587     2 root     SW       0  0.0   0  0.0 [RTW_CMD_THREAD]
  527     2 root     SW       0  0.0   0  0.0 [mmcqd/0]
    3     2 root     SW       0  0.0   0  0.0 [ksoftirqd/0]
  294     2 root     SW       0  0.0   0  0.0 [kswapd0]
  518     2 root     SW<      0  0.0   0  0.0 [kworker/0:1H]
  210     2 root     SW       0  0.0   0  0.0 [kcompactd0]
18237     2 root     SW       0  0.0   0  0.0 [kworker/u2:1]
21522     2 root     SW       0  0.0   0  0.0 [kworker/0:2]
  208     2 root     SW<      0  0.0   0  0.0 [writeback]
12661     2 root     SW       0  0.0   0  0.0 [kworker/u2:2]
    2     0 root     SW       0  0.0   0  0.0 [kthreadd]
11133     2 root     SW       0  0.0   0  0.0 [kworker/u2:0]
  534     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd4]
  448     2 root     SW       0  0.0   0  0.0 [irq/166-sdc0 cd]
  526     2 root     SW<      0  0.0   0  0.0 [bioset]
  418     2 root     SW<      0  0.0   0  0.0 [bioset]
  408     2 root     SW<      0  0.0   0  0.0 [bioset]
  403     2 root     SW<      0  0.0   0  0.0 [bioset]
  413     2 root     SW<      0  0.0   0  0.0 [bioset]
  383     2 root     SW<      0  0.0   0  0.0 [bioset]
  378     2 root     SW<      0  0.0   0  0.0 [bioset]
  388     2 root     SW<      0  0.0   0  0.0 [bioset]
  284     2 root     SW<      0  0.0   0  0.0 [spi0]
  263     2 root     SW<      0  0.0   0  0.0 [cfg80211]
  255     2 root     SW       0  0.0   0  0.0 [sys_user]
  269     2 root     SW<      0  0.0   0  0.0 [watchdogd]
  393     2 root     SW<      0  0.0   0  0.0 [bioset]
  211     2 root     SW<      0  0.0   0  0.0 [crypto]
  207     2 root     SW       0  0.0   0  0.0 [oom_reaper]
  212     2 root     SW<      0  0.0   0  0.0 [bioset]
    9     2 root     SW       0  0.0   0  0.0 [rcu_bh]
    8     2 root     SW       0  0.0   0  0.0 [rcu_sched]
   10     2 root     SW<      0  0.0   0  0.0 [lru-add-drain]
    5     2 root     SW<      0  0.0   0  0.0 [kworker/0:0H]
   11     2 root     SW       0  0.0   0  0.0 [kdevtmpfs]
  214     2 root     SW<      0  0.0   0  0.0 [kblockd]
  398     2 root     SW<      0  0.0   0  0.0 [bioset]
15889     2 root     SW       0  0.0   0  0.0 [kworker/0:1]
31488     2 root     SW       0  0.0   0  0.0 [kworker/0:0]

```

Solarbotics, 2023-11-19:
Update. I did turn off the "disable cloud" button (that allows the Yi App to see it), and it DOES stream in the Yi App to my Android. Weird. RTSP and ONVIF still dead though.

Well, good on ya for making the YiCam app still functional with this new firmware in place! Totally didn't expect that to work.

roleoroleo, 2023-11-19:
In the previous top command rRTSPServer uses 0.5% of CPU.
I don't understand why
`top -b -n 2 -d 1 | grep rRTSPServer | grep -v grep | tail -n 1 | awk '{print $8}'`
returns 0

Please try to increase the delay time:
`top -b -n 2 -d 2 | grep rRTSPServer | grep -v grep | tail -n 1 | awk '{print $8}'`
or
`top -b -n 2 -d 3 | grep rRTSPServer | grep -v grep | tail -n 1 | awk '{print $8}'`

Solarbotics, 2023-11-20:
```shell
root@YiPTZ:~# top -b -n 2 -d 2 | grep rRTSPServer | grep -v grep | tail -n 1 | awk '{print $8}'
0.0
root@YiPTZ:~#
``` 
```shell
root@YiPTZ:~# top -b -n 2 -d 3 | grep rRTSPServer | grep -v grep | tail -n 1 | awk '{print $8}'
0.3
root@YiPTZ:~#

```

Solarbotics, 2023-11-20:
netstat does indeed show nothing on port 554. Continuing investigation...
```shell
root@YiPTZ:/tmp/sd# /tmp/sd/yi-hack/bin/netstat -an
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
tcp        0    448 192.168.1.166:22        192.168.1.126:29892     ESTABLISHED
netstat: /proc/net/tcp6: No such file or directory
udp        0      0 0.0.0.0:3702            0.0.0.0:*
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node Path
unix  2      [ ]         DGRAM                      1888 /var/run/wpa_supplicant-global
unix  2      [ ]         DGRAM                      1909 /var/run/wpa_supplicant/wlan0
unix  2      [ ]         DGRAM                      1573
unix  2      [ ]         DGRAM                      2196
unix  2      [ ]         DGRAM                      2458
unix  2      [ ]         DGRAM                      2193
unix  2      [ ]         DGRAM                      1103
```

Solarbotics, 2023-11-20:
trying to manually run the rRTSPServer. I'm unsure what I'm looking for on the debug:
```shell
root@YiPTZ:/tmp/sd# rRTSPServer -m r30gb -r low -a no -p 544 -d 1
1700456857894: the size of the buffer is 1786156
1700456857895: capture - mapping file fshare_frame_buf, size 1786156, to b6ca1000
1700456857897: capture - closing the file fshare_frame_buf
1700456857897: capture - starting capture main loop
1700456857898: capture - buf_idx_end == buf_idx_end_prev
1700456857910: capture - buf_idx_end == buf_idx_end_prev
1700456857921: capture - buf_idx_end == buf_idx_end_prev
1700456857931: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1700456857932: capture - frame_sync == 0
1700456857942: capture - buf_idx_end == buf_idx_end_prev
1700456857952: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1700456857952: capture - frame_sync == 0
1700456857962: capture - buf_idx_end == buf_idx_end_prev
1700456857972: capture - buf_idx_end == buf_idx_end_prev
1700456857983: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1700456857983: capture - frame_sync == 0
1700456857993: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1700456857993: capture - frame_sync == 0
```

Solarbotics, 2023-11-20:
WHOA. Changed the RTSP server to model r35gb, and I have a feed!

roleoroleo, 2023-11-20:
Thank you for the donation.
But I don't understand why now it's working.
If your cam has fw 9.0.05 can't be a r35gb. It must work with r30gb.

Another detail. Editing the model_suffix file is not enough to change model.
You should overwrite /tmp/sd/lower_half_init.sh with the proper version.

JohnHolg, 2024-04-19:
Hi,

I installed FW 12.1.05.1 und YI-hack r30gb.

PTZ and rtsp-stream didn't work.

Can you help?

roleoroleo, 2024-04-20:
Please, open a new issue and add more details.

landgraf86, 2024-05-07:
> ## Whoa! Changed the RTSP server to model r35gb, and I have a feed!
> I editted the /tmp/sd/yi-hack/model_suffix file to r35gb, rebooted, and now I have RTSP and ONVIF functionality. Woohoo!
> I don't seem to have snapshots, but I can live without that for now. Also the PTZ page doesn't show video, but does control the cam movement. I see in the HTML that these are related. Will do more digging.
> Imma Dumbass. Just turn on the snapshot function. Sigh. ALL GOOD NOW. WONDERFUL! Looking for a donation link next...

+1. I change model_suffix file to r35gb too and feeds working for now. 
My camera:
Model: Yi Dome Guard, YRS 3019
SN: KFUSY36
Firmware: 12.1.05.1_202302091435

roleoroleo, 2024-05-10:
Same as above.
Editing the model_suffix file is not enough to change model.
You should overwrite /tmp/sd/lower_half_init.sh with the proper version.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/sysroot/r30gb/lower_half_init.sh