roleoroleo/yi-hack-Allwinner-v2 #487: RTSP/Snapshots not available on network without Internet access, why?

dzek69, 2022-08-30:
Hey, I just installed this software on `Yi 1080p Home IFUS / 9.0.36* / y211ga` and it was working all fine until I switched to a WIFI network that does not have access to the internet.

I tried switching various settings, but nothing helps. Switching back to WIFI with internet access fixed the issue. Switching again to no-internet network made the streams stop working.

On any network web interface is working, I just cannot access snapshots and rtsp. Is there any way around it?

Ports are visible as open via nmap:
```
Discovered open port 80/tcp on 192.168.100.237
Discovered open port 554/tcp on 192.168.100.237
Discovered open port 8080/tcp on 192.168.100.237
```

Trying to open snapshot via browser results in infinite loading blank tab,
trying to use ffmpeg to save the stream with
```
"ffmpeg" "-xerror" "-rtsp_transport" "tcp" "-i" "rtsp://192.168.100.237:554/ch0_0.h264" "-c:a" "aac" "-vsync" "2" "-async" "1" "-c:v" "copy" "-map" "0" "-movflags" "+faststart" "-tune" "zerolatency" "-f" "segment" "-segment_time" 60 "-segment_format" "mp4" "-reset_timestamps" "1" "-strftime" "1" "./dancing-full-%Y-%m-%d_%H-%M-%S.mp4"
```

results with this output:
```
ffmpeg version 4.3.4-0+deb11u1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --extra-version=0+deb11u1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
[rtsp @ 0x560af91cd140] method DESCRIBE failed: 404 Stream Not Found
rtsp://192.168.100.237:554/ch0_0.h264: Server returned 404 Not Found
```

I suspect something related to streaming is not starting correctly. 

@roleoroleo 

Any help, tips how to debug the issue that may help fixing it for everyone else, or anything will be appreciated.

Thanks,
Jacek

dzek69, 2022-08-30:
More info, result of `top` command:

```
Mem: 59604K used, 1308K free, 136K shrd, 6032K buff, 11148K cached
CPU: 93.5% usr  6.2% sys  0.0% nic  0.0% idle  0.0% io  0.0% irq  0.1% sirq
Load average: 3.60 1.45 0.54 6/118 3464
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
 1457     1 root     R<    4060  6.6   0 68.4 rRTSPServer -m y211ga -r low -a no -p 554
 1179     1 root     S    40676 66.5   0 26.8 ./rmm
 2752  2584 root     R      996  1.6   0  0.5 top
 2551  1310 root     S      904  1.4   0  0.3 /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p 0.0.0.0:22
 1274     1 root     S     3444  5.6   0  0.1 ./p2p_tnp
  638     1 root     S     1096  1.7   0  0.1 ./dispatch
 1493     1 root     S     5900  9.6   0  0.0 onvif_srvd --conf_file /tmp/onvif_srvd.conf
 1275     1 root     S     3512  5.7   0  0.0 ./oss
 2346  2301 root     S     3508  5.7   0  0.0 imggrabber -m y211ga -r low -w
 1272     1 root     S     2532  4.1   0  0.0 ./mp4record
 1260     1 root     S     1400  2.2   0  0.0 /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
 1458     1 root     S      988  1.6   0  0.0 {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
 3459  1458 root     S      988  1.6   0  0.0 {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
 2301  2300 root     S      984  1.6   0  0.0 {snapshot.sh} /bin/sh snapshot.sh
    1     0 root     S      980  1.6   0  0.0 /sbin/init
 2584  2551 root     S      980  1.6   0  0.0 -ash
 3461  3459 root     S      980  1.6   0  0.0 grep rRTSPServer
 3462  3459 root     R      980  1.6   0  0.0 grep -v grep
 3460  3459 root     R      980  1.6   0  0.0 top -b -n 2 -d 1
  549     1 root     S      980  1.6   0  0.0 -/bin/sh
 3464  3459 root     R      980  1.6   0  0.0 awk {print $8}
 1333     1 root     S      980  1.6   0  0.0 udhcpc -i wlan0 -b -s /backup/tools/default.script -x hostname:yi-bedroom
 1273     1 root     S      884  1.4   0  0.0 ./cloud
 1310     1 root     S      876  1.4   0  0.0 /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p 0.0.0.0:22
 2300  1291 root     S      872  1.4   0  0.0 {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf
 1525     1 root     S      864  1.4   0  0.0 {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/crontabs/
 1291     1 root     S      860  1.4   0  0.0 {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf
 1426     1 root     S      860  1.4   0  0.0 {ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/ntpd -p srv.lan
 3463  3459 root     S      860  1.4   0  0.0 {tail} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/bin/tail -n 1
 1277     1 root     S      676  1.1   0  0.0 ./watch_process
 1312     1 root     S      668  1.0   0  0.0 ipc_multiplexer
  266     2 root     SW       0  0.0   0  0.0 [kworker/0:1]
   12     2 root     SW       0  0.0   0  0.0 [kworker/u2:1]
  229     2 root     SW       0  0.0   0  0.0 [kworker/u2:2]
    7     2 root     SW       0  0.0   0  0.0 [rcu_preempt]
  449     2 root     SW       0  0.0   0  0.0 [irq/302-sunxi-m]
    3     2 root     SW       0  0.0   0  0.0 [ksoftirqd/0]
    6     2 root     SW       0  0.0   0  0.0 [kworker/u2:0]
  512     2 root     SW       0  0.0   0  0.0 [mmcqd/0]
    4     2 root     SW       0  0.0   0  0.0 [kworker/0:0]
  544     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd4]
  453     2 root     SW       0  0.0   0  0.0 [kworker/0:2]
  451     2 root     SW       0  0.0   0  0.0 [irq/166-sdc0 cd]
```

This doesn't look good:
> Mem: 59604K used, 1308K free

Web UI however reports:
> Free/Total Memory: 39136/60912 KB

dzek69, 2022-08-30:
Kind of solved.

Disabling NTPD fixed the issue. I don't know why NTPD cannot just synchronize the time with local server available on the network - all other cameras use it to sync the time and it's all fine.

Additionally:
After fixing the stream I enabled MQTT and just like NTPD MQTT seems not to connect at all (to the same machine that has NTP server), it's not even trying, but the mqtt service process seems to work.

I will continue to debug things, maybe it will get useful to somebody.

dzek69, 2022-09-02:
If i disable this one then all is ok
[image]

If I enable it then I cannot access RTSP and snapshots.

dzek69, 2022-09-02:
BTW: I can sync time manually via SSH. So NTP server is accessible from the camera

```
dzek@dzek-dell:~$ ssh root@192.168.100.237

BusyBox v1.27.2 () built-in shell (ash)

mount: mounting none on /sys/kernel/debug failed: Resource busy

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

root@yi-bedroom:~# date
Fri Apr 17 20:35:52 GMT 2020
root@yi-bedroom:~# /tmp/sd/yi-hack/usr/sbin/ntpd -q -np 192.168.30.130
ntpd: setting time to 2022-09-02 10:19:49.885870 (offset +74958231.863185s)
root@yi-bedroom:~# date
Fri Sep  2 10:19:52 GMT 2022
root@yi-bedroom:~#
```

roleoroleo, 2022-12-09:
I need a recap of this issue, please.
The cam runs without internet access.
If you enable ntp, rtsp and snapshot don't work.
Is it correct?

dzek69, 2022-12-09:
Correct

roleoroleo, 2022-12-10:
Sorry but I'm not able to replicate this behavior.
I configured the cam with my local ntp server and I added a rule in the firewall to block internet access.
But the cam continues to work properly.

Maybe the buffer is not properly filled. Run the following commands and check if the buffere changes:
```
root@yi-hack-h52ga:~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 ee 71 0f 00  2a 15 01 00 02 e2 0d 00  |.....q..*.......|
00000010  14 70 0e 00 8e 09 84 fb  4d 5e 00 00 00 00 00 00  |.p......M^......|
00000020
root@yi-hack-h52ga:~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 25 b3 0f 00  2a 15 01 00 fa dc 01 00  |....%...*.......|
00000010  d5 29 02 00 b7 12 84 fb  e2 5e 00 00 00 00 00 00  |.).......^......|
00000020
```

dzek69, 2022-12-10:
I enabled NTPD on both my cameras and they both work correctly now 🤷🏻. I did a few restarts to make sure it's not randomly getting stuck, but no luck this time. I didn't touch any config for these few months.

If it ever gets back I will try checking the buffer and I can even give you direct access to the camera if needed. 

Thanks for your work and your replies!

closed 2022-12-10 by dzek69

arnau-esteller, 2023-09-26:
Sorry to bump this old issue, I was fighting with the same issue for hours (if the camera has no internet connection at boot, the RTSP doesn't work).

The reason was having **Disable Cloud** and **Recording without Cloud** off.
Turning both on and having an NTP server available did the trick. Now it's working correctly.

I'm using a **Yi 1080p Home RFUS | 9.0.35* | y291ga**

Thanks @roleoroleo for your work!