roleoroleo/yi-hack-Allwinner-v2 #489: MQTT switching between online and offline constantly

fernandolcardoso, 2022-09-04:
There is some bug in the MQTT service

[image]

When I activate only the basic with MQTT connections and only the Home Assistant, it works normally.

If I enable any of the features below, the bug occurs.
- MQTT Information Advertisement
- MQTT Link Advertise
- MQTT Camera Settings Advertise
- MQTT Telemetry Advertise

The problem occurs in Firmware 0.2.2 of models r35gb, y211ga and y291ga (beta)

fernandolcardoso, 2022-09-04:
I forgot to mention that the Wi-Fi is at 100% on most cameras and I tested it on 3 different network systems (cable, regular Wi-Fi and Mesh Wi-Fi).

I also tested with snapshot off and swap on, getting the same result.

Without configuring, the countless times I formatted, changed the cards and reset the settings.

roleoroleo, 2022-09-06:
Check if you are using the same client id for both services:

[image]

[image]

fernandolcardoso, 2022-09-06:
All cameras are properly identified and in their separate topics.

I'm using the MQTT Explorer app which makes it easy to see all cameras.

Using the default settings, I can get the online status and binary motion sensor.

But if I enable any of the features below, the online and offline situation occurs.

- MQTT Information Advertisement
- MQTT Link Advertise
- MQTT Camera Settings Advertise
- MQTT Telemetry Advertise

[image]

[image]

[image]

[image]

roleoroleo, 2022-09-07:
There are 2 ids in the same cam.
One for the page http://IP:8080/?page=mqtt and another in the page http://IP:8080/?page=mqtt_adv
Are they different?

fernandolcardoso, 2022-09-08:
**/?page=mqtt**

[image]

[image]

**/?page=mqtt_adv**

[image]

roleoroleo, 2022-09-09:
Ok, change the id in mqtt_adv to "yi-850c-adv" and try if it works.

fernandolcardoso, 2022-09-09:
It does not solve...

?page=mqtt_adv

[image]

[image]

No matter what setting I apply, if I enable some of the services below, the problem occurs.
- MQTT Information Advertisement
- MQTT Link Advertise
- MQTT Camera Settings Advertise
- MQTT Telemetry Advertise

[image]

roleoroleo, 2022-09-09:
Try to overwrite mqtt_adv_* scripts with the content of this archive:
[mqtt_adv.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/9535388/mqtt_adv.tar.gz)
Please, make a backup copy of the files before overwriting.

fernandolcardoso, 2022-09-09:
I believe it solved, all services are active and with stability in status.

[image]

fernandolcardoso, 2022-09-09:
Since my last report, there have been no availability changes and all information is being presented correctly in Home Assistant.

I believe the problem has been resolved.

roleoroleo, 2022-09-10:
Yes, I will apply it in the next build.
The problem happens when you set hostname and mqtt client id at the same value.

fernandolcardoso, 2022-09-18:
I inform you that after updating to version 0.2.3, the problem remains

roleoroleo, 2022-09-20:
I know, I prepared 0.2.3 before these changes.
Please use the scripts I posted above until the next release.

closed 2023-01-18 by stale[bot]