roleoroleo/yi-hack-Allwinner-v2 #503: Live RTSP stream and Snapshot webview not working on Yi Dome Guard YRS r35gb

vielfarbig, 2022-09-20:
Hello,

one of my Yi Dome Guard YRS r35gb is successfully flashed and it also saves motion event videos, which I can view. Also there are recorded snapshot in subfolders of `/tmp/sd/yi-hack/www/record`.
But the live RTSP stream (e.g. trying to view with vlc or mpv) and Snapshot webview is not working.
Eventually it is related to no space available:
```
mount: mounting none on /sys/kernel/debug failed: Resource busy
```
and
```
# df
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1280      1280         0 100% /
devtmpfs                 30400      1824     28576   6% /dev
/dev/mtdblock3            2304      2304         0 100% /home
/dev/mtdblock4            1216      1176        40  97% /backup
tmpfs                    32768       132     32636   0% /tmp
/dev/mmcblk0p1        62351456    155424  62196032   0% /home/app/script/wifidhcp.sh
tmpfs                    32768       132     32636   0% /etc
/dev/mmcblk0p1        62351456    155424  62196032   0% /home/app/cloudAPI
/dev/mmcblk0p1        62351456    155424  62196032   0% /tmp/sd
/dev/mmcblk0p1        62351456    155424  62196032   0% /tmp/sd/yi-hack/www/record
```
The following process are running:
```
# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  207 root         0 SW   [oom_reaper]
  208 root         0 SW<  [writeback]
  210 root         0 SW   [kcompactd0]
  211 root         0 SW<  [crypto]
  212 root         0 SW<  [bioset]
  214 root         0 SW<  [kblockd]
  255 root         0 SW   [sys_user]
  262 root         0 SW   [kworker/0:1]
  263 root         0 SW<  [cfg80211]
  269 root         0 SW<  [watchdogd]
  284 root         0 SW<  [spi0]
  294 root         0 SW   [kswapd0]
  378 root         0 SW<  [bioset]
  383 root         0 SW<  [bioset]
  388 root         0 SW<  [bioset]
  393 root         0 SW<  [bioset]
  398 root         0 SW<  [bioset]
  403 root         0 SW<  [bioset]
  408 root         0 SW<  [bioset]
  413 root         0 SW<  [bioset]
  418 root         0 SW<  [bioset]
  446 root         0 SW   [irq/302-sunxi-m]
  448 root         0 SW   [irq/166-sdc0 cd]
  512 root         0 SW   [kworker/0:2]
  518 root         0 SW<  [kworker/0:1H]
  519 root         0 SW   [kworker/0:3]
  520 root         0 SW<  [bioset]
  521 root         0 SW   [kworker/u2:2]
  523 root         0 SW   [mmcqd/0]
  539 root         0 SWN  [jffs2_gcd_mtd4]
  546 root       980 S    -/bin/sh
  603 root      1152 S    ./dispatch
 1153 root     42016 S    ./rmm
 1220 root         0 SW   [RTW_CMD_THREAD]
 1236 root      1468 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
 1559 root      1048 S    udhcpc -i eth0 -b -s /home/app/script/default.script
 1586 root      2756 S    ./mp4record
 1598 root       860 S    {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf
 1616 root       868 S    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p 0.0.0.0:22
 1647 root       872 S    {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf
 1648 root       984 S    {snapshot.sh} /bin/sh snapshot.sh
 1688 root      3508 S    imggrabber -m r35gb -r low
 1689 root       860 S    {base64} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/bin/base64
 1723 root       996 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
 1770 root       864 S    {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/crontabs/
 2101 root       896 R    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p 0.0.0.0:22 -2
 2114 root       980 S    -ash
 3495 root      4060 S<   rRTSPServer -m r35gb -r high -p 554
 3501 root       980 S    sleep 10
 3536 root       980 R    ps
```

The system conf is
```
# cat /tmp/sd/yi-hack/etc/system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=no
RTSP=yes
RTSP_ALT=no
RTSP_STREAM=high
RTSP_AUDIO=no
SPEAKER_AUDIO=yes
SNAPSHOT=yes
ONVIF=no
ONVIF_WSDD=no
ONVIF_PROFILE=high
ONVIF_NETIF=wlan0
ONVIF_WM_SNAPSHOT=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
TIMEZONE=
FREE_SPACE=5
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=yes
SSH_PASSWORD=
CRONTAB=
SNAPSHOT_VIDEO=yes
SNAPSHOT_LOW=no
```
and newest version:
```
# cat /tmp/sd/yi-hack/version 
0.2.3
```
model:
```
# cat /tmp/sd/yi-hack/model_suffix 
r35gb
```

Btw: I wanted to try the unbrick method (https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick) but how do I get the mtdblocks?

Thanks for your help!

elbuit, 2022-11-08:
I have same camera and same behaviour, not snapshot or rstp although there is a /record/folder with recordings.
I changed firmware without using YI home app.
I've followed the this method.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Install-the-hack-without-the-Yi-app

In the PTZ menu I can move the camera but there isn't any image/screeshot.

Have you used Yi app to change the firmware?

vielfarbig, 2022-11-09:
I also changed the firmware with the method in the wiki without the Yi app.
I also can move the camera in PTZ menu, but also no image/screenshots are visible.

There are also other issues having problems with these, but I didn't something I could try yet.

roleoroleo, 2022-11-09:
> mount: mounting none on /sys/kernel/debug failed: Resource busy

This is normal.

> Btw: I wanted to try the unbrick method (https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick) but how do I get the mtdblocks?

The SD card should contain a folder with the backup made when you applied the hack.

If snapshots and stream are not working, maybe there is a problem reading the frame buffer.
Please send me a copy of the file /dev/shm/fshare_frame_buf
And check the yi firmware version.

roleoroleo, 2022-11-14:
Simply a cp and a ftp transfer:
```
cp /dev/shm/fshare_frame_buf /tmp
# ftp transfer with filezilla or similar
```
I need the whole file to check the structure.

About version:
`# cat /home/homever`

vielfarbig, 2022-11-14:
Here is the fshare_frame_buf
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10002646/fshare_frame_buf.zip)

for

```
# cat /home/homever 
9.0.05.33_202009251856
```

vielfarbig, 2022-11-14:
Yes it reads r30gb:

```
# cat /backup/init.sh | grep SUFFIX
SUFFIX=r30gb
if [ -f /home/home_${SUFFIX}m ]; then
    echo "---/home/home_${SUFFIX}m exist, update begin---"
	dd if=/home/home_${SUFFIX}m of=/tmp/newver bs=22 count=1
		rm -rf /home/home_${SUFFIX}m
	rm -rf mv /home/home_${SUFFIX}m
elif [ -f /tmp/sd/home_${SUFFIX}m ]; then
	echo "---tmp/sd/home_${SUFFIX}m exist, update begin---"
	dd if=/tmp/sd/home_${SUFFIX}m of=/tmp/newver bs=22 count=1
        cp -rf /tmp/sd/home_${SUFFIX}m /tmp/update/
		/tmp/update/extpkg.sh /tmp/sd/home_${SUFFIX}m
		mv /tmp/sd/home_${SUFFIX}m	/tmp/sd/home_${SUFFIX}m.done
	mv /tmp/sd/home_${SUFFIX}m	/tmp/sd/home_${SUFFIX}m.done
	echo "---update file(home_${SUFFIX}m) Not exist---"
```

but the model on the back of the camera reads `Yi Dome Guard YRS 3019` so according to the README it should be r35gb

vielfarbig, 2022-11-14:
@roleoroleo Can I simply apply the same installation steps (yi-hack r35gb -> yi-hack r30gb as @elbuit did it with the App) as for the original process (for yi firmware -> yi-hack) without the Yi App? Or do I have to reset something?
I think I cannot do it with the WebGUI and/or local firmware upgrading for switching.

closed 2023-03-14 by stale[bot]