roleoroleo/yi-hack-Allwinner-v2 #504: Can't remove YI watermark or change cameras IP

refriedfood, 2022-09-24:
I'm using YI 1080p Home Camera (y291ga_0.2.3), is there any way to specify this? Literally the only reason to hack this stupid camera was so that I could specify its IP address.
Any help appreciated.

roleoroleo, 2022-09-27:
There are other issues related to static IP.
But I didn't find how to set it because the yi processes check the connection and restart the dhcp request.
Please consider to add a reservation to you router.

refriedfood, 2022-09-28:
The problem is there is no way to assign backwards within most routers (like mine). You can turn off auto dhcp, like I have done, and specify each connection manually via ip and dns, but you cannot use the webUI interface to specify it on the camera, regardless if I have the ip available--which I do. Without dynamic assignment it will not connect, and with it on, it will take whatever isn't already assigned. Is there no way to force manual ip assignment in your hack? Couldn't its auto dchp search be turned off and allow for manual assignment instead?

There's also a problem that I have noticed, that every time the camera is powered off, even if the sd card is left in, and then powered back on, the camera is no longer cracked. I have to unplug, take the sd card out, copy the hack files to the sd card again, and then put the sd card back and power on it order for it to work again. It's a bit of a hassle that the crack doesn't stay over power downs. In fact, I see that as more of a pain then the ip it randomly grabs.

roleoroleo, 2022-09-30:
> There's also a problem that I have noticed, that every time the camera is powered off, even if the sd card is left in, and then powered back on, the camera is no longer cracked. I have to unplug, [...]

This behavior is not normal. If you don't remove the sd card, the hack must work.
Please check if the sd card is good, or try another one.

About static IP check this post:
https://github.com/roleoroleo/yi-hack-MStar/issues/208
but I think that there is no solution.

refriedfood, 2022-09-30:
It was a brand new card, and I tested it in a nintendo DS just to make sure, and it works fine after testing it. I'm not sure it would crack it to begin with it something was wrong with the card. The file structure changes on the card after its used to factory.done. Other than that there is no change. I will look at that link, but if you can't do it than you're probably right.

closed 2023-01-28 by stale[bot]

tempod, 2024-10-09:
I would also like to delete the dns that I am not interested in.
I created a file called startup.sh and put it in /tmp/sd/yi-hack/
inside this file I wrote
`sed -e '2,5d' /tmp/resolv.conf`
I restart the camera but when I go to check the file all the DNS are still present.
What am I doing wrong?