roleoroleo/yi-hack-Allwinner-v2 #507: Can't downgrade from 11.1.35

mousemn, 2022-09-30:
I have YI Home 1080 Camera RFUS. It comed with 9.0.36. First I made backup and upgraded it to 11.1.35.3_202207191613.
Now trying to soft downgrade to backuped version 9.0.36 and camera does not any reaction to SD. Yi application working well/
The other same camera (QFUS) was updated to 11.0.36 and succesfully downgraded to 9.0.36

roleoroleo, 2022-09-30:
Maybe 11.1 adds some downgrade protection.
Are you able to make a backup copy?

mousemn, 2022-09-30:
Yes, I made backup 11.1

roleoroleo, 2022-09-30:
Share it with me please.
I will check it.

mousemn, 2022-10-10:
https://disk.yandex.ru/d/hAGHzraWQOBfUg

Here is backup of RFUS_11.1.35.3_202207191613
Sorry for my late answer

roleoroleo, 2022-10-10:
This backup comes from a y291ga and not from a y211ga
So the old software had to be 9.0.35

mousemn, 2022-10-10:
yes, I was wrong, the original version was 9.0.35. 
I got confused with another YI Home 1080 Camera camera

mousemn, 2022-10-10:
Can I download a backup for y291ga RFUS 9.0.35 somewhere?

mousemn, 2022-10-10:
I upload y291ga_0.2.3.tgz to SD and looks like it was succsesfully installed to 11.1.35.

mousemn, 2022-10-10:
[image]

roleoroleo, 2022-10-10:
Downgrade is recommended for y211ga.
Probably, in your case it's not necessary.
Check if RTSP and snapshot are working properly.

evider2002, 2022-11-18:
Hi, 
I am currently in the exact same situation as mousemn. 
Firmware is 11.1.35.3_202207191613. 
y291ga_0.2.3.tgz is uploading and working properly. 
The only issue I am having after hacking, is that camera is not uploading footage to yi cloud anymore, and no app alerts are working either. 
How do I get the hack and yi cloud/alerts to work? 

Thank you.

roleoroleo, 2022-11-18:
Is "Disable cloud" on?

evider2002, 2022-11-18:
Thank you for your response roleoroleo! 
No, I did not turn on "Disable cloud". 
Right after freshly hacking (without changing any settings) motion does not work. 
Both cloud storage and alerts work from motion, so it seems like motion is what is not working. 
But the truth is that after I posted this question, I see that you addressed it here - 
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/388 . 
The problem is that 1- downgrading to 9.0 seems to be beyond my abilities, and 2- I want to stay on 11.1 because it has more motion alerts - person/vehicle/animal, whereas 9.0 only has person. 
So the question is, is there anyway to get yi-hack-Allwinner-v2 working on 11.1? 

Thank you!

evider2002, 2022-11-18:
Yes, if I activate specific alert, then I only get those specific alerts and won't get generic motion alert.
I can choose either specific alerts or generic motion alert. 
These alerts are the 6 second clips, but is also responsible for triggering motion cloud backup, which is not just 6 seconds, rather the complete motion. 
So if I select generic motion alert, then I will get generic motion 6 second alerts, and complete motion cloud backup. 
If I select specific motion alerts then I will get specific motion 6 second alerts, and complete specific motion cloud backup. 

When I install the hack, none of this works whether I choose specific or generic. Not the 6 second motion alerts, nor the complete motion cloud backup.

roleoroleo, 2022-11-18:
Are you able to connect with the cam via app?
Are all function working properly (except motion videos)?

evider2002, 2022-11-18:
Without the hack motion is working on 11, and worked on 9. 
With the hack I only know that its not working on 11, because I never tried the hack when it was on 9.

roleoroleo, 2022-11-18:
Please, try to remove the SD card and check if the motion works again.

evider2002, 2022-11-18:
It does work when I remove the sd, and when I replace the sd again without the hack. It only doesnt work with the sd and hack.

roleoroleo, 2022-11-19:
Never seen this problem.
Please, login to the cam with ssh and check the running processes.
`# ps`
Maybe something crashes...

evider2002, 2022-11-21:
roleoroleo, I am a bit confused. 
You are saying that you have never seen this this problem.
But isn't this https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/388 exactly my problem - that motion doesn't work in firmware version 11?

roleoroleo, 2022-11-21:
It's a different cam.
And, if I remeber correctly, motion detection was completely silent, even without hack.

ludoo, 2022-11-23:
Same issue here, RFUSY62 y291ga on firmware version 11, everything works great apart from motion detection, which works without the hack.

roleoroleo, 2022-11-23:
Please, explain me better.

1 - Motion detection is working when the hack is not running. You receive notifications from the yi app.
2 - Motion detection is not working when the hack is running. But...
- Is not working with the app (no notifications) or is not working with mqtt?
- Is the "Disable cloud" on or off?
- Is it working with the old 9.x.x firmware?
- y211ga, y291ga or another model?
- What's your configuration about other detections? Sound enable/disable, baby crying enable/disable, human detection enable/disable, face detection enable/disable?

ludoo, 2022-11-23:
@roleoroleo thanks for the quick reply, and forgive my noobness I just received my first camera.

> 1 - Motion detection is working when the hack is not running. You receive notifications from the yi app.

yes, I don't have notifications set up but in the camera timeline in the app I see snippets saved by motion or sound detection; this both before installing the hack, and after installation if I reformat the SD and restart

> 2 - Motion detection is not working when the hack is running. But...

yes, with the hack running there's no motion detection active; the snippets list is empty

- MQTT disabled
- disable cloud flag off (cloud enabled)
- smart detection has "motion detection" enabled (not Smart AI), sound enabled

Model is y291ga, or at least that's what I installed based on the serial number RFUSY62 (not clear if I should use y211ga or y291ga).

roleoroleo, 2022-11-25:
Please, check running processes.
`# ps`

You should see a list like this:
```
root@yi-hack-h52ga:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  211 root         0 SW   [oom_reaper]
  212 root         0 SW<  [writeback]
  214 root         0 SW   [kcompactd0]
  215 root         0 SW<  [crypto]
  216 root         0 SW<  [bioset]
  218 root         0 SW<  [kblockd]
  259 root         0 SW   [sys_user]
  266 root         0 SW   [kworker/0:1]
  267 root         0 SW<  [cfg80211]
  274 root         0 SW<  [watchdogd]
  288 root         0 SW<  [spi0]
  298 root         0 SW   [kswapd0]
  382 root         0 SW<  [bioset]
  387 root         0 SW<  [bioset]
  392 root         0 SW<  [bioset]
  397 root         0 SW<  [bioset]
  402 root         0 SW<  [bioset]
  407 root         0 SW<  [bioset]
  412 root         0 SW<  [bioset]
  417 root         0 SW<  [bioset]
  422 root         0 SW<  [bioset]
  454 root         0 SW   [irq/302-sunxi-m]
  456 root         0 SW   [irq/166-sdc0 cd]
  457 root         0 SW   [irq/303-sunxi-m]
  460 root         0 SW   [kworker/0:2]
  514 root         0 SW<  [bioset]
  515 root         0 SW   [mmcqd/0]
  534 root         0 SW<  [kworker/0:1H]
  547 root         0 SWN  [jffs2_gcd_mtd4]
  552 root       980 S    -/bin/sh
  585 root         0 SW   [kworker/0:3]
  593 root         0 SW   [ksdioirqd/mmc1]
  598 root         0 SW   [RTW_XMIT_THREAD]
  599 root         0 SW   [RTW_CMD_THREAD]
  600 root         0 SW   [RTWHALXT]
  609 root      1164 S    ./dispatch
 1166 root     39304 S    ./rmm
 1168 root         0 SW   [kworker/u2:2]
 1244 root      1488 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
 1256 root      1948 S    ./mp4record
 1257 root      1152 S    ./cloud
 1258 root      3128 S    ./p2p_tnp
 1259 root      3188 S    ./oss
 1261 root       676 S    ./watch_process
 1275 root       860 S    {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf
 1287 root       844 S    pure-ftpd (SERVER)
 1299 root       868 S    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p 0.0.0.0:22
 1328 root      1048 S    udhcpc -i wlan0 -b -s /backup/tools/default.script -x hostname:yi-hack-h52ga
 1423 root       860 S    {ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/ntpd -p ntp1.inrim.it
 1451 root       996 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
 1504 root      5900 S    onvif_srvd --conf_file /tmp/onvif_srvd.conf
 1524 root      1988 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --type tdn:NetworkVideoTransmitter --xaddr http://%s --scope onvif://www.onvif.org/name/Unknown onvif://www.onvif.org/Profile/Str
 1552 root       864 S    {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/crontabs/
 1817 root      3636 S<   rRTSPServer -m h52ga -r both -a alaw -p 554
 3016 root       892 R    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p 0.0.0.0:22 -2
 3038 root       980 S    -ash
 3163 root       980 S    sleep 10
 3198 root       980 S    /bin/sh -c /tmp/sd/yi-hack/script/thumb.sh cron
 3199 root      1100 S    {thumb.sh} /bin/ash /tmp/sd/yi-hack/script/thumb.sh cron
 3221 root      1164 S    {thumb.sh} /bin/ash /tmp/sd/yi-hack/script/thumb.sh cron
 3355 root       980 R    ps
```

ludoo, 2022-11-25:
Here it is

<details>
<summary>ps</summary>
```
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  211 root         0 SW   [oom_reaper]
  212 root         0 SW<  [writeback]
  214 root         0 SW   [kcompactd0]
  215 root         0 SW<  [crypto]
  216 root         0 SW<  [bioset]
  218 root         0 SW<  [kblockd]
  259 root         0 SW   [sys_user]
  266 root         0 SW   [kworker/0:1]
  267 root         0 SW<  [cfg80211]
  273 root         0 SW<  [watchdogd]
  288 root         0 SW<  [spi0]
  298 root         0 SW   [kswapd0]
  382 root         0 SW<  [bioset]
  387 root         0 SW<  [bioset]
  392 root         0 SW<  [bioset]
  397 root         0 SW<  [bioset]
  402 root         0 SW<  [bioset]
  407 root         0 SW<  [bioset]
  412 root         0 SW<  [bioset]
  417 root         0 SW<  [bioset]
  422 root         0 SW<  [bioset]
  450 root         0 SW   [irq/302-sunxi-m]
  452 root         0 SW   [irq/166-sdc0 cd]
  453 root         0 SW   [irq/303-sunxi-m]
  454 root         0 SW   [kworker/0:2]
  458 root         0 SW   [kworker/0:3]
  525 root         0 SW<  [kworker/0:1H]
  526 root         0 SW   [kworker/0:4]
  527 root         0 SW<  [bioset]
  528 root         0 SW   [kworker/u2:2]
  530 root         0 SW   [mmcqd/0]
  546 root         0 SWN  [jffs2_gcd_mtd4]
  584 root       988 S    /usr/sbin/telnetd
  596 root         0 SW<  [phy0-atbm_wq]
  597 root         0 SW   [phy0-usb_atbm_b]
  642 root      1264 S    ./dispatch
 1197 root     41352 S    ./rmm
 1277 root      1488 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
 1283 root      2808 S    ./mp4record
 1284 root      1224 S    ./cloud
 1285 root      3556 S    ./p2p_tnp
 1286 root      3880 S    ./oss
 1288 root       744 S    ./watch_process
 1304 root       860 S    {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf
 1315 root       844 S    pure-ftpd (SERVER)
 1335 root       868 S    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p 0.0.0.0:22
 1369 root      1048 S    udhcpc -i wlan0 -b -s /home/app/script/default.script
 1459 root       860 S    {ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/ntpd -p pool.ntp.org
 1490 root       996 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
 1552 root      5900 S    onvif_srvd --conf_file /tmp/onvif_srvd.conf
 1579 root      1988 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --type tdn:NetworkVideoTransmitter --xaddr http://%s --scope onvif://www.onvif.org/name/Unknown onvif://www.o
 1596 root       864 S    {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/crontabs/
 1847 root      4060 S<   rRTSPServer -m y291ga -r high -a yes -p 554
 1926 root       980 S    /bin/sh -c /tmp/sd/yi-hack/script/thumb.sh cron
 1927 root      1032 S    {thumb.sh} /bin/ash /tmp/sd/yi-hack/script/thumb.sh cron
 1950 root      1032 S    {thumb.sh} /bin/ash /tmp/sd/yi-hack/script/thumb.sh cron
 2485 root       980 S    sleep 10
 2508 root       980 S    -ash
 2543 root      2700 R    imggrabber -f /tmp/sd/record/2022Y11M23D15H/E148M00S60.h26x -r low -w
 2544 root         0 Z    [init]
 2545 root       980 R    ps
```
</details>

ludoo, 2022-11-25:
What is the process I should be seeing? I can log in and try to troubleshoot if you point me in the right direction.

roleoroleo, 2022-11-25:
The processes needed for detection to work are: rmm, cloud and p2p_tnp.
Nothing strange in your ps.
Please, post your /tmp/sd/lower_half_init.sh

ludoo, 2022-11-25:
here it is

```bash
#!/bin/sh
mount -t vfat /dev/mmcblk0 /tmp/sd
if [ "${SUFFIX}" = "y211ga" ] || [ "${SUFFIX}" = "y211ba" ] || [ "${SUFFIX}" = "y291ga" ];then
    echo "need reset gpio198"
    echo 198 > /sys/class/gpio/export 
    echo out > /sys/class/gpio/gpio198/direction  
    echo 0 > /sys/class/gpio/gpio198/value 
    sleep 1
    echo 1 > /sys/class/gpio/gpio198/value
fi

### wifi 8188 ###
if [ "${enable_4g}" = "y" ];then
    echo "4g is running...."
else
    if [ -f /home/base/wifi/8188fu.ko ];then
                insmod /home/base/wifi/8188fu.ko
    elif [ -f /home/base/wifi/8189fs.ko ];then
        insmod /home/base/wifi/8189fs.ko
    elif [ -f /backup/ko/8188fu.ko ];then
                insmod /backup/ko/8188fu.ko
    elif [ -f /backup/ko/8189fs.ko ];then
        insmod /backup/ko/8189fs.ko
    elif [ -f /backup/ko/hi3881.ko ];then
        insmod /backup/ko/hi3881.ko
    elif [ -f /backup/ko/8192fu.ko ];then
                insmod /backup/ko/8192fu.ko         
    elif [ -f /backup/ko/atbm603x_wifi_usb.ko ];then
        insmod /backup/ko/atbm603x_wifi_usb.ko
    elif [ -f /backup/ko/rdawfmac.ko ];then
                        insmod /backup/ko/rdawfmac.ko    
    fi
        
    if [ -f /backup/ko/ssv6x5x.ko ];then
        if [ "${SUFFIX}" = "d071qp"  ];then
            if [ -f /home/base/firmware/ssv6x5x/ssv6152-wifi.cfg ];then
                insmod /backup/ko/ssv6x5x.ko stacfgpath="/home/base/firmware/ssv6x5x/ssv6152-wifi.cfg" wifi_type=SDIO
            else
                echo "not found ssv6x5x-wifi.cfg"
            fi  
        else
            if [ -f /home/base/firmware/ssv6x5x/ssv6x5x-wifi.cfg ];then
                insmod /backup/ko/ssv6x5x.ko stacfgpath="/home/base/firmware/ssv6x5x/ssv6x5x-wifi.cfg" wifi_type=$SSV_WIFI_TYPE
            else
                echo "not found ssv6x5x-wifi.cfg"
            fi  
        fi      
    fi
fi

echo "--------------------------insmod sensor--------------------------"
insmod /home/base/ko/videobuf2-core.ko
insmod /home/base/ko/videobuf2-memops.ko
insmod /home/base/ko/videobuf2-dma-contig.ko
insmod /home/base/ko/videobuf2-v4l2.ko
insmod /home/base/ko/vin_io.ko

#统一版本
if [ "${SUFFIX}" = "b091qp" ];then
insmod /backup/ko/cam_sensor.ko
insmod /home/base/ko/vin_v4l2.ko ccm0=$SENSOR_DRIVE_NAME i2c0_addr=$SENSOR_ADDR
else
insmod /home/base/ko/cam_sensor.ko
insmod /home/base/ko/vin_v4l2.ko
fi

if [ -f /home/base/ko/icplus.ko ];then
    insmod /home/base/ko/icplus.ko
elif [ -f /backup/ko/icplus.ko ];then
    insmod /backup/ko/icplus.ko
fi

if [ -f /home/base/ko/sunxi_gpadc.ko ];then
    insmod /home/base/ko/sunxi_gpadc.ko
fi

#/home/app/script/factory_test.sh

#echo "MTK 7601" > /tmp/MTK
#echo /tmp/sd/core.exe[%e].pid[%p].sig[%s] > /proc/sys/kernel/core_pattern

sleep 1
ifconfig lo up

ifconfig ${NETWORK_IFACE} up
#设置最大功率 wlan0 up之后
#if [ -f /backup/ko/rdawfmac.ko ];then
#    /backup/ko/rda5995-usb/firmware/rda_tools wlan0  write_txp 0x4e 0x3a
#fi

ethmac=d2:`ifconfig ${NETWORK_IFACE} |grep HWaddr|cut -d' ' -f10|cut -d: -f2-`
#if [ "${enable_4g}" = "y" ];then
#    ifconfig usb0 up
#    ethmac=d2:`ifconfig usb0 |grep HWaddr|cut -d' ' -f10|cut -d: -f2-`
#else
#    ifconfig wlan0 up
#    ethmac=d2:`ifconfig wlan0 |grep HWaddr|cut -d' ' -f10|cut -d: -f2-`
#fi

ifconfig eth0 hw ether $ethmac
a=1
if [ "${SUFFIX}" = "b111qp" ] || [ "${SUFFIX}" = "b101qp" ] || [ "${SUFFIX}" = "b092qp" ] || [ "${SUFFIX}" = "b091qp" ] || [ "${SUFFIX}" = "q321br_aldz_3m" ]; then
    while ( ! ifconfig eth0 up)
    do
        echo "ifconfig eth0 up failed"
        let a++
        if [ $a -eq 10 ]; then
            break
        fi
    done
else
    ifconfig eth0 up
fi

echo "============================================= home low_half_init.sh... ========================================="
echo "============================================= begin to start app... ========================================="
cd /home/app
if [ -f /home/app/property ];then
    ./property &
fi
#./log_server &

if [ -f "/tmp/sd/Factory/factory_test.sh" ]; then
        /tmp/sd/Factory/config.sh
        exit
fi

if [ -f "/tmp/sd/factory_aging_test.sh" ]; then
         #/tmp/sd/factory_aging_test.sh
    ./dispatch &
    sleep 2
    ./rmm &
    sleep 2
    ./mp4record &
        exit
fi

LD_PRELOAD=/tmp/sd/yi-hack/lib/ipc_multiplex.so ./dispatch &
#sleep 2
#./rmm &
#sleep 2
#./mp4record &
#./cloud &
#./p2p_tnp &
#./oss &
#./rtmp &
#./watch_process &

chmod 777 /tmp/sd/debug.sh
sh /tmp/sd/debug.sh &

chmod 755 /tmp/sd/yi-hack/script/system.sh
sh /tmp/sd/yi-hack/script/system.sh &
```

ludoo, 2022-11-25:
I downgraded to 9.x using the backup you provided in #535 and now motion detection works as expected. I will be getting another camera next week, if you want to try and nail down the issue with the new firmware let me know and I'll be glad to test stuff.

roleoroleo, 2022-11-26:
Ok.

evider2002, 2022-11-29:
What is the user name and password to login to the cam with ssh?

roleoroleo, 2022-11-29:
user: root
password: <blank>

evider2002, 2022-11-29:
That worked! Thank you!

evider2002, 2022-11-29:
root@yi-f9ed:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  211 root         0 SW   [oom_reaper]
  212 root         0 SW<  [writeback]
  214 root         0 SW   [kcompactd0]
  215 root         0 SW<  [crypto]
  216 root         0 SW<  [bioset]
  218 root         0 SW<  [kblockd]
  259 root         0 SW   [sys_user]
  266 root         0 SW   [kworker/0:1]
  267 root         0 SW<  [cfg80211]
  273 root         0 SW<  [watchdogd]
  288 root         0 SW<  [spi0]
  298 root         0 SW   [kswapd0]
  382 root         0 SW<  [bioset]
  387 root         0 SW<  [bioset]
  392 root         0 SW<  [bioset]
  397 root         0 SW<  [bioset]
  402 root         0 SW<  [bioset]
  407 root         0 SW<  [bioset]
  412 root         0 SW<  [bioset]
  417 root         0 SW<  [bioset]
  422 root         0 SW<  [bioset]
  450 root         0 SW   [irq/302-sunxi-m]
  452 root         0 SW   [irq/166-sdc0 cd]
  453 root         0 SW   [irq/303-sunxi-m]
  511 root         0 SW<  [bioset]
  512 root         0 SW   [mmcqd/0]
  533 root         0 SW<  [kworker/0:1H]
  546 root         0 SWN  [jffs2_gcd_mtd4]
  586 root       980 S    /usr/sbin/telnetd
  598 root         0 SW<  [phy0-atbm_wq]
  599 root         0 SW   [phy0-usb_atbm_b]
  645 root      1264 S    ./dispatch
 1210 root     41308 S    ./rmm
 1293 root      1468 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
 1299 root      3076 S    ./mp4record
 1300 root      1224 S    ./cloud
 1301 root      3556 S    ./p2p_tnp
 1302 root      5000 S    ./oss
 1304 root       744 S    ./watch_process
 1319 root       860 S    {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf
 1332 root       844 S    pure-ftpd (SERVER)
 1343 root       868 S    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p 0.0.0.0:22
 1381 root      1048 S    udhcpc -i wlan0 -b -s /home/app/script/default.script
 1384 root       892 S    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p 0.0.0.0:22 -2
 1385 root       980 S    ash -c /tmp/sd/yi-hack/usr/libexec/sftp-server
 1386 root       776 S    /tmp/sd/yi-hack/usr/libexec/sftp-server
 1493 root       860 S    {ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/ntpd -p pool.ntp.org
 1520 root       996 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
 1565 root      5900 S    onvif_srvd --conf_file /tmp/onvif_srvd.conf
 1588 root      1988 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --type tdn:NetworkVideoTransmitter --xaddr http://%s --scope onvif://www.onvif.org/name/Unknown onvif://www.onv
 1612 root       864 S    {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/crontabs/
 1770 root      4060 S<   rRTSPServer -m y291ga -r high -p 554 -u pi -w raspberry
20179 root         0 SW   [kworker/u2:2]
24230 root       916 S    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sbin/dropbear -R -B -p 0.0.0.0:22 -2
25147 root       984 S    -ash
28821 root       980 S    sleep 10
28840 root         0 Z    [init]
28843 root       980 R    ps

evider2002, 2022-11-29:
```
#!/bin/sh
mount -t vfat /dev/mmcblk0 /tmp/sd
if [ "${SUFFIX}" = "y211ga" ] || [ "${SUFFIX}" = "y211ba" ] || [ "${SUFFIX}" = "y291ga" ];then
    echo "need reset gpio198"
    echo 198 > /sys/class/gpio/export 
    echo out > /sys/class/gpio/gpio198/direction  
    echo 0 > /sys/class/gpio/gpio198/value 
    sleep 1
    echo 1 > /sys/class/gpio/gpio198/value
fi

### wifi 8188 ###
if [ "${enable_4g}" = "y" ];then
    echo "4g is running...."
else
    if [ -f /home/base/wifi/8188fu.ko ];then
		insmod /home/base/wifi/8188fu.ko
    elif [ -f /home/base/wifi/8189fs.ko ];then
        insmod /home/base/wifi/8189fs.ko
    elif [ -f /backup/ko/8188fu.ko ];then
		insmod /backup/ko/8188fu.ko
    elif [ -f /backup/ko/8189fs.ko ];then
        insmod /backup/ko/8189fs.ko
    elif [ -f /backup/ko/hi3881.ko ];then
        insmod /backup/ko/hi3881.ko
    elif [ -f /backup/ko/8192fu.ko ];then
		insmod /backup/ko/8192fu.ko	    
    elif [ -f /backup/ko/atbm603x_wifi_usb.ko ];then
        insmod /backup/ko/atbm603x_wifi_usb.ko
    elif [ -f /backup/ko/rdawfmac.ko ];then
			insmod /backup/ko/rdawfmac.ko    
    fi
	
    if [ -f /backup/ko/ssv6x5x.ko ];then
        if [ "${SUFFIX}" = "d071qp"  ];then
            if [ -f /home/base/firmware/ssv6x5x/ssv6152-wifi.cfg ];then
                insmod /backup/ko/ssv6x5x.ko stacfgpath="/home/base/firmware/ssv6x5x/ssv6152-wifi.cfg" wifi_type=SDIO
            else
                echo "not found ssv6x5x-wifi.cfg"
            fi	
        else
            if [ -f /home/base/firmware/ssv6x5x/ssv6x5x-wifi.cfg ];then
                insmod /backup/ko/ssv6x5x.ko stacfgpath="/home/base/firmware/ssv6x5x/ssv6x5x-wifi.cfg" wifi_type=$SSV_WIFI_TYPE
            else
                echo "not found ssv6x5x-wifi.cfg"
            fi	
        fi	
    fi
fi

echo "--------------------------insmod sensor--------------------------"
insmod /home/base/ko/videobuf2-core.ko
insmod /home/base/ko/videobuf2-memops.ko
insmod /home/base/ko/videobuf2-dma-contig.ko
insmod /home/base/ko/videobuf2-v4l2.ko
insmod /home/base/ko/vin_io.ko

#统一版本
if [ "${SUFFIX}" = "b091qp" ];then
insmod /backup/ko/cam_sensor.ko
insmod /home/base/ko/vin_v4l2.ko ccm0=$SENSOR_DRIVE_NAME i2c0_addr=$SENSOR_ADDR
else
insmod /home/base/ko/cam_sensor.ko
insmod /home/base/ko/vin_v4l2.ko
fi

if [ -f /home/base/ko/icplus.ko ];then
    insmod /home/base/ko/icplus.ko
elif [ -f /backup/ko/icplus.ko ];then
    insmod /backup/ko/icplus.ko
fi

if [ -f /home/base/ko/sunxi_gpadc.ko ];then
    insmod /home/base/ko/sunxi_gpadc.ko
fi

#/home/app/script/factory_test.sh

#echo "MTK 7601" > /tmp/MTK
#echo /tmp/sd/core.exe[%e].pid[%p].sig[%s] > /proc/sys/kernel/core_pattern

sleep 1
ifconfig lo up

ifconfig ${NETWORK_IFACE} up
#设置最大功率 wlan0 up之后
#if [ -f /backup/ko/rdawfmac.ko ];then
#    /backup/ko/rda5995-usb/firmware/rda_tools wlan0  write_txp 0x4e 0x3a
#fi

ethmac=d2:`ifconfig ${NETWORK_IFACE} |grep HWaddr|cut -d' ' -f10|cut -d: -f2-`
#if [ "${enable_4g}" = "y" ];then
#    ifconfig usb0 up
#    ethmac=d2:`ifconfig usb0 |grep HWaddr|cut -d' ' -f10|cut -d: -f2-`
#else
#    ifconfig wlan0 up
#    ethmac=d2:`ifconfig wlan0 |grep HWaddr|cut -d' ' -f10|cut -d: -f2-`
#fi

ifconfig eth0 hw ether $ethmac
a=1
if [ "${SUFFIX}" = "b111qp" ] || [ "${SUFFIX}" = "b101qp" ] || [ "${SUFFIX}" = "b092qp" ] || [ "${SUFFIX}" = "b091qp" ] || [ "${SUFFIX}" = "q321br_aldz_3m" ]; then
    while ( ! ifconfig eth0 up)
    do
        echo "ifconfig eth0 up failed"
        let a++
        if [ $a -eq 10 ]; then
            break
        fi
    done
else
    ifconfig eth0 up
fi

echo "============================================= home low_half_init.sh... ========================================="
echo "============================================= begin to start app... ========================================="
cd /home/app
if [ -f /home/app/property ];then
    ./property &
fi
#./log_server &

if [ -f "/tmp/sd/Factory/factory_test.sh" ]; then
	/tmp/sd/Factory/config.sh
	exit
fi

if [ -f "/tmp/sd/factory_aging_test.sh" ]; then
	 #/tmp/sd/factory_aging_test.sh
    ./dispatch &
    sleep 2
    ./rmm &
    sleep 2
    ./mp4record &
	exit
fi

LD_PRELOAD=/tmp/sd/yi-hack/lib/ipc_multiplex.so ./dispatch &
#sleep 2
#./rmm &
#sleep 2
#./mp4record &
#./cloud &
#./p2p_tnp &
#./oss &
#./rtmp &
#./watch_process &

chmod 777 /tmp/sd/debug.sh
sh /tmp/sd/debug.sh &

chmod 755 /tmp/sd/yi-hack/script/system.sh
sh /tmp/sd/yi-hack/script/system.sh &
```

evider2002, 2022-11-29:
So I think at this point it is clear that motion will not work with Yi hack on 11.  
Is there any plan to have yi hack support 11 too?  
12 just came out, should upgrade and test on 12?

roleoroleo, 2022-12-08:
I don't know if I'm able to make it work in version 11, because I don't onw this model.
You could try 12.

evider2002, 2022-12-08:
It doesn't work in 12 either as mentioned here https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/535 . 
At this point, I downgraded to 9 and trying to make peace with it. 

If at any point you do add support for 11 / 12, please update this thread so that I will be notified.   

Thank you.

closed 2023-04-07 by stale[bot]