roleoroleo/yi-hack-Allwinner-v2 #508: No sound_detection in MQTT (Yi Home 1080p - IFUS)

IvanUpa, 2022-10-04:
Good afternoon.

Please help me solve the problem. Topics for "sound_detection" and "crying" are not published in MQTT. All other "motion_start", "motion_stop", "human", "motion_file" events are created.
In the YI Home application, these events work fine.
What has already been done:
1) Several times completely reset the camera.
2) Several times completely installed Hack.
3) I tried to install Hack from other cameras of similar models (not successful).
4) All sorts of combinations of settings via app and web.
5) I re-read all the created issues, I did not find a solution.

Made noise around the camera, but there is only motion. The application has both motion and sound.:
```
root@yi-c23c:/bin# mqttv4
Starting mqttv4 v0.1.0
Trying to connect... Client yi-cam sending CONNECT
Client yi-cam received CONNACK (0)
Client yi-cam sending PUBLISH (d0, q1, r1, m1, 'yicam/status', ... (6 bytes))
.
connected!
Client yi-cam received PUBACK (Mid: 1)
CALLBACK MOTION START
Thread 0 available
Client yi-cam sending PUBLISH (d0, q1, r0, m2, 'yicam/motion_detection', ... (12 bytes))
Client yi-cam received PUBACK (Mid: 2)
CALLBACK MOTION STOP
Client yi-cam sending PUBLISH (d0, q1, r0, m3, 'yicam/motion_detection', ... (11 bytes))
Thread 0 waiting
Thread 0 starting
Client yi-cam received PUBACK (Mid: 3)
```

```
Firmware Version	0.2.3
Base Version	9.0.36.00_202011101611
Serial Number	IFUSY65WX******
Model Suffix	y211ga
Load Average	2.09 1.82 1.50
Free/Total Memory	36240/60912 KB
```

mqttv4.conf:

```
# -----------------------------------------------------------------------------
# Set the topics where the messages will be published
# The final topic will be MQTT_PREFIX/TOPIC_MOTION
# -----------------------------------------------------------------------------

TOPIC_BIRTH_WILL=status
TOPIC_MOTION=motion_detection
TOPIC_MOTION_IMAGE=
MOTION_IMAGE_DELAY=0.5
TOPIC_MOTION_FILES=
TOPIC_SOUND_DETECTION=sound_detection

# -----------------------------------------------------------------------------
# Set the topics messages
# -----------------------------------------------------------------------------

BIRTH_MSG=online
WILL_MSG=offline
MOTION_START_MSG=motion_start
MOTION_STOP_MSG=motion_stop
AI_HUMAN_DETECTION_MSG=human
BABY_CRYING_MSG=crying
SOUND_DETECTION_MSG=sound_detected

# -----------------------------------------------------------------------------
# Other settings
# -----------------------------------------------------------------------------

MQTT_KEEPALIVE=120
MQTT_QOS=1
MQTT_RETAIN_BIRTH_WILL=1
MQTT_RETAIN_MOTION=0
MQTT_RETAIN_MOTION_IMAGE=0
MQTT_RETAIN_MOTION_FILES=0
MQTT_RETAIN_SOUND_DETECTION=0
```

camera.conf

```
SWITCH_ON=yes
SAVE_VIDEO_ON_MOTION=yes
SENSITIVITY=low
AI_HUMAN_DETECTION=no
FACE_DETECTION=no
MOTION_TRACKING=no
SOUND_DETECTION=yes
SOUND_SENSITIVITY=50
LED=yes
ROTATE=no
IR=yes

```

roleoroleo, 2022-10-05:
Do you receive audio messages in the app?

IvanUpa, 2022-10-05:
> Do you receive audio messages in the app?

Yes. The screenshot:
https://ltdfoto.ru/images/2022/10/05/IMG_0558.jpg

IvanUpa, 2022-10-06:
It works, thanks!
But the sensitivity is very bad, but this is a problem in the camera itself. Or I have a defective camera.

roleoroleo, 2022-10-07:
I think it's a problem of the model, not your camera.

IvanUpa, 2022-10-07:
Maybe. Thanks again for your help.

closed 2022-10-07 by IvanUpa

IvanUpa, 2022-10-10:
> I think it's a problem of the model, not your camera.

Tell me, in your experience, which model is the most stable?

roleoroleo, 2022-10-10:
I have:
- MStar y203c
- Allwinner y20ga
- Allwinner-v2 y21ga
- Allwinner-v2 h51ga

The best is the old one: y203c.