roleoroleo/yi-hack-Allwinner-v2 #511: Error setting up with Hass / yi-hack 0.2.3

wired-dev, 2022-10-12:
Logger: homeassistant.config_entries
Source: custom_components/yi_hack/__init__.py:60
Integration: Yi Cam with yi-hack ([documentation](https://github.com/roleoroleo/yi-hack_ha_integration))
First occurred: 7:37:25 PM (1 occurrences)
Last logged: 7:37:25 PM

Error setting up entry yi_hack_a2_000000 for yi_hack
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/yi_hack/__init__.py", line 60, in async_setup_entry
    CONF_TOPIC_BABY_CRYING: mqtt[CONF_TOPIC_BABY_CRYING],
KeyError: 'TOPIC_BABY_CRYING'

from a fresh install; when i upgraded my other units they didn't have the same issue; just this one that i had to hard reset cause it refused to download the file. I restored an config file from one of my other cameras and the issues resolved themselfs.

roleoroleo, 2022-10-14:
> from a fresh install; when i upgraded my other units they didn't have the same issue; just this one that i had to hard reset cause it refused to download the file. I restored an config file from one of my other cameras and the issues resolved themselfs.

Is it working, now?

duhow, 2022-10-23:
@wired-dev it seems to be fixed as of https://github.com/roleoroleo/yi-hack_ha_integration/commit/e624c4fdd6d42957cce6c85181c0cb7ae180fa45 , please check with latest version?

closed 2023-02-20 by stale[bot]