roleoroleo/yi-hack-Allwinner-v2 #513: Question to roleoroleo about which cam to buy today?

chilman408, 2022-10-18:
Hello!

I have 7 Yi Home 1080p cameras that are currently running the AllWinner (V1) yi-hack firmware.  They're doing great.

I'm wondering if I want to buy another one today, which camera should I get that is most bug-free and easy to setup?  (PTZ option would be nice too)

I tried to buy the Yi Dome U and the Yi Dome U Pro.  Unfortunately, the Yi Dome U only comes in RFUS* S/N and the Yi Dome U Pro (in my setup, following instructions here) cannot survive a power-loss reset after applying the hack...

roleoroleo, 2022-10-22:
r30gb is not recommended because it has hecv compression.
I have a h51ga but I'm hot happy.
Probably h52ga is the right choice.
Unforunately MStar and Allwinner (v1) are no longer available (I think).

About your bricked cam, I'm pretty sure that we can unbrick it.

chilman408, 2022-11-03:
Is there a direct way to buy h52ga in BFUS serial?

I tried to buy a couple from Amazon and they came in with RFUS serial...

roleoroleo, 2022-11-04:
I bought this model and it's a BFUS.
https://aliexpress.com/item/1005001894600719.html

But I think it's not a problem to support a different serial number.
If the yi firmware revision is the same, there are no differences between BFUS and RFUS.

> I tried to buy a couple from Amazon and they came in with RFUS serial...

If you still have them check the fw version.

closed 2023-06-08 by stale[bot]