roleoroleo/yi-hack-Allwinner-v2 #521: Log files location?

pblxptr, 2022-10-31:
Are there any helpful log files where I could have a look and try to find out why my camera keeps restarting from time to time?
In general, I'm using Yi Outdoor, all features except RTSP server are disabled but the camera works poorly - it's hard to get a stable rtsp stream (even low stream), and the camera restarts randomly e.g. after some time, sometimes it is 30 seconds, sometimes it's few minutes.

roleoroleo, 2022-11-01:
It often happens that the camera restarts if you are using snapshots.
The snapshot procedure requires a lot of memory and the kernel call a reboot.
Try to disable snapshot or to enable swap file.

pblxptr, 2022-11-01:
The same idea came to my mind so I disabled the snapshots. It looks better now.

closed 2023-03-02 by stale[bot]