roleoroleo/yi-hack-Allwinner-v2 #522: Yi Dome Guard YRS

MicheleMussi, 2022-11-05:
Hello,
 I have a Yi Dome Guard YRS, with firmware 9.0.46.00_20210081405.
I copied the files to my sd (formatted Fat32) from sdhack\r35gb and sysroot\r35gb.
So I have in the Factory folder with 2 files, a yi-hack folder with 1 fw_upgrade_in_progress 0kb file and in the root lower_half_init.sh.
After insertion the camera freezes, I wait a few minutes and after if I look in the sd I have the following content:
backup
Factory.done
log
yi-hack
lower_half_init.sh
If I remove the card it works if the re insert does not go. I also tried with another sd card. What am I doing wrong? Thank you

roleoroleo, 2022-11-05:
Check the procedure.
You have to download the released archive.

MicheleMussi, 2022-11-06:
Thank you, now it works fine.

NikNikolay7, 2023-02-03:
hello, if your camera is an international version for europe, please give a backup copy of your factory firmware
I have such a camera and had the imprudence to update the official firmware