roleoroleo/yi-hack-Allwinner-v2 #535: Accidental upgrade to 12.0.37 on y211ga IFUS

ArekInc, 2022-11-25:
Hi there, I have accidentally managed to update my IFUS y211ga to latest version witch is 12.0.37 and with 0.2.5 hack on it causes camera to reset on and off every minute or so. I wonder if someone have maybe a link to older firmware so I could downgrade my camera?

roleoroleo, 2022-11-25:
Please, before to restore the old firmware, backup the new one.
[y211ga_9.0.36.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10094551/y211ga_9.0.36.tar.gz)

ludoo, 2022-11-25:
sorry for the off-topic, but you don't happen to have a restore to 9.x for y291ga?

ArekInc, 2022-11-25:
> Please, before to restore the old firmware, backup the new one. [y211ga_9.0.36.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10094551/y211ga_9.0.36.tar.gz)

Thanks, much appreciated. Could you drop a link to the issue for restoring soft on y211ga -  or #305 is the one?

roleoroleo, 2022-11-25:
> sorry for the off-topic, but you don't happen to have a restore to 9.x for y291ga?

[y291ga_9.0.35.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10094960/y291ga_9.0.35.tar.gz)

> https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/305 is the one?

Yes. This is the issue and this is the wiki: https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/unbrick/README.md

ArekInc, 2022-11-26:
Cheers, all worked for me, back to ver 9.0.x 
Just a small tip for others. I was running the script on RPi and had to change the build.sh file permissions to 777, I had in on 644 by default for some reason.

roleoroleo, 2022-12-08:
Do you have a backup of version 12?

Jasonli8400, 2022-12-21:
> Cheers, all worked for me, back to ver 9.0.x Just a small tip for others. I was running the script on RPi and had to change the build.sh file permissions to 777, I had in on 644 by default for some reason.

I did the same mistake updating to the new version. Can you please help me with steps by steps how to revert back  to the old version 9.0.36 QFUS?

masyanich, 2023-01-22:
same problem here,  ifusy cam upgraded to 12x,  hack on it causes camera to reset on and off every minute or so. I need way to downgrade, according to the instructions, I created files from a backup for a similar camera of the same model with firmware 9 (backup_y21ga,home_y21ga,rootfs_y21ga) ,files are in the root of the card, the camera just does not want to flash them, I tried different cards of different capacities, nothing helps :(

roleoroleo, 2023-01-22:
12.0.37 is not a y211ga.
Probably it's an unsupported cam.
Please, send me a backup (version 9) if you have.

masyanich, 2023-01-22:
https://disk.yandex.ru/d/w5S3t--dIQ4R8w
y211GA v1.1 - is labled on the cam box

roleoroleo, 2023-01-22:
The fw is 9.0.36, so the cam is a y211ga.
But the new version should be 12.0.36 not 12.0.37.

Bodengriller, 2023-02-04:
`Do you have a backup of version 12?`
@roleoroleo here's my dump for your collection
[Uploading y211ga_12.0.37.0_202211241030.zip…]()

djRicky, 2024-05-08:
Hi @roleoroleo thanks for your great job. I have a camera y211ga with the same 12.0.37.0_202211241030 firmware. The camera was updated using original Y! app. And now after a reset I can't connect anymore to any wifi. Always getting Wifi connection error. Then I try with the hacks y211ga_0.3.1.tgz and y211ga_0.3.2.tgz and the backup is done automatically but nothing more. Setting up the wifi with configure_wifi.cfg doesn't work and using QR code still getting the same error. I don't know how to downgrade or fix this error.

djRicky, 2024-05-08:
This is my backup
[y211ga_12.0.37.0_202211241030.zip](https://drive.google.com/file/d/1SmlgrIOWJyzYw9pQOrc7W6jMERP6qjQn/view?usp=sharing)

djRicky, 2024-05-09:
Board photos 
[image]
[image]

roleoroleo, 2024-05-10:
Did you try a reset with the button?

TVC2024, 2024-10-31:
I need FW Yi Cam 1080p Home (y211ga).
can you send it to me? may be send previous version FW
[image]

roleoroleo, 2024-11-04:
[mtdblocks.tgz](https://github.com/user-attachments/files/17617719/mtdblocks.tgz)