roleoroleo/yi-hack-Allwinner-v2 #540: Problem 291ga - continuous reboot loop

xpinguinx, 2022-12-06:
The YI-HACK firmware works perfectly in fact I can enter the configuration page http://[IP-ADDRESS:8080].
The problem is that unfortunately after 10 seconds the CAM keeps restarting and this happens endlessly. If instead I extract the SD memory the cam resumes working regularly. I also tried to load the old versions but without success
I haven't found any reference how to fix it.
Regards

roleoroleo, 2022-12-06:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/FAQ#why-does-my-cam-often-reboot

xpinguinx, 2022-12-06:
Thanks for the reply.
I performed all the tricks and suggestions step by step before even sending the request for help but unfortunately with negative results.
I disabled the snapshot, I enabled the swap memory, I replaced the power supply with a bigger capacity one, I also disabled other services like FTP, ONVIF to optimize the ram. But unfortunately nothing!
Help me

roleoroleo, 2022-12-06:
Maybe something went wrong during the installation phase.
Try to install again using another SD card.

xpinguinx, 2022-12-06:
I tried 32GB memory from three different vendors (SanDisk, Kinkstone, Kodak). After a few seconds the cam loses connection because it restarts as can also be seen from the ping test.
What could this problem be caused by?
Thanks for the support

[image]
[image]

roleoroleo, 2022-12-07:
As you can see in this table https://github.com/roleoroleo/yi-hack-Allwinner-v2#supported-cameras
your cam is not supported.
Please send me a backup of the flash and I will try to support it.

xpinguinx, 2022-12-07:
RoleoRoleo thanks for the reply.
The cam is listed, in fact you can see the photo but unfortunately it doesn't seem to work :(
I backed up the flash as requested. Thanks again for the support.
Regards
[image]
[image]
[backup-fash.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10174281/backup-fash.zip)

roleoroleo, 2022-12-07:
Only fw 9.x is supported at the moment.

xpinguinx, 2022-12-07:
Overwritten the file as recommended. 
Unfortunately the orange LED stays on and won't start :(

Thank you for the time you are dedicating to me..
[image]

roleoroleo, 2022-12-07:
Is it bricked?
Or it works when you remove the sd...

xpinguinx, 2022-12-07:
When I remove the SD it works perfectly

roleoroleo, 2022-12-07:
Please, try again with this file.
[lower_half_init.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10177272/lower_half_init.zip)

roleoroleo, 2022-12-07:
I will add your cam to the list.

ec-blaster, 2022-12-07:
Hey!

I flashed it again and now it's stable.
I'll keep up trying tonight and tell you tomorrow...

ec-blaster, 2022-12-08:
I don't know why, but it was stable for a while, but as soon as I powered off the camera and switched in on again, the boot loop continues

ec-blaster, 2022-12-08:
> try enabling swap and disabling snapshot

It's done already. Both things. And memory seems ok.

The problem comes when I reboot or power off and on again.

But, if I remove the SD card, boot the original firmware, power off, insert the SD again and power it on, it works.

Strange, but it seems that it needs that the original firmware boots before the yi-hack.

ec-blaster, 2022-12-08:
OK, sorry to bother you, but it was my mistake:

As I had the browser windows open, I thought that the settings changes had been saved, but they weren't. So I went step by step, saving a change every time and waiting for the next reboot, and finally I was able to make it work.

The settings changed to make it work are:

* Swap file: enabled.
* Snapshot for recorded video: disabled.
* FTP: disabled.
* Telnet: disabled.

With these settings everything works OK, so I think this issue could be closed.

Thanks.

closed 2022-12-08 by roleoroleo

yivo, 2022-12-14:
> Please, try again with this file. [lower_half_init.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10177272/lower_half_init.zip)

This also helped me. Thank you.

Here are the details about my camera.

[image]

Blayven, 2023-01-14:
+1
Yi Home 1080p RFUS 12.0.35 - working! Thanks.

roleoroleo, 2023-01-21:
Please, wait the next release for a full support.

babsz02, 2023-01-30:
J'ai modifié le nouveau fichier et ça fonctionne merci !

> lower_half_init.zip