roleoroleo/yi-hack-Allwinner-v2 #541: Problems Running On Yi 1080p Home IFUS. No issues before. Did not change any setting.

gavsiu, 2022-12-08:
I was running v0.1.9 on firmware v8.* for a while with no issues until it stopped working one day. I have internet access blocked except for NTP.

I could never get it working again while v0.2.3 came and never figured it out (also didn't have much time to try). I thought the SD card was dead, so I bought a new one. Eventually v0.2.5 came and I realized the stock firmware still boots fine when the SD card is taken out. With a fresh copy of yi-hack, it boots, but whenever I change the settings, it never boots. I tried updating the stock firmware from v8 to v11. The camera works on stock firmware.

Eventually, I have now figured out I can change any setting in the yi-hack firmware other than the Timezone. If I try using the setting I used before, which is what "this page" specifies (PST8PDT,M3.2.0,M11.1.0), the camera will stop booting.

The other issue I'm encountering is that the low quality RTSP feed is now corrupted on my old tablet (SM-T377W) on Android 7.1.1. It can display the high quality feed, but FPS is low and I rather use the low quality setting. I have no issues with high/low feed on my Samsung Galaxy S22 Ultra with Android 13 (other than FPS). It could be a codec issue, but I did not have any issues on my tablet for the longest time.

I have 2 older Yi 1080p Home 9FUS webcams that have no issues and I haven't had to touch forever. I realize they run a different version of yi-hack, but I wish they didn't send me the newer one.

Any insights on why I can no longer change the Timezone and why the low quality feed is corrupted on certain devices?

roleoroleo, 2022-12-08:
> Eventually, I have now figured out I can change any setting in the yi-hack firmware other than the Timezone. If I try using the setting I used before, which is what "this page" specifies (PST8PDT,M3.2.0,M11.1.0), the camera will stop booting.

This commit should fix your TZ problem: https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/4db939ee69d7b134313dd2610be2b71cc328bf32

Check again and let me know.
About stream corrupted I don't know how to fix it. Try to use "Alternative RTSP server".

dzek69, 2022-12-09:
@gavsiu disable ntpd and check again, this is what fixes issues for me

gavsiu, 2022-12-10:
> > Eventually, I have now figured out I can change any setting in the yi-hack firmware other than the Timezone. If I try using the setting I used before, which is what "this page" specifies (PST8PDT,M3.2.0,M11.1.0), the camera will stop booting.
> 
> This commit should fix your TZ problem: [4db939e](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/4db939ee69d7b134313dd2610be2b71cc328bf32)
> 
> Check again and let me know. About stream corrupted I don't know how to fix it. Try to use "Alternative RTSP server".

This seems to have worked this time.

I think I rushed it last time after applying the fix by trying to restore settings saved right before rather than key in every change again. Or maybe there was an issue with me running firmware v12... Yeah I updated again while trying to fix it.

What I did this time was downgrade from v12 to v9.0.36 following https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/388, reset camera, upload v0.2.5, apply https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/4db939ee69d7b134313dd2610be2b71cc328bf32 fix, setting only timezone in yi-hack and reboot, then set every other setting manually. I've blocked the webcam from internet again through my router.

For whatever reason, the tablet now only works using software decoder, but whatever I guess.

It seems to be working, but fingers crossed. Last time it stopped working without me touching it.

Thank you again for your work.

closed 2022-12-10 by gavsiu