roleoroleo/yi-hack-Allwinner-v2 #542: Bricked(?) Yi 1080p Home IFUS

lazarevski, 2022-12-10:
So I've had this camera for a year or so, and when I got it, I successfully hacked it. And the cam worked great up until few months ago when the cam just stopped working - constant yellow light, no audio prompts. I don't have a backup and I don't remember what I flashed it with originally, so I don't know where to start. Can someone help me try to revive it?

[image]
[image]
[image]

roleoroleo, 2022-12-10:
It could be a y291ga or a y211ga.
Does it work if you switch on without the sd card?

lazarevski, 2022-12-10:
Same behavior with or without SD card. Was support for y291ga added just recently? I ordered this cam back in April '21 from ali, so it was probably flashed May/June 2021.

roleoroleo, 2022-12-10:
Do you have a backup inside your sd card?

lazarevski, 2022-12-11:
> Do you have a backup inside your sd card?

That's the thing, I've misplaced or have overwritten the card.

> Support for y291ga is from august.
> 
> If it's not a hardware failure, you could try to restore backup partition. If the file is not related to your model, it will not loaded. [y291ga.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10200658/y291ga.zip) [y29ga.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10200659/y29ga.zip) [y211ga.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10200660/y211ga.zip)
> 
> Read the wiki,

Thanks for the prompt replies, ill give it a try and report back.

wolverine-0, 2022-12-17:
@lazarevski did you have any luck?  I have the same model and I am in exactly the same boat as you.  I have tried the backup files linked above (thanks @roleoroleo) but haven't had any joy.  Still just a static orange light on boot.  I'm half tempted to think that the hardware has failed but I am willing to try anything else that can be suggested.

babsz02, 2023-02-02:
> @lazarevskiavez-vous eu de la chance? J'ai le même modèle et je suis exactement dans le même bateau que toi. J'ai essayé les fichiers de sauvegarde liés ci-dessus (merci@roleoroleo) mais je n'ai eu aucune joie. Toujours juste une lumière orange statique au démarrage. Je suis à moitié tenté de penser que le matériel a échoué, mais je suis prêt à essayer tout ce qui peut être suggéré.

Have you heard from your camera?
Because I have the same problem, fixed orange light

roleoroleo, 2023-02-02:
The only method to fix the orange led is restoring the original partition using a backup (if it's not a hardware failure).
If you have it, please follow the unbrick procedure.

lazarevski, 2023-02-02:
No luck on my end, probably a bricked device.

roleoroleo, 2023-02-04:
Are you sure about the model suffix?

twillist, 2023-02-21:
I had a similar problem and I had to solder wires on the uart pads and hook it to a serial port on a computer and the it would flash the recovery images from an sd card.

closed 2023-06-21 by stale[bot]