roleoroleo/yi-hack-Allwinner-v2 #544: Yi Outdoor - 9FUSY26

hjpfourie, 2022-12-12:
I have a newer Yi Outdoor Camera 1080p, which starts with 9FUSY26. Doesn't look like this camera is supported yet? 

Tried the YI Hack V5 which worked for my older Yi Outdoor cameras, but no luck on this one. 

Any plans to support this model @roleoroleo?

roleoroleo, 2022-12-12:
I need to know the fw version, you can read it inside the app.

roleoroleo, 2022-12-12:
> After some digging, I did see there was an attempt some time ago to create a firmware hack for this model, but it was abandoned due to what I can only imagine was the complexity to get it working. [roleoroleo/yi-hack-MStar#220](https://github.com/roleoroleo/yi-hack-MStar/issues/220) has the discussion.

If the model is the same discussed above, I can't support it. Too different than other platforms.

hjpfourie, 2022-12-12:
Thank you, thought as much.

closed 2022-12-12 by hjpfourie