roleoroleo/yi-hack-Allwinner-v2 #545: Yi 1080p Home RFUS - 12.0.35 - y291ga_0.2.5 - Does not record video in Motion Detection mode

schuchelov, 2022-12-12:
The camera records video correctly in "24/7" mode, but does not record video in "Motion Detection" mode (without y291ga, the camera works correctly in both modes)

Serial Number: RFUSY62V************
Base Version: 12.0.35.8_202211241029
Firmware Version: 0.2.5
Model Suffix: y291ga
UPDATE: lower_half_init.sh from https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/540

backup: https://disk.yandex.ru/d/WN2ccQuH0sLmSQ

roleoroleo, 2022-12-12:
This problem is hard to fix.
I need your support because I don't have this model.
When motion detection doesn't work, you mean through the app, I think. Does motion detection work via mqtt?
Could you list the running processes in both configurations (with and without the hack)?

schuchelov, 2022-12-12:
I'll be happy to help.

Switching "24/7" and "Motion Detection" is performed in "YI Home". 
"Motion Detection" does not work - I mean the absence of records for this event.
I checked "MQTT" - the camera does not send messages. 
(With identical settings, I checked the test for QFUS_9.0.36_y211ga_0.2.5 - messages come about "motion_detection", "motion_detection_image" and "motion_files")

roleoroleo, 2022-12-12:
Ok, next step...
I need a list of the running processes in both configurations.
Without the hack and with the hack configured with "Disable cloud"=off.

schuchelov, 2022-12-12:
Sorry, I do not know how to get a "list of running processes". Do you mean information from web pages (configurations, mqtt, camera_settings, eventsdir)? Or save the configuration? config_RFUS_12.0.35.tar.bz2 - https://disk.yandex.ru/d/SPLnmMMmqpF9lw
"With the hack" - maybe there is some command to output this information? (but "Without the hack" - is it possible to get such information from the camera somehow in the "YI Home" application?)

roleoroleo, 2022-12-12:
With or without the sd card containing the hack, you should be able to enter to the cam using telnet.
Open putty and run a telnet session.
The command to show the processes is 'ps'.

schuchelov, 2022-12-12:
Ok. List of running processes "With the hack" and "Without the hack" - https://disk.yandex.ru/d/FSzf1Wi9CJno-w

roleoroleo, 2022-12-13:
You should try to edit the file lower_half_init.sh in the root folder of your sd card:
1. Remove the sd from the cam.
2. Put the sd into your pc.
3. Make a backup copy of the file lower_half_init.sh lower_half_init.sh_ori
4. Open the file lower_half_init.sh with your preferred text editor.
5. Change the line
LD_PRELOAD=/tmp/sd/yi-hack/lib/ipc_multiplex.so ./dispatch &
with
./dispatch &
near the end of the file.
6. Save and eject the sd card.
7. Put the sd into the cam.
8. Check if motion detection works with the original app.

schuchelov, 2022-12-13:
Sorry. Both lines are found in the file.

`line 146: ./dispatch &`
`line 177: LD_PRELOAD=/tmp/sd/yi-hack/lib/ipc_multiplex.so ./dispatch &`

I tried
changing "line 146" to `LD_PRELOAD=/tmp/sd/yi-hack/lib/ipc_multiplex.so ./dispatch &`
and
changing "line 177" to `./dispatch &`

I didn't notice any difference. There are no records.

roleoroleo, 2022-12-15:
Try to remove the file /tmp/sd/yi-hack/bin/cloudAPI_real
`# rm /tmp/sd/yi-hack/bin/cloudAPI_real`
and restart the cam.

schuchelov, 2022-12-15:
1) "lower_half_init.sh" - returned the original
2) executed the command `rm /tmp/sd/yi-hack/bin/cloudAPI_real` and rebooted the camera

I didn't notice any changes. There are no records.

Opened the SD via the computer: in the folder "H:\yi-hack\bin " the file "cloudAPI_real" is present, but without the "creation/modification date". After restarting the camera - did it create itself?
https://disk.yandex.ru/i/FwYd98tgNdJGFQ

roleoroleo, 2022-12-16:
> Opened the SD via the computer: in the folder "H:\yi-hack\bin " the file "cloudAPI_real" is present, but without the "creation/modification date". After restarting the camera - did it create itself?

The hack creates this file. It's ok.

I probably found the problem, Try this beta please:
[y291ga_0.2.5.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10245404/y291ga_0.2.5.tar.gz)

schuchelov, 2022-12-16:
1) "this beta" is a continuous reboot loop.
2) update "lower_half_init.sh" from #540 - continuous reboot loop.
3) The "Swap file" is enabled - the camera has stopped restarting.

There are no video recordings.

4) additionally checked: "lower_half_init.sh" from "this beta" And "Swap File" is Enable = continuous reboot loop.

roleoroleo, 2022-12-17:
Another try please:
[y291ga_0.2.5.tgz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10251352/y291ga_0.2.5.tgz)

schuchelov, 2022-12-17:
1) "last beta" - the camera does not reboot. There are no records.
2) disable: a) Snapshot for recorded video b) ONVIF c) ONVIF WS-Discovery d) ONVIF watermark e) SSH - camera does not reboot. There are no records.
3) RTSP Audio - I change the value "null" to "Disabled (default)" - https://disk.yandex.ru/i/JAgyO1UuHROI6Q - continuous reboot loop.
4) "Reset to yi-hack default" - rebooted several times, but then stopped rebooting. There are no records.
5) The "Swap file" is enabled AND "RTSP Audio" - I change the value "null" to "Disabled (default)" - camera does not reboot. There are no records.

roleoroleo, 2022-12-18:
We need to check if the cam creates records internally.
Using the last beta, login with putty and check the folder `/tmp/sd/record` when you move the cam.
If there is a file tmp.mp4.tmp your cam has detected the motion.
Run the command:
`ls /tmp/sd/record`

schuchelov, 2022-12-18:
Run the command: `ls /tmp/sd/record` - The directory is empty. 
I move my hand in front of the camera - Run the command: `ls /tmp/sd/record` - The directory is empty.

roleoroleo, 2022-12-18:
Please, check if cloud application is running:
`ps`

schuchelov, 2022-12-18:
```
root@RFUS:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  211 root         0 SW   [oom_reaper]
  212 root         0 SW<  [writeback]
  214 root         0 SW   [kcompactd0]
  215 root         0 SW<  [crypto]
  216 root         0 SW<  [bioset]
  218 root         0 SW<  [kblockd]
  259 root         0 SW   [sys_user]
  266 root         0 SW   [kworker/0:1]
  267 root         0 SW<  [cfg80211]
  274 root         0 SW<  [watchdogd]
  288 root         0 SW<  [spi0]
  298 root         0 SW   [kswapd0]
  382 root         0 SW<  [bioset]
  387 root         0 SW<  [bioset]
  392 root         0 SW<  [bioset]
  397 root         0 SW<  [bioset]
  402 root         0 SW<  [bioset]
  407 root         0 SW<  [bioset]
  412 root         0 SW<  [bioset]
  417 root         0 SW<  [bioset]
  422 root         0 SW<  [bioset]
  450 root         0 SW   [irq/302-sunxi-m]
  452 root         0 SW   [irq/166-sdc0 cd]
  453 root         0 SW   [irq/303-sunxi-m]
  454 root         0 SW   [kworker/0:2]
  458 root         0 SW   [kworker/0:3]
  527 root         0 SW<  [kworker/0:1H]
  528 root         0 SW<  [bioset]
  529 root         0 SW   [kworker/u2:2]
  531 root         0 SW   [mmcqd/0]
  547 root         0 SWN  [jffs2_gcd_mtd4]
  585 root       988 S    /usr/sbin/telnetd
  597 root         0 SW<  [phy0-atbm_wq]
  598 root         0 SW   [phy0-usb_atbm_b]
  651 root      1356 S    ./dispatch
 1213 root     47964 S    ./rmm
 1297 root      1468 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.c
 1314 root      2736 S    ./mp4record
 1315 root      1228 S    ./cloud
 1316 root      3564 S    ./p2p_tnp
 1317 root      3536 S    ./oss
 1319 root       744 S    ./watch_process
 1333 root       860 S    {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/
 1344 root       844 S    pure-ftpd (SERVER)
 1368 root      1048 S    udhcpc -i wlan0 -b -s /backup/tools/default.script -
 1428 root       860 S    {ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/u
 1463 root       996 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.
 1518 root       860 S    {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/
 1768 root      4060 S<   rRTSPServer -m y291ga -r high -a no -p 554
 1902 root       980 S    -ash
 1951 root       980 S    sleep 10
 1977 root         0 Z    [init]
 1979 root       980 R    ps
```

roleoroleo, 2022-12-19:
cloud application is ok, I don't understand...
Just a little recap:
- If you remove the sd card, motion detection works.
- When the hack is running, motion detection doesn't work.
- cloudAPI application is updated and running.
- cloud application is running.
- All other mandatory applications are running (dispatch, rmm and p2p_tnp).

If that's the case I don't know how to fix it.
I need to connect in console to your cam and do some testing.

schuchelov, 2022-12-19:
If you remove the sd card, motion detection works. - YES
When the hack is running, motion detection doesn't work. - YES

"I need to connect in console to your cam and do some testing." - `OK`

My English is not good enough to have a conversation in "real-time voice" - but if a website with a translator is opened in parallel and we type, then we will understand each other =) You can connect either through Chrome Remote Desktop or AnyDesk. Having previously written off to Skype or WhatsApp or another messenger.

roleoroleo, 2022-12-21:
@schuchelov What do you think about tomorrow?
I'm UTC+1.

> You can connect either through Chrome Remote Desktop or AnyDesk. Having previously written off to Skype or WhatsApp or another messenger.

Search roleoroleo on telegram.

evider2002, 2022-12-30:
@roleoroleo After reading all the recent threads, this is what I believe, please tell me if I am correct. 
fw11 motion doesn't work. 
fw12 original issue was that camera was rebooting, but you fixed that issue with updated lower_half_init.sh. Now the issue is that motion doesn't work. 
So the last fw version that motion works with is 9? Motion does not work with 11/12? 

Thank you.

roleoroleo, 2023-01-03:
As explained here the motion detection feature is changed between 9 and 11.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/550
It's not true that it doesn't work, simply it changed.
I already updated the hack to work with this new fw but I need to test it.
If you want, I can send you a beta to test.

evider2002, 2023-01-03:
@roleoroleo This comment was from 4 days ago. 
By now I already know that you have the hack working on 12... 
I indeed tested it and commented on it here https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/547#discussioncomment-4579267

schuchelov, 2023-01-09:
Sorry I'm missing. New Year. Indicated in the "YI Home" application, "motion detection" is enabled in two places.
#1) SD recording mode - "recording detected activity" or "Recording continuously 24/7"
#2) Settings - (Intelligent Detection) Settings - Movement

The v9 - it was enough to switch only in the #1 setting. The v12 - everything worked correctly when I turned on setting #2.

(I don't know what the rules are here - I got a consultation - the problem is solved - the branch can be closed)

closed 2023-05-09 by stale[bot]

gabest11, 2023-06-20:
I have the same problem. If I do 1+2 as above then it does record a single event, but only every 8-10 minutes, nothing in between. Sensitivity high has no effect.

roleoroleo, 2023-06-20:
1 means "recording detected activity" or "Recording continuously 24/7"?

gabest11, 2023-06-20:
Only activity. I am still testing it. The alert frequency seems to have an effect on it. If I set to the highest, it only leaves a very small gap between the 1min recordings. But this also sends a lot of notifications to the phone, so silly that it is linked at all.

gabest11, 2023-06-20:
I have one more observation. "Motion Detection" (yi-hack, camera settings) cannot be turned on, only if I edit the config file. If I switch it on in the web UI, I get a single recording and it is off again.

gabest11, 2023-06-20:
It was a dual package, so I have another camera. This one records clips just as erratically without the yi-hack fw. So disappointing, since I had a 6FUS (MStar) before and that worked flawlessly.

roleoroleo, 2023-06-20:
> I have one more observation. "Motion Detection" (yi-hack, camera settings) cannot be turned on, only if I edit the config file. If I switch it on in the web UI, I get a single recording and it is off again.

This bug was fixed here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/b40b880afa217446582eeaa6664541f661ccb59c

> It was a dual package, so I have another camera. This one records clips just as erratically without the yi-hack fw. So disappointing, since I had a 6FUS (MStar) before and that worked flawlessly.

I don't understand exactly what do you want to obtain.
Record activity? Notifications?
Please, explain me better.

Regarding the hack, it doesn't change these features on the cam. It just allows to enable/disable them without using the app (and maybe the cloud).

gabest11, 2023-06-20:
Okay, I'm going to summerize my experience.

- before yi-hack, I setup the camera in yi app, changed recording type to "Activity Detection Recording", nothing was recorded, it did not work out-of-the-box...
- turned off "Activity Detection Recording", it recorded everything 24*7, as it should
- put yi-hack onto sdcard
- "Activity Detection Recording" ON again, nothing was recorded, again
- searched online, arrived here, figured I also have to enable "Motion Detection" for yi-hack, a partial sub-1min file and the following 1min file was recorded, but nothing more, I could jump up and down in front of the camera, only after 10 minutes did it record again
- played with the sensitivity and notification frequency in the yi app and I managed to shorten the delay between two recordings to 5 or 1 minutes, no idea how, but the notification seems to be part of it
- now, the strangest part, set "Motion Detection" OFF in the yi app and it started recording for more than 1 minutes continously, and nothing when there was nothing to record
- also turned off "Motion Detection" in yi-hack, rebooted, still records correctly (fingers crossed)