roleoroleo/yi-hack-Allwinner-v2 #549: Determine which Yi 1080 Home Camera From DID

burchis, 2022-12-22:
I have RFUSY6259TBVC... with FW 12.0.35.8* .  Like others I have lost motion record to SD-card and wishing to downgrade firmware.  Here is a screen shot from Kami firmware download page. The third option (XFXXX65XXX..) has a y211ga file but my camera is XFXXX62. Can you confirm which option I need?  I have flashed two other cameras (9FUS) with y20ga and all went well.  I believe I use to have 9.0.35 which uses y291ga and that is not listed on Kami. 

[image]

roleoroleo, 2022-12-22:
Your cam is a y291ga and the firmware is not available in the Kami web site (AFAIK).
In any case, a downgrade is not a good idea if you don't have a complete fw.
Normally, the fw available as a download file is only a partial update.

You can restore fw 9.x if you have a backup of the old fw or if another user share its own backup.
Or you can wait the next update of the hack which will probably support fw 12.

burchis, 2022-12-22:
Thanks for your reply. I found and downloaded a backup file named 'backup_rfus_9_0_35'.  I am assuming it is valid.  After extracting the zip file a folder titled 'backup' existed with these files: 

[image]

[image]

Do I place the backup folder in the SD-card root directory or copy all the individual bin files or just certain bin files. 

Any tips much appreciated, thanks in advance.

roleoroleo, 2022-12-22:
Read the wiki about unbrick procedure.
It's the same.
Try to overwrite mtdblock3 and mtdblock4.
It should be enough.

closed 2023-04-22 by stale[bot]