roleoroleo/yi-hack-Allwinner-v2 #551: How to disable startup sound

mopicus, 2022-12-26:
Hello, I have a MIBAO G1 1296p dome camera
I'm looking if there is any way on how to disable the startup sound of the camera, which is very annoying.
Any help will much appreciated!

roleoroleo, 2022-12-27:
Are you able to detect which file is played?
Audio file are in /home/app/audio_file
If you find the file we can create an empty file and add a mount bind.

mopicus, 2022-12-28:
Yes, it is in /home/app/audio_file/common
How I could the mount?

roleoroleo, 2022-12-28:
Try this procedure.
Open the file /tmp/sd/lower_half_init.sh
and add these new lines:
```
touch /tmp/poweron.aac
mount --bind /tmp/poweron.aac /home/app/audio_file/common/poweron.aac
```
at the end, between these 2 lines:
```
mount --bind /tmp/sd/yi-hack/script/wifidhcp.sh /backup/tools/wifidhcp.sh

LD_PRELOAD=/tmp/sd/yi-hack/lib/ipc_multiplex.so ./dispatch &
```

mopicus, 2022-12-28:
It worked, thank you very much!

closed 2022-12-28 by mopicus

roleoroleo, 2022-12-28:
When you update the hack, this change will be overwritten.
I will add it to the next release.