roleoroleo/yi-hack-Allwinner-v2 #552: YI Dome Guard IFUS - camera is restarting when connected to RTSP stream

SirAd, 2022-12-27:
Hi, 
I have installed r30gb 0.2.5 hack on yi dome guard camera with firmware starting from IFUS. Camera is restarting when I'm trying to connect to RTSP. I've captured debug from rRTSPServer.

SirAd, 2022-12-27:
see  rRTSPServer output with debug
[rRTSPServer.err.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10307040/rRTSPServer.err.txt)

Giuserver, 2022-12-27:
Hello. Try to enable Swap space! After that, use tcp transport.
If you use ffplay it's like:
ffplay -rtsp_transport tcp -i "rtsp://IP-CAM/ch0_0.h264"

SirAd, 2022-12-28:
I'm closing this ticked. I applied change described in #484 - swith to h264 and now camera is stable.
Thanks for assistance.

closed 2022-12-28 by SirAd