roleoroleo/yi-hack-Allwinner-v2 #553: FTP upload keeps uploading lots of files

mopicus, 2022-12-30:
Hello, I have setup the yiihack options to upload files to FTP
In camera settings I have set Low detection sensitivity and AI Human detection on, but the camera uploads a new file every 30 seconds, the minimum movement triggers it, without any human motion.
I have tried different settings but the result is always the same.

Model is a MIBAO G1 1296p dome with 0.2.5 fw

Thanks!

roleoroleo, 2023-01-03:
Check if this option is true:
[image]

mopicus, 2023-01-04:
Yes I have tried both options, checked and unchecked. In both cases, as soon as I enable the FTP upload, the camera starts uploading lots of files

roleoroleo, 2023-01-04:
FTP upload is not related to the quantity of video files.
The problem is: why your cam saves lots of files.

mopicus, 2023-01-05:
These days is working ok, but there is no wind at all, I will keep an eye on it
Thanks!

closed 2023-01-05 by mopicus