roleoroleo/yi-hack-Allwinner-v2 #556: Difference between y291ga and y211ga

cjgarciao, 2023-01-01:
Hello,
I've a YI 1080p Home with firmware 9.0.35 and I've mistakenly installed y211ga instead y291ga. The camera looks work fine with the hack, what is exactly the difference? Should I remove y211 and install y291?.
Thanks,
Carlos.

roleoroleo, 2023-01-02:
The hack is pretty much the same for all models.
The big difference is the structure of the frame grabber file but for y211ga and y291 it's the same.
Another difference is the file /tmp/sd/lower_half_init.sh

In general you should install the hack for y291ga. If you don't want to create from scratch your sd, simply overwrite the file `/tmp/sd/lower_half_init.sh` and edit the file `/tmp/sd/yi-hack/model_suffix`

cjgarciao, 2023-01-02:
Hello,
Ok, I will change the two files on the sd card.
If I want to remove the hack, is removing the sd card enough or should I restore the backup?
Thanks.
Carlos.

roleoroleo, 2023-01-03:
There are a couple of modified file in the backup partition.
But it's not a big problem, the cam works properly.

If you want to remove the hack completely you can edit these files or restore the original partition.

closed 2023-05-04 by stale[bot]