roleoroleo/yi-hack-Allwinner-v2 #557: Dome RFUS camera

Dmario-dev, 2023-01-02:
Hi,
Just got a new Yi Dome Camera.
Serial is:
RFUSY73*
Firmware:
9.0.46.00

Is there a firmware in the making for it?
If so, any way to Beta Test it?

a-t-a-r-i, 2023-01-03:
Hi,

I got a RFUSY73 up and running today, but I can't say if it's stable yet.

```
root@yi-22c2:~# cat /tmp/sd/yi-hack/version 
0.2.5
root@yi-22c2:~# cat /home/homever
9.0.46.00_202110081405
root@yi-22c2:~# cat /tmp/sd/yi-hack/model_suffix 
r35gb
```

I have done this

1. Installed the https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.2.5/r35gb_0.2.5.tgz release
3. Reconfigured the cam in h264: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/484#issuecomment-1235411673

roleoroleo, 2023-01-03:
@Dmario-dev The cam is probably a r35gb. Try it.

@a-t-a-r-i You don't need to reconfigure the cam in h264 because your cam is a r35gb not a r30gb.

a-t-a-r-i, 2023-01-04:
Ok, I guess I got a little confused when I tried it out.

I deleted the SD card and tried this again.

With swap and most services turned off I get the low res rtsp stream to run. The high res stream is not quite working yet. 

Before I turned on swap the rRTSPServer server was restarted about every 10 seconds.

Now I just have to find out what I did to get the high res stream running.

roleoroleo, 2023-01-06:
Normally the swap file fixes a problem related to snapshots.
RTSP should work with or without swap.

closed 2023-05-06 by stale[bot]