roleoroleo/yi-hack-Allwinner-v2 #558: I have installed the wrong Firmware in my cameras 2x RFUS and IFUS

keecy, 2023-01-05:
HI Everyone, 

Since I didn't read properly i did shit.
_"If you want to know if your cam is supported, please check the serial number (first 4 letters) and the **firmware version**."_
I forgot to read the firmware version part.

So I have:
 - RFUS 
   Base firmware: 9.0.35*
   By the supported table: y291ga
   What I have  installed:   y211ga
   Status: Working  
   Backup: no
   
 - RFUS 
   Base firmware: 9.0.35*
   By the supported table: y291ga
   What I have  installed:   y211ga
   Status: Bricked (When turned on just shows the yellow led)
   Backup: yes
   
  - IFUS 
   Base firmware: 9.0.19*
   By the supported table: y21ga
   What I have  installed:   y211ga
   Status: Working
   Backup: no
   
   
  @roleoroleo you were right, my camera was not a y211ga. #305 
  
  Can you guide me to install the correct firmware in my cameras ?
  and unbrick the RFUS,
  
  Thanks in advance, 
  Br, 
  keecy

Giuserver, 2023-01-05:
If I'm not mistaken, I've read that it can be done like this: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/305#issuecomment-1350534700 .Use first backup, then home file.

keecy, 2023-01-06:
**UPDATE**
Using the the unbrick procedure.

RFUS
Base firmware: 9.0.35*
Supported table: y291ga
What I have installed: **y291ga**
Status: back to Live with the correct firmware
Backup: yes

RFUS
Base firmware: 9.0.35*
Supported table: y291ga
What I have installed: **y291ga*
Status: back to Live
Backup: yes

The one below is more complicated, after to load the backup file doesn't come up.
IFUS
Base firmware: 9.0.19*
Supported table: y21ga
What I have installed: y211ga
Status: Working
Backup: yes

Br, 
keecy

roleoroleo, 2023-01-06:
About your y21ga you should try to restore the backup partition with 2 different files:
- the file produced with build.sh factory
- the file produced with build.sh hacked

The bootloader saves the hash of last partition loaded, so you can't load 2 times the same file.
If you tried factory, try hacked.
If you tried hacked, try factory.

keecy, 2023-01-06:
Hi Roleo, 
Thanks for the answer.
I had it work, after use the unbricked procedure after i have connected the RX TX GND to the FDTI serial converter.
i turned it off and it doen't start. strange.

I connected over uart
and i got this
`[310]BOOT0 commit : 81c18ee
[312]board init start
[315]set pll start
[317]set pll end
[319][pmu]: bus read error
[321]board init ok
[323]chip id check OK
[325]DRAM BOOT DRIVE INFO: V0.41
[329]DRAM CLK = 528 MHz
[331]DRAM Type = 2 (2:DDR2,3:DDR3)
[334]DRAMC read ODT  off.
[337]DRAM ODT off.
[339]DRAM SIZE =64 M
[346]DRAM simple test OK.
[349]rtc standby flag is 0x0, super standby flag is 0x0
[354]dram size =64
[356]key press : [
[358]spinor id is: ef 40 17, read cmd: 03
[362]Succeed in reading toc file head.
[366]The size of toc is 4c000.
[419]Entry_name        = optee
[422]Entry_name        = u-boot
[426]Entry_name        = dtb
[429]Jump to secend Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-12-g97f2688 #1 2019å¹´ 11æœˆ 08æ—¥ æ˜ŸæœŸäº” 08:26:51 UTC arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:84: no pmu node

U-Boot 2018.05-00011-gf4c6972 (Oct 29 2020 - 17:38:02 +0800) Allwinner Technology

[00.475]DRAM:  64 MiB
[00.477]Relocation Offset is: f9f94000
[00.489]secure enable bit: 0
[00.491]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.498]gic: sec monitor mode
[00.500]flash init start
[00.502]workmode = 0,storage type = 3
SF: Detected w25q64cv with page size 256 Bytes, erase size 64 KiB, total 8 MiB
[00.519]sunxi flash init ok
[00.522]Loading Environment from SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x3e00, len: 0x100
OK
[00.583]try sprite_led_gpio config
xxxxxxxxxxxxxxlijun uboot xiaoyiledinit
[00.591]sprite_led_gpio start
[00.594](weak)update dtb dram start
[00.598]update dtb dram  end
[00.600]update dts
root_partition is rootfs
[... 77 lines cut ...]
part_name:env start_offset:0x3b20 part_size:131072 
Nothing updated, start normal boot
Hit any key to stop autoboot:  1  0
=> 
=> 
=> 
=> 
=> U-Boot 2018.05-00011-gf4c6972 (Oct 29 2020 - 17:38:02 +0800) Allwinner Technology
Unknown command 'U-Boot' - try 'help'
=> 
=> 
=> 
=> [00.475]DRAM:  64 MiB
Unknown command '[00.475]DRAM:' - try 'help'
=> 
=> [00.477]Relocation Offset is: f9f94000
Unknown command '[00.477]Relocation' - try 'help'
=> 
=> [00.489]secure enable bit: 0
Unknown command '[00.489]secure' - try 'help'
=> 
=> [00.491]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
Unknown command '[00.491]CPU=816' - try 'help'
=> 
=> [01>>>>> U-Boot 2018.05-00011-gf4c6972 (Oct 29 2020 - 17:38:02 +0800) Allwinner Technolog>>>> [00.475]DRAM:  64 Mi>> [00.477]Relocation Offset is: f9f9400>> [00.489]secure enable bit: >> [00.491]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mh>> [01>>>>> U-Boot 2018.05-00011-gf4c6972 (Oct 29 2020 - 17:38:02 +0800) Allwinner Technolog>>>> [00.475]DRAM:  64 Mi>> [00.477]Relocation Offset is: f9f9400>> [00.489]secure enable bit: >> [00.491]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mh>> [01>>>>> U-Boot 2018.05-00011-gf4c6972 (Oct 29 2020 - 17:38:02 +0800) Allwinner Technolog>>>> [00.475]DRAM:  64 Mi>> [00.477]Relocation Offset is: f9f9400>> [00.489]secure enable bit: >> [00.491]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mh>> [01>>>>> U-Boot 2018.05-00011-gf4c6972 (Oct 29 2020 - 17:38:02 +0800) Allwinner Technolog>>>> [00.475]DRAM:  64 Mi>> [00.477]Relocation Offset is: f9f9400>> [00.489]secure enable bit: >> [00.491]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz`

keecy, 2023-01-06:
My y21ga is a bit crazy, with some power supplies starts, with others no.

I guess there is some problem with it.

Thanks  Roleo for your support.

Br,

closed 2023-01-06 by keecy