roleoroleo/yi-hack-Allwinner-v2 #559: YI firmware update (reset necessary)?

tunerooster, 2023-01-08:
Is it necessary to factory reset, re-pair and reinstall yi-hack when doing a YI Firmware update?

1. My camera is hard to access physically to get at the reset button.
2. I have made a number of additions to the current yi-hack, which will be time consuming ti redo.

I hesitate to just "try it" because last time I did a YI Firmware update my camera would no longer boot (https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/454).

I am hoping that my recent upgrade to yi-hack update 0.2.5 might now allow a YI Firmware update without completely starting over.  My yi-hack model is h30ga.

Can anyone offer any advice?  Has anyone upgraded YI Firmware recently on top of yi-hack 0.2.5?

Thanks!

roleoroleo, 2023-01-09:
> Is it necessary to factory reset, re-pair and reinstall yi-hack when doing a YI Firmware update?

There is no answer to this question.
It depends on the content of the firmware update.
There are additive firmware update that don't touch the hack and full firmware update that overwrite part of the hack.

Usually, neither reset nor re-pairing is necessary.
The hack should be reinstalled only if it doesn't work anymore (in case of full firmware update).

tunerooster, 2023-01-09:
I assume there is no way to know if the YI Firmware is a full update. or additive.

In that case, i guess the rule is:

_Don't update YI Firmware unless you are prepared to retrieve camera, reset and re-pair,  in the event you are unlucky, and the update was a full update that overwrites part of the hack_.  Which is what happened to me in #454.

Since then (over 6 months ago), I have not updated any of my cameras with new YI Firmware.  Is there any reason to ever update?  I sometimes fear I am missing out on something.   :)

Thanks for your advice!

closed 2023-01-11 by tunerooster