roleoroleo/yi-hack-Allwinner-v2 #562: FTP directory name not matching timestamp

Plop, 2023-01-09:
Hi,

I've got an issue with the directories names on the FTP : it always set the time one hour behind the real (and displayed on the cam overlay) time : [image]

Also, is it possible to rename the pattern of the folders and files ?

Also, I don't understand why some files starts with "E0" and others with "E1" ?

[image]

[image]

Thanks for your help !

roleoroleo, 2023-01-14:
> I've got an issue with the directories names on the FTP : it always set the time one hour behind the real

What's the correct time?
"Date & time" column or "Directory name" column?

> Also, is it possible to rename the pattern of the folders and files ?

No, it's defined by Yi software.

> Also, I don't understand why some files starts with "E0" and others with "E1" ?

I don't know, probably it's the type of alarm.

Plop, 2023-01-15:
> > I've got an issue with the directories names on the FTP : it always set the time one hour behind the real
> 
> What's the correct time? "Date & time" column or "Directory name" column?

The "Date & time" column

Thanks for your answers !

closed 2023-05-15 by stale[bot]