roleoroleo/yi-hack-Allwinner-v2 #563: ONVIF server for r35gb bad media URL and motion events for Scrypted

xorguy, 2023-01-15:
Hi.

Seems ONVIF server for r35gb is not working, requesting its capabilities returns URLs with "http://&#x7;:80/onvif/media_service" and not with the correct IP address, like "http://192.168.0.12:80/onvif/media_service".

Tried to add to Scrypted and it did not get any snapshot/video, tried with cURL and returned that bad URL.

Reproducible with:
1. Create request.xml with content:
`<s:Envelope xmlns:s="http://www.w3.org/2003/05/soap-envelope" xmlns:tds="http://www.onvif.org/ver10/device/wsdl">
  <s:Header/>
  <s:Body>
    <tds:GetCapabilities>
      <tds:Category>All</tds:Category>
    </tds:GetCapabilities>
  </s:Body>
</s:Envelope>` 
2. Send cURL with setting current camera IP address:
`curl -X POST -H "Content-Type: application/soap+xml; charset=utf-8" -d @request.xml http://192.168.0.12/onvif/device_service`

Returned XML:
`<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:xmime="http://tempuri.org/xmime.xsd" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:c14n="http://www.w3.org/2001/10/xml-exc-c14n#" xmlns:ds="http://www.w3.org/2000/09/xmldsig#" xmlns:saml1="urn:oasis:names:tc:SAML:1.0:assertion" xmlns:saml2="urn:oasis:names:tc:SAML:2.0:assertion" xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd" xmlns:xenc="http://www.w3.org/2001/04/xmlenc#" xmlns:wsc="http://docs.oasis-open.org/ws-sx/ws-secureconversation/200512" xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd"><SOAP-ENV:Header></SOAP-ENV:Header><SOAP-ENV:Body><tds:GetCapabilitiesResponse><tds:Capabilities><tt:Device><tt:XAddr>http://&#x7;:80/onvif/device_service</tt:XAddr><tt:Network></tt:Network><tt:System><tt:DiscoveryResolve>false</tt:DiscoveryResolve><tt:DiscoveryBye>false</tt:DiscoveryBye><tt:RemoteDiscovery>false</tt:RemoteDiscovery><tt:SystemBackup>false</tt:SystemBackup><tt:SystemLogging>false</tt:SystemLogging><tt:FirmwareUpgrade>false</tt:FirmwareUpgrade><tt:SupportedVersions><tt:Major>20</tt:Major><tt:Minor>12</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>17</tt:Major><tt:Minor>6</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>6</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>4</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>0</tt:Minor></tt:SupportedVersions></tt:System><tt:IO></tt:IO><tt:Security><tt:TLS1.1>false</tt:TLS1.1><tt:TLS1.2>false</tt:TLS1.2><tt:OnboardKeyGeneration>false</tt:OnboardKeyGeneration><tt:AccessPolicyConfig>false</tt:AccessPolicyConfig><tt:X.509Token>false</tt:X.509Token><tt:SAMLToken>false</tt:SAMLToken><tt:KerberosToken>false</tt:KerberosToken><tt:RELToken>false</tt:RELToken></tt:Security></tt:Device><tt:Media><tt:XAddr>http://&#x7;:80/onvif/media_service</tt:XAddr><tt:StreamingCapabilities><tt:RTPMulticast>false</tt:RTPMulticast><tt:RTP_TCP>false</tt:RTP_TCP><tt:RTP_RTSP_TCP>true</tt:RTP_RTSP_TCP></tt:StreamingCapabilities></tt:Media><tt:PTZ><tt:XAddr>http://&#x7;:80/onvif/ptz_service</tt:XAddr></tt:PTZ></tds:Capabilities></tds:GetCapabilitiesResponse></SOAP-ENV:Body></SOAP-ENV:Envelope`

Tried with a r30gb camera and XML contained correct URL.

Both cameras tried to add to Scrypted and even the video can be played, adding those to HomeKit never triggered for any movement. Is there any option to be able to add the cameras to Scrypted with motion alerts with ONVIF?

roleoroleo, 2023-01-21:
Did you set this option properly?
[image]

xorguy, 2023-01-22:
Right, one of the cameras was on Ethernet and the setting was not set.
Regarding Scrypted, I think there is a problem with movement detection not being sent via onvif, is it possible?

roleoroleo, 2023-01-23:
Sorry but event management is not implemented in this onvif version.
I tried but I have not been able to fix memory leaks.
Through onvif you can access to strean and snapshot.

0xgrm, 2023-01-23:
Fortunately you can easily implement motion detection in Scrypted with MQTT. 
It works perfectly.

xorguy, 2023-01-23:
> Fortunately you can easily implement motion detection in Scrypted with MQTT.
> It works perfectly.

That's what I'm looking at now, but I am fighting with the lines for the MQTT at the motion sensor so it gets both the motion_start and the motion_stop, if it is not much to ask, do you have how to setup?

Thanks in advance!

closed 2023-05-24 by stale[bot]