roleoroleo/yi-hack-Allwinner-v2 #566: OUTDOOR YHS3017 - RFUS Y59N8L

bettiii, 2023-01-22:
Hallo,
I try many fw but nothing... MayBe the cam isn't yet support?
many thanks

bettiii, 2023-01-22:
the fw installated now is 9.0.26_08_202203181055

roleoroleo, 2023-01-22:
h30ga

bettiii, 2023-01-22:
thanks, I try, I see the main page 8080, the single image photo is ok but the streaming not run. Now I retry

roleoroleo, 2023-01-23:
Which rtsp client are you using?
Did you try with vlc?

TZSENS, 2023-01-27:
> thanks, I try, I see the main page 8080, the snapshot image photo is ok but the streaming not run. Now I retry rtsp://192.168.178.129/ch0_0.h264, when click here nothing happen

which firmware you installed working? I have YI outdoor RFUSXXXX ,can not install the hack firmware.

kubikub, 2023-01-31:
Hello, 

It work for me, same serial root RFUSY59 and same firmware. I was trying Yi-Hack v5 first but it did'nt work. With the yi-hack-Allwinner-v2 it work great. Just copy the rtsp link into VLC/new network stream and fill up login and password if setted.

guido-30, 2023-02-02:
Hello,
I try to install yi -hack-Allwinner-v2 r30gb_0.2.5.tgz but not run. I format in FAT 32 the SD and copy the 2 folder and 1 file in the SD (is correct?)

guido-30, 2023-02-02:
Hello kubikub,
have you a backup the original fw the camera? I lost and I can't ,connect it.
many thanks.

roleoroleo, 2023-02-04:
> Hello,
> I try to install yi -hack-Allwinner-v2 r30gb_0.2.5.tgz but not run. I format in FAT 32 the SD and copy the 2 folder and 1 file in the SD (is correct?)

If it doesn't work, the card is not correctly recognized or the model is not a r30gb.

Frostbite1991, 2023-02-15:
I'm having the same problem, except snapshot doesn't work either. Also I can only access the webUI if I enter in the IP address. "IP-CAM:8080" does nothing for me. Tried streaming through VLC and it goes into a "cannot connect" error loop until I end the program process. Tried with full credentials, without, and with only username to no avail. It seems the Outdoor RFUS cam is not quite stable yet.  

Hostname | yi-2e46

Firmware Version | 0.2.5
Base Version | 9.0.26.08_202203181055
Model Suffix | h30ga
Serial Number | RFUSY5921OQP9E220620

kubikub, 2023-02-15:
> Hello kubikub, have you a backup the original fw the camera? I lost and I can't ,connect it. many thanks.

Hi, unfortunately I did the upgrade to the 11 version which is not yet supported. Of course without saving the 9.26 version .... 😖...
If you got it please share it 😉

closed 2023-06-15 by stale[bot]