roleoroleo/yi-hack-Allwinner-v2 #568: 3 days after update camera down

tarbax, 2023-01-23:
have 3 cameras updated them all and 1 out of 3 went down after couple of days.
i rebooted them multiple times but nothing pops up anymore.
got a blue status light and no camera feed, web interface or ssh

roleoroleo, 2023-01-24:
Strange.
When the led is blue, normally the cam is working properly.
Maybe the sd card is corrupted and the hack doesn't start.

tarbax, 2023-01-25:
> Strange. When the led is blue, normally the cam is working properly. Maybe the sd card is corrupted and the hack doesn't start.

so i got a new sd took all the files onto it and wont boot only got a solid yellow light.
reset button dousnt work what should i do now ?

roleoroleo, 2023-01-25:
Are you using the original cable and power adapter?

tarbax, 2023-01-31:
> Are you using the original cable and power adapter?

was a wifi issue. now i got a blue solid light and connects to network but no webpage shows up.
do i have to reset the camera after a new sd install (clean install from download) ?

roleoroleo, 2023-02-02:
No, reset is not necessary.
Are you using port 8080?

tarbax, 2023-02-02:
> No, reset is not necessary.
> Are you using port 8080?

Yes

roleoroleo, 2023-02-02:
Try to login with telnet.
If the cam is hacked you will enter.
If not, the hack didn't work.

tarbax, 2023-02-12:
> Try to login with telnet. If the cam is hacked you will enter. If not, the hack didn't work.

i reformatted on a windows instead of mac and now it works. thanks!

closed 2023-02-12 by tarbax