roleoroleo/yi-hack-Allwinner-v2 #571: New beta release for fw 12

roleoroleo, 2023-02-02:
If your cam is updated with fw 12 and you want to try a new beta, please try these files:
[r35gb_x.x.x.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10567167/r35gb_x.x.x.zip)
[y21ga_x.x.x.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10567172/y21ga_x.x.x.zip)
[y211ga_x.x.x.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10567169/y211ga_x.x.x.zip)
[y291ga_x.x.x.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10567173/y291ga_x.x.x.zip)

Unzip the file inside the sd card (/tmp/sd folder of the cam).
You should have the file with the .tgz extension.
Go to "Maintenance" webpage and click "UPGRADE FIRMWARE" button.

NikNikolay7, 2023-02-05:
hello, many people who updated their YI cameras regret that they updated the cameras to version 12 because they got problems with human detection, and they don’t know they can’t return the previous firmware version

NikNikolay7, 2023-02-05:
Существует ли такой же лёгкий способ, где нужно просто на отформатированную карту памяти распаковать прошивку для того чтобы вернуть предыдущую заводскуюпрошивку камеры  yi dome guard YRS3521
Я смотрел вашу статью с рекомендациями по восстановлению заводского бэкапа но Насколько я понял там применяется виртуальная машина Linux К сожалению для меня это пока очень сложно

NikNikolay7, 2023-02-05:
Is there an easy way where you just need to unpack the firmware on a formatted memory card in order to restore the previous factory firmware of the camera yi dome guard YRS3521
 I looked at your article with recommendations for restoring a factory backup, but as far as I understand, a Linux virtual machine is used there. Unfortunately, this is still very difficult for me.

roleoroleo, 2023-02-05:
If you want to create the image from windows, you can user mkimage from here:
https://www.mediawiki.compulab.com/w/index.php?title=U-Boot:_Images
but it works only for "factory" image.

morenod, 2023-02-05:
Can you compile it for h30?

NikNikolay7, 2023-02-05:
Большое спасибо ,попробую сделать так

roleoroleo, 2023-02-05:
I added the tools for windows.
Please try to use build.bat

NikNikolay7, 2023-02-05:
Thank you very much, I’m sure that someone who is more advanced than me
will help, I still can’t figure it out, I’ll hope that yi will fix it in
the next firmware, on many resources and on the application page in the
play store, many users complain about the updated version 12  yi camera
software

 Thank you for your help

pafade89, 2023-02-13:
Hi @roleoroleo,

I am checking your new beta with my Yi Home IFUS (y211ga) and it works fine.

I also have a Yi Outdoor 1080p IFUS Cam (h30ga). Is it possible to get a beta for these model from you as well for testing? The new firmware is called 11.1.26.

Cheers

roleoroleo, 2023-02-16:
@pafade89 
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/576#issuecomment-1432859992

pafade89, 2023-02-17:
Thank you very much. Installation worked and is currently running without problems.

maRoff, 2023-04-13:
Do you plan to support YI Dome Guard YRS.3019 (r30gb) at new firmware version 12.1.05?

roleoroleo, 2023-04-15:
This version is new for me.
Please, send me a backup copy.

maRoff, 2023-04-19:
Copy from: YI Dome Guard YRS.3019 (r30gb), firmware version 12.1.05
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/11274929/backup.zip)

roleoroleo, 2023-04-20:
Please, overwrite this file in your sd card:
[lower_half_init.sh.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/11283839/lower_half_init.sh.gz)
and check if the cam works.

maRoff, 2023-04-20:
Web panel is working, but rtsp stream is not.
[image]

roleoroleo, 2023-04-20:
Maybe Yi changed the frames.
Please, send me a copy of /dev/shm/fshare_frame_buf

maRoff, 2023-04-24:
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/11314559/fshare_frame_buf.zip)

roleoroleo, 2023-04-26:
The frame buffer is not changed, it should work.
Please try to run the server manually.
```
# kill wd_rtsp.sh
# kill rRTSPServer
# rRTSPServer -m r30gb -r both -a alaw -p 554
```
and check the output.

JohnHolg, 2023-04-28:
I have the same problem

roleoroleo, 2023-04-30:
Enable debug log with "-d" please.
Is rmm running?

maRoff, 2023-04-30:
What is rmm?

roleoroleo, 2023-04-30:
Sorry...
rmm is the main process that manages the cam (a Yi process).
You can check if it's running with a `ps` command executed in the shell.

maRoff, 2023-05-04:
Yes, rmm is running:
[image]

roleoroleo, 2023-05-04:
Maybe the frame buffer is locked and rRTSPServer is in loop.
Run this command twice:
`hexdump -C -n 16 /dev/shm/fshare_frame_buf`
You should see different outputs.

maRoff, 2023-05-09:
YI APP is working without any problem.

[image]

roleoroleo, 2023-05-14:
I don't understand...
Please try to run the process without audio and with log:
```
# killall wd_rtsp.sh
# killall rRTSPServer
# rRTSPServer -m r30gb -r both -a no -p 554 -d 7
```

maRoff, 2023-05-18:
It's not helped:
[image]
If is an option I will try to downgrade camera firmware.

maRoff, 2023-05-18:
I downgraded this camera firmware, but if you have other idea I can test it again. Thank you.

roleoroleo, 2023-05-19:
After downgrade is it working properly?

maRoff, 2023-05-19:
Yes, without any problem with your official version. I didn't test modified files from this thread.

closed 2023-09-17 by stale[bot]

Rohja, 2023-09-24:
Hi !

Seems like I'm encountering the same issue as @maRoff with my `Yi Dome Camera YRS.3019`.

I didn't use it for a long time and it was using an outdated firmware. After a backup (of version 9.x), I upgraded the firmware to `12.1.05_xx` using the Yi Home application on Android.

The app doesn't offer any other upgrade after that.

Following the setup instructions I get a working web panel but the RTSP server isn't responding (port seems closed) and any configuration change followed by a reboot result in (at least) the telnet and web servers being unavailable.

I checked that the `rmm` process is running and it is:

```
root@(none):~# ps | grep rmm
 1215 root     39944 S    ./rmm
 5533 root       980 S    grep rmm
```

Running `hexdump -C -n 16 /dev/shm/fshare_frame_buf` twice always output the same data:

 - First time:
```
root@(none):~# hexdump -C -n 16 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000010
```
 - Second time:
```
root@(none):~# hexdump -C -n 16 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000010
```

With the hack on the SD card and the web panel/telnet working, the video stream isn't loading in the Yi Home app: it get stuck loading `25%` until I get a timeout error.

I tried running `rRTSPServer` manually like suggested above:

```
root@(none):~# killall wd_rtsp.sh
root@(none):~# killall rRTSPServer
root@(none):~# rRTSPServer -m r30gb -r both -a no -p 554 -d 7
1695557197390: the size of the buffer is 1786156
1695557197395: mapping file fshare_frame_buf, size 1786156, to b6be7000
1695557197395: closing the file fshare_frame_buf
1695557197395: starting capture main loop
1695557197396: buf_idx_end == buf_idx_end_prev
1695557197406: buf_idx_end == buf_idx_end_prev
1695557197416: buf_idx_end == buf_idx_end_prev
1695557197427: buf_idx_end == buf_idx_end_prev
1695557197441: buf_idx_end == buf_idx_end_prev
1695557197452: buf_idx_end == buf_idx_end_prev
1695557197462: buf_idx_end == buf_idx_end_prev
1695557197472: buf_idx_end == buf_idx_end_prev
1695557197483: buf_idx_end == buf_idx_end_prev
1695557197493: buf_idx_end == buf_idx_end_prev
1695557197503: buf_idx_end == buf_idx_end_prev
1695557197513: buf_idx_end == buf_idx_end_prev
1695557197523: buf_idx_end == buf_idx_end_prev
1695557197534: buf_idx_end == buf_idx_end_prev
1695557197544: buf_idx_end == buf_idx_end_prev
[TRUCATED]
```

Do you want me to try [this version](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/571#issuecomment-1554619154) of `rRTSPServer` or to replace `lower_half_init.sh` by [this one](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/571#issuecomment-1515898469) and post logs ?

Let me know if I can help with any other informations/logs.

I can provide you with a backup of both the old `9.x` firmware as well as the recent `12.1.05_xx` firmware if needed.

If needed, I can get hardware serial access by opening the camera and soldering some wires, I'll just need some documentation for that task. 

Thanks !

roleoroleo, 2023-09-26:
rmm is running but the stream doesn't start (the buffer is empty).
So RTSP server can't work.

A telnet access is enough to debug the problem.
Did you restart from scratch with the sd card, after upgrade 9-12?
If no, please erase your card and apply the hack again.

Please send me a dump of these commands:
- `cat /backup/init.sh`
- `ps`
- `cat /tmp/sd/yi-hack/etc/system.conf`

Clean personal info from the last command.

Rohja, 2023-09-26:
First of all, thanks for your answer.

I tried the hack on version 9.x, the rtsp server wasn't working so I upgraded the camera without the sdcard after rebooting.
I then re-installed the hack on the SD card and did the steps of my previous message.

Content of `/backup/init.sh`:

```
#!/bin/sh
echo "--------------------------home app init.sh--------------------------"
SUFFIX=r30gb

PATH="/usr/bin:/usr/sbin:/bin:/sbin:/home/base/tools:/home/app/localbin:/home/base"
LD_LIBRARY_PATH="/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib:/tmp/sd:/tmp/sd/gdb"
export PATH
export LD_LIBRARY_PATH

ulimit -c unlimited
ulimit -s 1024
#echo /tmp/sd/core.exe[%e].pid[%p].sig[%s] > /proc/sys/kernel/core_pattern

echo 100 > /proc/sys/vm/dirty_writeback_centisecs
echo 50 > /proc/sys/vm/dirty_expire_centisecs
#echo 2048 > /proc/sys/vm/extra_free_kbytes

echo 400 > /proc/sys/vm/vfs_cache_pressure
echo 0 > /proc/sys/vm/drop_caches
echo 256 > /proc/sys/vm/min_free_kbytes

sysctl -w vm.dirty_background_ratio=2
sysctl -w vm.dirty_expire_centisecs=500
sysctl -w vm.dirty_ratio=2
sysctl -w vm.dirty_writeback_centisecs=100

sysctl -w fs.mqueue.msg_max=256
mkdir /dev/mqueue
mount -t mqueue none /dev/mqueue

mount tmpfs /tmp -t tmpfs -o size=32m
mkdir /tmp/sd
mkdir /dev/shm
mkdir /tmp/var
mkdir /tmp/var/run
mkdir /tmp/run

checkdisk
rm -fr /tmp/sd/FSCK*.REC
umount -l /tmp/sd
mount -t vfat /dev/mmcblk0p1 /tmp/sd

#----insmod hardware cyption------

#extract files
mkdir /tmp/ko/
mkdir /tmp/app/
mkdir /tmp/tools/

if [ -f /home/home_${SUFFIX}m ]; then
    echo "---/home/home_${SUFFIX}m exist, update begin---"
        dd if=/home/home_${SUFFIX}m of=/tmp/newver bs=22 count=1
        newver=$(cat /tmp/newver)
        if [ -f /home/homever ]; then
                curver=$(cat /home/homever)
        else
                curver=0
        fi
        echo check version: newver=$newver, curver=$curver
        if [ $newver != $curver ]; then
                ### cipher ###
                sleep 1
[... 179 lines cut ...]
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
HTTPD_PORT=80
USERNAME=
PASSWORD=
TIMEZONE=
FREE_SPACE=0
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=yes
SSH_PASSWORD=
CRONTAB=
DEBUG_LOG=no
```

Note: I didn't do any kind of configuration in the web panel. 

Let me know if you need anything else.

roleoroleo, 2023-09-26:
The cam is ok.
The hack is properly installed and all processes are running.
So it's not clear why it's not working.

- Clean the cache of the browser and try to configure the cam.
- Run ipc_cmd -x and check if the buffer is filled.

Rohja, 2023-09-26:
Running `ipc_cmd -x` does solve the buffer being empty and the rtsp server is now working !
(Btw this arg isn't documented in `ipc_cmd -h`, would you accept a PR for that change ?)

I'll check if this solution survive a reboot and try more in-depth configuration in the web UI later.

roleoroleo, 2023-09-26:
This parameter is intentionally not documented...
:)
And it won't survive reboot.

But the command is already executed in the system.sh.
You should try to add a sleep after line 27: https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/d69fdfc5072f2153c526db8c49aec11908709a4c/src/static/static/yi-hack/script/system.sh#L28
Maybe 1 second.

Rohja, 2023-09-26:
Ok, so I think I figured out what's (part of) the issue.

In `/tmp/sd/yi-hack/etc/system.conf` I have `DISABLE_CLOUD=no` by default so [this part of the script run](https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/d69fdfc5072f2153c526db8c49aec11908709a4c/src/static/static/yi-hack/script/system.sh#L175-L201) and [start_buffer](https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/d69fdfc5072f2153c526db8c49aec11908709a4c/src/static/static/yi-hack/script/system.sh#L228) is never executed. 

Setting `DISABLE_CLOUD=yes` did the job. 

Is that a normal behavior ? Shouldn't the buffer be filled no matter is cloud features are enabled or not ?

roleoroleo, 2023-09-30:
When DISABLE_CLOUD=no, you shouldn't need to start the buffer, it should do it by itself.
It's the standard behavior of the cam.

It's definitely a race condition.

Does the cam access to internet?

Rohja, 2023-10-02:
Yes the camera has access to the internet.

Alexselutin, 2023-10-07:
> Я добавил инструменты для Windows. Пожалуйста, попробуйте использовать build.bat

Can you send it to me too please?

roleoroleo, 2023-10-12:
> Yes the camera has access to the internet.

Try to add a sleep (2 seconds) before line 185:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/abaefd499f1e79e7a3f1ba3075c327b994897652/src/static/static/yi-hack/script/system.sh#L185

> > Я добавил инструменты для Windows. Пожалуйста, попробуйте использовать build.bat
> 
> Can you send it to me too please?

It's released:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick

Cofffeek, 2023-10-28:
> If your cam is updated with fw 12 and you want to try a new beta, please try these files: [r35gb_x.x.x.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10567167/r35gb_x.x.x.zip) [...]

Good afternoon, I am the owner of two cameras YRS 3521. 
One camera failed during the update, when copying a jump from a working camera using ch341a, I accidentally deleted the camera in the application. After that, I can't connect in the application to add not the first not the second camera.  Says the Wifi connection, then after 1-2 minutes passes and says: - Bearing has timed out, please reset the camera. Overloaded that one that the second camera nothing happens please help!

roleoroleo, 2023-11-03:
If the hack is installed, you should use the unbrick procedure to restore the factory fw.
Than, reset the cam with the button and try again.
If it doesn't connect to the wifi, try another access point.
And disable 5 GHz network if you have it.