roleoroleo/yi-hack-Allwinner-v2 #572: hack version 0.2.5 Yi Dome Guard chinese version can not use app YI

tummerry, 2023-02-02:
### Discussed in https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/570

<div type='discussions-op-text'>

<sup>Originally posted by **tummerry** February  2, 2023</sup>

help me 
hack version 0.2.5 can not link app YI Home and YI Life
Is there a way to help me connect?
I change xxCHxxxxxxxx xxUSxxxxxxxx finish
[image]
[image]
</div>

roleoroleo, 2023-02-02:
I don't understand.
Did you change the serial number from CH to US?

tummerry, 2023-02-03:
yes i change CN    HFCNY36SN04IT8191216    to US    HFUSY36SN04IT8191216 succeed.
But unable to connect YI Home app and YI Life.
I have a backup file for you to look at.
[image]
[backup_CN.ZIP](https://github.c
[backup_US.ZIP](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10574574/backup_US.ZIP)
om/roleoroleo/yi-hack-Allwinner-v2/files/10574571/backup_CN.ZIP)

Please help me

roleoroleo, 2023-02-04:
Another user told me that after the serial number change the cam would not connect to the app.
Maybe there is a check in the number.
I will study your backup but I don't know if I'm able to fix the problem.

roleoroleo, 2023-02-04:
Did the cam connect to the cloud before changing serial number?

roleoroleo, 2023-02-04:
Please, try to change also TNPCHNG-xxxxxx-YYYYY to TNPUSAL-xxxxxx-YYYYYY.

tummerry, 2023-02-06:
[image]
I changed and still can't connect YI App

roleoroleo, 2023-02-09:
The first char is not ok.
[image]
Please, check again your changes.