roleoroleo/yi-hack-Allwinner-v2 #574: Yi dome guard YRS3521problems after updating the official factory firmware

NikNikolay7, 2023-02-03:
hello, my camera yi dome guard YRS 3521, international EU version, after updating the official firmware version to version 12.1.47.2_202301121850, it began to react much worse to the appearance of a person, I want to return the previous old version of the firmware, but I do not have a backup copy and nowhere on the Internet  I can not find the firmware for this camera, YI technical support refuses to send the previous firmware
maybe someone has a firmware for a camera like mine or share a backup copy, please help

NikNikolay7, 2023-02-03:
[image]

roleoroleo, 2023-02-04:
Here it is a backup:
[mtdblocks.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10609179/mtdblocks.tar.gz)

NikNikolay7, 2023-02-04:
Спасибо ,попробую восстановить

NikNikolay7, 2023-02-04:
Извините, если вам не трудно пожалуйста подскажите как мне устновить этот бэкап

Мне нужно  отформатировать карту памяти в fat 32 и распаковать на карту памяти файлы которые вы мне скинули?

NikNikolay7, 2023-02-04:
> Вот бэкап: [mtdblocks.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10609179/mtdblocks.tar.gz)
 Спасибо

NikNikolay7, 2023-02-04:
Чтобы из этого файла восстановить старую версию прошивки мне нужно иметь linux машину и уметь ей пользоваться?
Просто отформатировав карту памяти в fat 32 и  распаковать на неё эти файлы ,не получится восстановить старую версию заводского программного обеспечения?
Я никогда не имел дело с linux машиной

NikNikolay7, 2023-02-05:
hello thank you

 Sorry, but do you have instructions on how to restore the previous version
of the firmware using this backup, I read your publications and realized
that I need a linux virtual machine and the ability to use it

 I have never used linux, I only have a laptop with win 7

NikNikolay7, 2023-02-05:
Do you have a file ready for unpacking to a memory card for flashing and returning the old firmware version of this camera?

roleoroleo, 2023-02-05:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/571#issuecomment-1417983679

NikNikolay7, 2023-03-08:
Нигде не могу найти прошивку для моей камеры, инструменты для восстановления бекапа не запускаются на моём компьютере
У меня есть вторая такая же камера ,которую я не обновлял
Если кто то знает другой способ как вернуть старую прошивку на обновленную камеру ,пожалуйста помогите 
Или может у кого то есть прошивка
9.0.46.00_202110081405
Для камеры
YRS3521
Dome guard

roleoroleo, 2023-03-09:
If you have a second camera (the same model), you can backup the firmware on the second cam and restore it to the 1st one.

NikNikolay7, 2023-03-09:
Да у меня есть вторая такая же камера которую я не обновлял
Ну те инструменты для Windows которые нужны для того чтобы восстановить резервную копию прошивки на моём компьютере не запускаются
Я устанавливал последнюю версию вашего хака и видел что она на карте памяти автоматически создаёт папку Backup
Вы не знаете какой-то другой способ чтобы восстановить резервную копию прошивки на обновлённой камере?

Rudebois, 2023-04-08:
> I installed your hack on the camera yesterdayYRS3521 9.0.46.00_202110081405 and it clearly quickly and accurately records people on video, I haven’t figured out the settings yet, I want notifications about people in the frame to come, but even so it works very well, thank you

can you Help me share a step by step for downgrade the firmware from 12. to 9. ??

i have same issue and same model like you
look what the yi dev done to my dome guard, i put the camera outside, the traffic of the road so bussy and then the playback must be longgg, not like this!
[image]
[image]
[image]

YRS 3521
RFUS 
R35GB

closed 2023-08-08 by stale[bot]

Alexselutin, 2023-10-09:
> Here it is a backup: [mtdblocks.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10609179/mtdblocks.tar.gz)

what firmware version is this backup from?

roleoroleo, 2023-10-12:
r35gb

Cofffeek, 2023-10-26:
> I installed your hack on the camera yesterdayYRS3521 9.0.46.00_202110081405 and it clearly quickly and accurately records people on video, I haven’t figured out the settings yet, I want notifications about people in the frame to come, but even so it works very well, thank you

Добрый день, вы решили данную проблему? Можете скинуть пожалуйста инструкцию, как через программатор, или через что делали флешку меняли прошивку. Было тоже две камеры одну обновил и прошивка слетела. Скачал через программатор со старой. И в итоге две камеры слетели xD

Cofffeek, 2023-10-26:
> I installed your hack on the camera yesterdayYRS3521 9.0.46.00_202110081405 and it clearly quickly and accurately records people on video, I haven’t figured out the settings yet, I want notifications about people in the frame to come, but even so it works very well, thank you

Скиньте пожалуйста на почту копию  вашей камеры dump dingoo999@gmail.com

Cofffeek, 2023-10-28:
> Я вчера установил ваш хак на камеру YRS3521 9.0.46.00_202110081405 и он четко быстро и точно записывает людей на видео, с настройками пока не разобрался, хочу, чтобы уведомления о людях в кадре приходили, но даже так работает очень что ж, спасибо тебе

Скиньте пожалуйста инструкцию...