roleoroleo/yi-hack-Allwinner-v2 #575: cannot disable Alarm-Sound

Bodengriller, 2023-02-04:
I had to downgrade my firmware from 12.0.37.0_202211241030 and did it like this:
- first I loaded the original y211ga_11.0.36.7_202202170022 from [Kami](https://www.kamihome.com/firmware/) onto my IFUS 
- and then restored the y211ga_9.0.36.02_202105081647 backup uploaded and as described [here](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/535) in this thread.  (first backup, then home)
- I then "installed" yi-hack y211ga_0.2.5.

Everything looks fine now, but my camera always makes this silly alarm sound when it detects motion.

In the app I am missing the slider to disable the alarm sound.
[like here](https://help.yitechnology.com/hc/article_attachments/1260802072269/SmartDetectionH31_copy.png)

is there a way to disable it by ssh or something?
That Sound is driving me crazy

what can I do?

Bodengriller, 2023-02-04:
hm... 
I loaded the y211ga_9.0.36.02_202105081647 backup again, to get rid of yi-hack,
then I updated again to 12.0.37
after that I used [this Backup from @dreimer1986 ](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/388#issuecomment-1062645815) to go back to 9.0.36 and installed yi-hack again.

Now it works.

Perhaps a problem with [this backup-files from @roleoroleo ](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/535#issuecomment-1327754657)?

closed 2023-02-04 by Bodengriller

roleoroleo, 2023-02-04:
> Perhaps a problem with [this backup-files from @roleoroleo ](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/535#issuecomment-1327754657)?

Maybe but strange.