roleoroleo/yi-hack-Allwinner-v2 #576: YI OUTDOOR  1080 reboot loop after firmware updated to 11.1.26.1

morenod, 2023-02-05:
Hack was working fine with 0.2.5 but yesterday I updated the stock firmare to 11.1.26.1 and now camera is on a reboot loop.

I have flashed card again with 0.2.5 yi-hack but when using it, camera reboots (without SD card, it works fine)

```
$ ls -1
backup
Factory.done
log
lower_half_init.sh
record
yi-hack
```

morenod, 2023-02-05:
I have already enabled swap and disabled snapshots but it keeps rebooting

roleoroleo, 2023-02-05:
What's your model suffix?
At the moment fw 11 is not supported.
Please, make a backup copy of the flash and share it with me.

morenod, 2023-02-05:
[backup.tar.gz](https://github.com/morenod/yi-hack-Allwinner-v2/raw/yi_outdoor_fw_11/backup/backup.tar.gz)

Model: Yi Outdoor Camera 1080p (YHS_3020)

Base Version: 11.1.26.1_202301191526
Suffix: IFUSY593GX

Thanks!

micruz76, 2023-02-11:
Same here.

YI Outdoor Camera 1080p YHS_3020

Firmware after update: 11.1.26.1_202301191526

kubikub, 2023-02-15:
Hello, 
I did the upgrade to 11.1.26.1_202301191526 and now the hack do not work. 
I there a way to downgrade the 9.26 firmware ?
I have the yi Outdoor YHS-3017 with RFUS Y59IR

roleoroleo, 2023-02-16:
Beta release for fw 11:
[h30ga_0.2.5.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10754875/h30ga_0.2.5.tar.gz)

morenod, 2023-02-16:
Confirmed it is working fine

[image]

Thanks!!

micruz76, 2023-02-21:
Works fine. Thanks alot

roleoroleo, 2023-02-21:
Did you install the released version?

pafade89, 2023-02-21:
> Beta release for fw 11:
> [h30ga_0.2.5.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10754875/h30ga_0.2.5.tar.gz)
> 

Hi @roleoroleo your beta version worked great for me.  Now I have installed the new official release 0.2.6 which contains all features and now the person detection no longer works with hg30. What can be the reason?

Thx in advanced P

closed 2023-07-09 by stale[bot]

grzesiczek1, 2026-11-01:
Data point for 11.1.26.7, not a fix, but it may help narrow this down.

Three Yi Outdoor 1080p (h30ga) on stock 11.1.26.7_202606301651 (OTA from 11.1.26.6) run yi-hack 0.3.7 from the SD card with no reboot loop: 8–10 h uptime so far plus several manual reboots, with DISABLE_CLOUD=yes. One of them also ran 11 min with the cloud enabled, also fine.

The difference from the reports above: these cameras got the hack in 2024, before the OTA. The OTA rewrites only the home partition, so `/backup/init.sh` (mtdblock4) still has the patch from that install (telnetd + `source /tmp/sd/lower_half_init.sh`). So the hack starts through that old patch, not through a fresh install on .6/.7.

That suggests the 0.3.7 runtime works on the 11.1.26.7 home partition. If you see the loop after a fresh install, comparing your `/backup/init.sh` with the one described above might show whether the install step or the runtime is the problem. Note also the version difference: 0.3.7 here vs 0.3.4/0.3.6 in the reports.

One more thing worth ruling out: one of these cameras had a "reboot loop" in late 2024 that I first blamed on the firmware, but its SD card turned out to be dying (writes reported success but did not persist). A known-good card is a cheap test.

I can't share full flash dumps (they contain Wi-Fi credentials), but I'm happy to compare specific files or run checks.