roleoroleo/yi-hack-Allwinner-v2 #577: Yi 1080p FTP Upload doesnt tranfer files when "create directory tree" is enables

TwistedBE, 2023-02-05:
I've got a few Yi 1080p IFUS and BFUS cam's running Yi-Hack 0.2.5 with 64gb SD cards for a while now.

Cam's upload the video clips to an FTP server that is running on a small server thats always on inside my house.
But this ONLY works when "create directory tree" is not enabled !!!

Although I can make it work without the map structure its a real pain in the ass to filter inside one big map with thousands of files.
When enabeling "create directory tree" nothing gets transferd to the FTP server... (cam has been restared several times)

A second question i have is will the cam keep transfering new files once's the SD card is full (64gb).
Or will the oldest file be deleted on the FTP server just like it happens on the SD card ? (and if not can this be enabled ?)

Thanks for the help !

roleoroleo, 2023-02-05:
Already seen this problem.
The command for directory creation fails, probably it's not compatible with your ftp server.
Try with another daemon, if you can.
If you want to debug what it happens, try to run the script manually:
- Open a ssh session
- Run `/tmp/sd/yi-hack/script/ftppush.sh cron`
- Check the output

The cam removes old files in your SD card trying to mantain free space, based on the setting:
[image]
About FTP server, no files are deleted.

TwistedBE, 2023-02-05:
I'm using "Filezilla server" thats running on a windows 10 media pc.

2023-02-05 [20-02-57] [INFO] === SERVICE START ===
2023-02-05 [20-02-57] [INFO] checkFiles
ftpput: unexpected server response to STOR: 550 Couldn't open the file or directory
2023-02-05 [20-02-58] [ERROR] uploadToFtp: ftpput FAILED.
2023-02-05 [20-02-58] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2023Y02M05D14H/12M00S60.mp4]. Retrying in 45 s.
2023-02-05 [20-02-58] [INFO] === SERVICE STOPPED ===

This is the log from Filezilla itself, READ+WRITE premissions are given (hence file upload without folder structure works)

[image]

roleoroleo, 2023-02-09:
Does folder /Zijgevel exist?
Maybe the user with a space char is not recognized.
Try to run this command manully from a terminal:
`echo -e "USER YOUR_USERNAME\r\nPASS YOUR_PASSWORD\r\nmkd /Zijgevel/2023Y02M05D14H\r\nquit\r\n" | nc -w 5 YOUR_IP 21 | grep "2023Y02M05D14H"`
changing YOUR_USERNAME, YOUR_PASSWORD and YOUR_IP.

TwistedBE, 2023-02-10:
Yeah the folder exists and I dont think that the user is a problem.
Cause everything works perfectly without "create directory tree" enabled.
The camera uploads all the video files to the ftp server in single files inside the "Zijgevel" folder.

roleoroleo, 2023-02-11:
mkdir command and copy command are sent in a different manner.
Please try to run the command above from the command line.

closed 2023-06-11 by stale[bot]