roleoroleo/yi-hack-Allwinner-v2 #580: Issue with Yi 1080p Home RFUS on 12.0.35

nbernier, 2023-02-11:
I copied the firmware to the sd card and insert it to the cam. Hack is deploying but the camera keep rebooting after it complete. It never comes back to a working state. I see the folder Factory gets renamed on the sd card. I was trying to find the logs in the folder but didn't find anything. My cam was working with the previous firmware.

roleoroleo, 2023-02-11:
Fw 12 is not supported at the moment.

nbernier, 2023-02-11:
Yes I saw that is was still a WIP but I would gladly help.

zenhaust, 2023-02-12:
This works for me, for same problem.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/540#issuecomment-1352252171

nbernier, 2023-02-13:
The cam boot up now but I can't access port 8080 nor ssh. Seems like the custom firmware is not present

zenhaust, 2023-02-13:
Mmm. Strange situation. The patch works fine for many users. ¿Can you see your cam in the router?

roleoroleo, 2023-02-13:
I published a new beta release, if you want:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/571

nbernier, 2023-02-13:
Hi,
Yes. I can see the cam in my network. The cam works with the YI Home app and I can see the ip adresse from the info tab. 

I have installed the beta version. It boots up but I can't connect on port 8080 or 22. Anything else I can try? Where can I find the logs?

nbernier, 2023-02-15:
yes it completed. The Factory is renamed to Factory.done and there is a backup folder

roleoroleo, 2023-02-15:
Try to enter with telnet (port 23) and list the running processes:
`ps`

nbernier, 2023-02-15:
I can't connect to it with Telnet. Is Telnet open by default on the cam? I am on the same network and I can see it online on the router with hostname yi-hack

roleoroleo, 2023-02-16:
Yes, telnet should work if the hack completed successfully.

nbernier, 2023-03-28:
latest version fixed it. Thank you very much

closed 2023-03-28 by nbernier